Mrnt/OctoPrint-FlashForge #2: State: Error: Failed to autodetect serial port, please set it manually.

Dmurph24, 2019-08-07:
I followed the instructions listed and was unable to connect my PowerSpec Ultra. I receive the following error when trying to connect:

```
State: Error: Failed to autodetect serial port, please set it manually.
```

Mrnt, 2019-09-04:
@Dmurph24 apologies for the delay in responding. 

I have a PowerSpec Ultra that I tested with but there may be some difference in the device id based on hardware versions etc. Mine uses vendor ID 0x2b71 and a device ID 0x00ff but maybe I fubarred the ID's when I tried to reflash its firmware.  Please have another go with the plugin and if it still does not detect your printer then please provide the vendor and device ID's. I have added some debugging to the detection code which should provide the vendor and device ID's if you enable logging for the plugin (Settings > Logging  and under "Logging Levels" select "DEBUG" for octoprint.plugins.flashforge then hit "Save"). Alternatively, you can look them up using the tools of your OS platform.

Mrnt, 2019-09-04:
@Dmurph24 also:
1) check the FlashForge plugin is enabled in Settings > Plugin Manager
2) go to the Terminal tab to look for some debug messages from the plugin

Dmurph24, 2019-09-04:
@Mrnt Im out of town this week, but will be back next week and will try this at that point. Thanks for getting back to me!

Dmurph24, 2019-09-04:
If I remember correctly, I installed the plugin successfully but it never actually appeared in the plugin manager, even after a restart.

Mrnt, 2019-09-07:
@Dmurph24 I finally set up OctoPi (which is painfully slow for me on a Pi B 1 or 2), previously I was using my desktop as I was in "proof of concept" mode. It looks like there were a couple of issues that would trip up an install.
1) libusb1 was not being installed automatically, which would cause the plugin not to appear in the plugin list. As of 0.1.3 this should be addressed
2) OctoPi/Octoprint on Linux does not have permission to access the USB interface of the printer, which caused the connection to fail before getting any useful feedback in the logs.
It should now throw an error when trying to connect in the case of a permissions problem and indicate that a udev rule needs to be created with instructions on what the rule should be and where it should be created.

Dmurph24, 2019-09-10:
It looks like this build installs correctly and I see it in the plugin manager now, but Im getting a new error message when connecting now:
```
Unexpected error while connecting to serial port: AUTO FlashForgeError: 'Unable to connect to FlashForge printer - permission error.

On OctoPi/Linux add the following line to
 /etc/udev/rules.d/99-octoprint.rules:

SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"

Then reboot your system for the rule to take effect.

' @ comm.py:_openSerial:2611 (hook flashforge)
```

Dmurph24, 2019-09-10:
Followed the instructions to this message and everything is now working correctly. This is incredible, thank you!

closed 2019-09-10 by Dmurph24