Mrnt/OctoPrint-FlashForge #3: Feature Request: Allow 'Load and Print' to work

Dmurph24, 2019-09-17:
I know this is currently in the documentation and the only way to currently print the object is by uploading to the sd card, but it would be very nice if you could save files to Octoprint and decide to print them later. It also looks like this could help include the print time remaining and gcode layers.

Is this possible with the terminal output received from the usb?

Mrnt, 2019-09-17:
The way Octoprint works is that when you select files to print that have been saved to Octoprint, it feeds the file directly to the printer, line by line. That is where we get a bit unstuck with these FlashForge printers - while they will accept commands sequentially, they do not seem to give enough feedback as to when the buffer is full etc, so it is proving hard to print directly from Octoprint without the communication to the printer hanging because the printer stops accepting commands due to full buffer/waiting for print head to heat up/being bloody minded. However,  I am trying to do exactly that, just know that it is hard to get it to work and may prove unfeasible.

You should see progress and be able to pause/resume/restart/cancel when printing from an sd upload however and as you mentioned it is using the feedback via usb for those progress updates. I have to check if it does the print time remaining when doing it that way. 

re gcode layers, you are right - they will typically only work when printing directly from octoprint as it opens the file and parses the code to build that layer simulation.

Dmurph24, 2019-09-17:
I figured this would be difficult, if you need help testing or anything, let me know.

brett19, 2020-08-29:
@Mrnt: This new release that allows you to print directly from OctoPrint.  Does this still stream the data to the SD card on the printer, or does it send it byte-wise over USB?

Mrnt, 2020-08-29:
@brett the "Upload to SD" card functionality >should< still work as before, but the new release should now allow uploading of a file to OctoPrint (so it appears on the left side in the File list) and then you can print from within OctoPrint (and see print progress in the GCode Viewer tab) which will stream commands over the USB connection to the printer one by one...

Please note that I only have access to two models of printer (FlashForge Finder v1, PowerSpec Ultra) and so I am reliant on users to test compatibility of features with other models. So far it appears that there may be issues with the Finder v2 with the direct print/command streaming, which I am working through. It would appear that FlashForge may have made at least that printer (the Finder v2) not handle certain commands via direct print although it apparently will handle them when in a file uploaded to the printer's SD card.

Dmurph24, 2020-09-05:
@Mrnt I updated and tried this new feature today. I loaded and started printing the file, however, it started printing about 10mm higher than the build plate. I checked to make sure it wasnt the GCode and it printed properly using the upload to SD method. Looks like there is some sort of offset issue on the Z axis with this?

Mrnt, 2020-09-07:
@Dmurph24 Others are reporting the same thing. I'm not sure right now what is happening - seems like something that would happen if the Z axis isn't homing properly.

tckb, 2021-02-25:
@Mrnt thanks for making this plugin! this works like a charm for "Upload from SD". But as you know this is highly restrictive.  I can't fully utilize the full capabilities of octoprint :-( 

as reported by others ⬆️  I have same issue. Print directly results in z-axis issues. Somehow the bed is doesnt level with nozzel and nozzel doesn't extrude. 

Machine:
Flashforge Dreamer NX
Firmware: 1.6.XXX

Mrnt, 2021-03-11:
@tckb does print directly fail every time for you? I am usually able to get it to work so if you have a sample that consistently fails can you turn on debugging for the plugin (see instructions on the ReadMe) and upload the log and the sample file.

tckb, 2021-03-11:
@Mrnt  Well it fails eveytime when you print _directly_  if works flawlessly when you "Upload to SD". The print via SD card upload is funtionally very limited. I am unable to use the awesome features of OctoPrint :-( 

I will upload the debugging logs soon-ish.

tckb, 2021-03-13:
@Mrnt  here's the latest model I tried.  
Image:  
[image]
gcode
[Cam3_0.2mm_ABS_1h24.g.txt](https://github.com/Mrnt/OctoPrint-FlashForge/files/6135251/Cam3_0.2mm_ABS_1h24.g.txt)

Mrnt, 2021-12-06:
Sorry for the delay!

That gcode file does not look very comparable with FlashForge printers - take a look at the commands in the file and then take a look at the [gcode](https://github.com/Mrnt/OctoPrint-FlashForge/wiki/G-Code-Reference#example-14) Wiki. It's hard to imagine that it would work if you uploaded it directly to the SD card...

Mrnt, 2021-12-06:
Closing since this feature has been added (though admittedly is not super stable).

closed 2021-12-06 by Mrnt