Mrnt/OctoPrint-FlashForge #4: Support for Flashforge Ultra 3D Printer

twistedcustomdevel, 2019-12-06:
Hello. Thanks for writing up this wonderful piece of software that has allowed me to take control over the functionality of this proprietary printer. It feels good to have control from my little Raspberry Pi instead of my Windows workstation.

I ran into a technical glitch trying to connect to my 3D printer. Apparently FlashForge has used multiple USB Vendor ID's. Here is the output from my lsusb -v command:

T:  Bus=01 Lev=02 Prnt=02 Port=01 Cnt=02 Dev#= 21 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=0315 ProdID=0001 Rev= 2.00
S:  Manufacturer=Flashforge
S:  Product=FlashForge Dreamer 3D Printer
S:  SerialNumber=00000000050C

Bus 001 Device 021: ID 0315:0001
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0315
  idProduct          0x0001
  bcdDevice            2.00
  iManufacturer           1 Flashforge
  iProduct                2 FlashForge Dreamer 3D Printer
  iSerial                 3 00000000050C
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002e
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           4
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

Apparently they assigned a Vendor ID of 0x0315 instead of the one that is currently encoded in __init__.py for FlashForge that is 0x2b71. Changing the 0x2b71 value to 0x0315 and manually installing the modified source in the plugin manager yielded another error in the log:

' @ comm.py:_openSerial:2661 (hook flashforge)
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2661, in _openSerial
    serial_obj = factory(self, self._port, self._baudrate, settings().getFloat(["serial", "timeout", "connection"]))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_flashforge/__init__.py", line 129, in printer_factory
    serial_obj = flashforge.FlashForge(self, comm, self.vendor_id, self.device_id, read_timeout=float(read_timeout))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_flashforge/flashforge.py", line 55, in __init__
    "SUBSYSTEM==\"usb\", ATTR{{idVendor}}==\"{:04x}\", MODE=\"0666\"\r\n\r\nThen reboot your system for the rule to take effect.\r\n\r\n".format(vendor_id))
FlashForgeError: Unable to connect to FlashForge printer - permission error.

On OctoPi/Linux add the following line to
 /etc/udev/rules.d/99-octoprint.rules:

SUBSYSTEM=="usb", ATTR{idVendor}=="0315", MODE="0666"

Then reboot your system for the rule to take effect.

After using a text editor to create a new udev rule to reflect the one in the log I was able to successfully connect to and control all functions of the printer. 

Please find the attached modified __init__.py file for your perusal. Note it is modified only for my use case and it may be helpful to others to modify it to account for both Vendor ID's.

[octoprint_flashforge_tmi.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/3932693/octoprint_flashforge_tmi.zip)

Mrnt, 2019-12-10:
@twistedcustomdevel Thanks for all the info! I'm a bit confused about the title - are you saying it is actually a PowerSpec Ultra 3D Printer but is reporting itself as a FlashForge Dreamer?

I have updated the plugin with that vendor ID for FlashForge from your output so you should be able to run the update from within OctoPi and get the latest version with the fix and hopefully it will now see your printer automatically.

twistedcustomdevel, 2019-12-11:
Thanks for the update. 

Yes it reports as a Flashforge Dreamer but it is most certainly not.

[image]
[image]

Mrnt, 2019-12-11:
@twistedcustomdevel Oh that is curious - so I should probably change the code so that it reports your machine as a "PowerSpec Ultra 3DPrinter" in the debug log.

I also have an Ultra 3D but it reports as a USB device called 'Powerspec Ultra 3D Printer' but with the FlashForge vendor ID and a product ID of 0x00ff (which is where I got the PowerSpec USB ID already in the code). It reported the following to Octoprint:
Recv: Machine Type: PowerSpec Ultra 3DPrinter
Recv: Machine Name: PS3D-Ultra
Recv: Firmware: V2.4 20160407
Not sure if you can update your firmware with newer PowerSpec firmware or whether it matters.

Since it reports with the FlashForge USB ID I tried to re-flash the firmware with Dreamer firmware which did not work but I was able to swap the logic board for one from a FF Dreamer which works (unfortunately it means no WiFi unless I fork out money for the Dreamer WiFi module). 

Now, like you I'm trying to drive it with this project running on a Pi... :-)

twistedcustomdevel, 2019-12-11:
Wi-Fi on mine is hit or miss. Mostly miss. I am definitely a proponent of either USB wired connections or printing from the SD Card.

I am currently out of acetone. Need a little to clean off the plate and make some more juice to make sure the prints stick. Will attempt to run some benchmark prints tomorrow.

Next upgrade will be a borosilicate glass plate.

Further do you plan on supporting the LED lights on this printer? It would be nice to flip them up from the Octoprint interface if it was not too much to find what commands activate them.

Mrnt, 2020-01-08:
I had not thought about the lights. I know that FlashPrint supports them so obviously it is possible. I want to try to get the file upload and printing to be more solid first.

closed 2020-01-08 by Mrnt