Mrnt/OctoPrint-FlashForge #7: Error connecting / printing: USB Error readraw() (-9)

Dmurph24, 2020-01-03:
Everything has been working for awhile. I recently updated and Im having a new issue. Im not sure if its related to the update or not, but I receive this error when attempting to connect:

```
Unexpected error while reading serial port, please consult octoprint.log for details: FlashForgeError: 'USB Error readraw() (-9)' @ comm.py:_readline:2793
```

Occasionally it will connect, but after I upload and print, I receive this error again.

Mrnt, 2020-01-07:
@Dmurph24 - sorry for the delay. A couple of questions:
- what platform are you using for Octoprint (Windows/Mac/Pi/etc)?
- do you see that issue only after attempting a print?
- what file are you uploading to the SD card - a .gx file created by FlashPrint?

ammoniak, 2020-01-08:
I have the same error on a new installation on a pi. Do I need to install some kind of driver or configure anything special to get the tty working? I believe that might be the issue in my case.

Dmurph24, 2020-01-08:
@Mrnt 
1. Raspberry Pi 
2. The error occurs when I try to connect to the printer. Occasionally it will connect, but when I click on "Upload + Print" button it would show that error and not print
3. I am uploading a gcode file

I would like to note this worked great until the most recent update

Mrnt, 2020-01-08:
@ammoniak - did you enable permissions for USB access? In the Troubleshooting section of the ReadMe there's a note about it and the Terminal tab of OctoPi should supply the specific dev rule you need to add.

Mrnt, 2020-01-08:
@Dmurph24:

- What printer are you using?
- Did you also recently update OctoPrint?
- Can you enable logging for the plugin and then paste in the section of octoprint log where it attempts the upload? It might help shed some light.

[image]

ammoniak, 2020-01-09:
I deleted the Octopi (put the image on the pi again) and went through all the setup steps again, including the udev rules, etc. and it works now. I probably made a mistake on the first try... I'm not sure where though.

Mrnt, 2020-01-09:
@ammoniak glad it's working now! For me at least, connecting to the printer and controlling the temp, homing, etc is usually pretty reliable. 
Uploading to the SD card however is flaky and DOES sometimes result in connection loss with the USB stack left in a weird state. I almost have a fix ready for making that more stable.

Dmurph24, 2020-01-09:
Printer: Powerspec Ultra
Yes, I recently upgraded octoprint to the newest version, as well as your plugin from the previous version

Below is a log of the failure:
```
printer_safety_check
2020-01-09 16:09:01,973 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-01-09 16:09:01,978 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-01-09 16:09:01,992 - octoprint.events.fire - DEBUG - Firing event: UpdatedFiles (Payload: {'type': 'gcode'})
2020-01-09 16:09:01,993 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x69977d70>>
2020-01-09 16:09:01,988 - octoprint.plugins.flashforge - INFO - Starting SDCard upload from tesla_body_pla.gcode to tesla_~1.gx
2020-01-09 16:09:01,998 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-01-09 16:09:02,023 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-01-09 16:09:02,023 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-01-09 16:09:02,025 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2020-01-09 16:09:02,040 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-01-09 16:09:02,044 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-01-09 16:09:02,048 - octoprint.events.fire - DEBUG - Firing event: MetadataAnalysisStarted (Payload: {'origin': 'local', 'path': u'tesla_body_pla.gcode', 'type': 'gcode', 'name': u'tesla_body_pla.gcode', 'file': u'tesla_body_pla.gcode'})
2020-01-09 16:09:02,051 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x69977d70>>
2020-01-09 16:09:02,054 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-01-09 16:09:02,055 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-01-09 16:09:02,058 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-01-09 16:09:02,061 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2020-01-09 16:09:02,066 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-01-09 16:09:02,086 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-01-09 16:09:02,092 - octoprint.events.fire - DEBUG - Firing event: TransferStarted (Payload: {'remote': u'tesla_~1.gx', 'local': u'tesla_body_pla.gcode'})
2020-01-09 16:09:02,095 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x69977d70>>
2020-01-09 16:09:02,103 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-01-09 16:09:02,108 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-01-09 16:09:02,112 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-01-09 16:09:02,115 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2020-01-09 16:09:02,120 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-01-09 16:09:02,121 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-01-09 16:09:02,123 - octoprint.events.fire - DEBUG - Firing event: Upload (Payload: {'path': u'tesla_body_pla.gcode', 'name': u'tesla_body_pla.gcode', 'file': u'tesla_body_pla.gcode', 'target': u'sdcard'})
2020-01-09 16:09:02,125 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x69977d70>>
2020-01-09 16:09:02,143 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-01-09 16:09:02,146 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-01-09 16:09:02,160 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-01-09 16:09:02,161 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2020-01-09 16:09:02,162 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-01-09 16:09:02,165 - octoprint.plugin - DEBUG - Calling on_event on tracking
2020-01-09 16:09:02,521 - octoprint.events.fire - DEBUG - Firing event: TransferFailed (Payload: {'remote': u'tesla_~1.gx', 'local': u'tesla_body_pla.gcode', 'time': 10})
2020-01-09 16:09:02,532 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x69977d70>>
2020-01-09 16:09:02,537 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-01-09 16:09:02,551 - octoprint.plugin - DEBUG - Calling on_event on announcements
2020-01-09 16:09:02,558 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2020-01-09 16:09:02,564 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2020-01-09 16:09:02,570 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2020-01-09 16:09:02,576 - octoprint.plugin - DEBUG - Calling on_event on tracking
```

Mrnt, 2020-01-10:
@Dmurph24 I think you not have enabled debugging specifically for the FlashForge plugin (as in the pic above) or may need to restart Octoprint to see debug messages from the plugin which should be something like this for an upload:

`
2020-01-10 09:15:12,292 - octoprint.plugins.flashforge - DEBUG - Sent: 99.00% 47104/47279
2020-01-10 09:15:12,292 - octoprint.plugins.flashforge - DEBUG - FlashForge.writeraw() called by thread SD Uploader
2020-01-10 09:15:12,293 - octoprint.plugins.flashforge - DEBUG - Sent: 100.00% 47279/47279
2020-01-10 09:15:12,293 - octoprint.plugins.flashforge - DEBUG - FlashForge.sendcommand() M29
2020-01-10 09:15:12,294 - octoprint.plugins.flashforge - DEBUG - FlashForge.writeraw() called by thread SD Uploader
2020-01-10 09:15:12,294 - octoprint.plugins.flashforge - DEBUG - FlashForge.readraw() called by thread: SD Uploader, timeout: 10000
2020-01-10 09:15:12,533 - octoprint.plugins.flashforge - DEBUG - FlashForge.readraw() CMD M28 Received. | Writing to file: 0:/user/surpri~1.gx  Length: 47279 | ok | 
2020-01-10 09:15:12,533 - octoprint.plugins.flashforge - DEBUG - FlashForge.sendcommand() got an ok
2020-01-10 09:15:12,533 - octoprint.plugins.flashforge - DEBUG - FlashForge.readraw() called by thread: SD Uploader, timeout: 1000
2020-01-10 09:15:12,537 - octoprint.plugins.flashforge - DEBUG - FlashForge.readraw() CMD M29 Received. | File save failed. | ok | 
2020-01-10 09:15:12,538 - octoprint.plugins.flashforge - DEBUG - Upload failed: File transfer incomplete
2020-01-10 09:15:12,548 - octoprint.plugins.flashforge - DEBUG - FlashForge.readline() called by thread comm._monitor
2020-01-10 09:15:12,549 - octoprint.events.fire - DEBUG - Firing event: TransferFailed (Payload: {'remote': u'surpri~1.gx', 'local': u'Surprising_Lappi-4.gx', 'time': 10})
2020-01-10 09:15:12,549 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x10556c310>>
2020-01-10 09:15:12,552 - octoprint.plugins.flashforge - DEBUG - FlashForge.readline() called by thread comm._monitor
2020-01-10 09:15:12,553 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2020-01-10 09:15:12,553 - octoprint.plugins.flashforge - DEBUG - FlashForge.readline() called by thread comm._monitor
2020-01-10 09:15:12,554 - octoprint.plugin - DEBUG - Calling on_event on announcements
`

Dmurph24, 2020-01-10:
I restarted and confirmed that flashforge is added here: 
[image]

Mrnt, 2020-01-10:
@Dmurph24 and after you select it you have to hit the "+" button then hit the "Save" button at the bottom of the dialog...

Mrnt, 2020-01-15:
@Dmurph24 I just made a new release that I am hoping will help a little. You will need to update a setting under Serial Connection > Firmware & Protocol >Protocol fine tuning: open up the "Advanced" section and set the value for "Hello" command to `M601 S0`

Dmurph24, 2020-01-17:
@Mrnt I updated to the new version and it worked without adding this protocol!

FruitBoi, 2020-02-20:
I'm also having this issue. I've followed the instructions, added the udev rule, etc.

Right as the print was about to begin, after heating, I got this error and octoprint was disconnected.

I'm only able to reconnect again once I reboot the printer. 

I've got a PowerSpec Ultra 3D

``Recv: MachineStatus: BUILDING_FROM_SD
Recv: MoveMode: HOMING
Recv: ok
Send: M119
Recv: CMD M27 Received.
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
Unexpected error while writing to serial port: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_TIMEOUT [-7])' @ comm.py:_do_send_without_checksum:3450
Connection closed, closing down monitor
Changing monitoring state from "Printing from SD" to "Offline (Error: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_TIMEOUT [-7])' @ comm.py:_do_send_without_checksum:3450)"``

Any help is appreciated.

Mrnt, 2020-02-20:
Hi there - a few questions to help me help you:

1. Confirm that you are printing by uploading directly to the internal SD card by using the "Upload to SD" button in OctoPrint (at the current time printing from within OctoPrint, ie hitting "Upload" then printing will not work due to timeout issues).
2. What type of file are you uploading .g/.gcode/gx/etc and what did you use as the slicer (PowerPrint/Cura/etc)?
3. Are you running OctoPrint on Windows/Mac/Linux or OctoPi on Raspberry Pi?
4. Can you enable debugging, try again, zip up your octoprint.log file and upload it here?

FruitBoi, 2020-02-20:
I was indeed using "Upload to SD", I was using Simplify3D as a slicer, exported to a .gcode file.
OctoPrint is running on Raspbian on a Pi. It can connect to the printer, and everything works as expected, except now and then, (especially after finishing a print) I'll get the above error and I won't able to reconnect until I reboot the printer. I'll upload my octoprint log when I get home.

Mrnt, 2020-02-21:
A couple of things to check -

- The GCODE scripts under Settings > GCODE Scripts: make sure that you don't have something set from another printer setup. Quite a few of the Marlin GCode commands are not recognized by FlashForge printers and actually seem to cause them to hang.
- Check the GCode commands at the end of the file that you printed for something that may be trying to reset the printer etc that has the same issue as above.

If you can provide the octoprint.log especially with debugging turned on for the flashforge plugin it might be possible to see what is happening.

FruitBoi, 2020-02-21:
I did remove the gcode scripts, and that helped a little. Here are my logs.
[octoprint (5).zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/4233738/octoprint.5.zip)

Mrnt, 2020-02-21:
May not be related, but I did notice:

```
2020-02-20 23:28:39,172 - octoprint.plugins.tracking - INFO - Sent tracking event system_throttled, payload: {'throttled_mask': 327685, 'throttled_voltage_now': True, 'throttled_voltage_past': True, 'throttled_overheat_now': False, 'throttled_now': True, 'throttled_past': True, 'throttled_overheat_past': False}
```
which presumably means the Pi is throttling because the PSU is having trouble delivering sufficient current.

Can you also upload your print file so I can take a look?

I actually have the same printer with I think the same  firmware, however I disabled the GPX plugin but I am also only (currently) testing with files generated by FlashForge's FlashPrint or PowerPrint.

FruitBoi, 2020-02-21:
Yeah, not sure why it's throttling. It has sufficient current from the PSU. The PSU is rated for 3A. It's an IPad Wall Charger and tried a couple of different cords. Anyways, here's the Gcode I sent to the printer.
[Can_Handle_V4.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/4233836/Can_Handle_V4.zip)

Mrnt, 2020-02-21:
Thanks I'll take a look.

re the power: sometimes the USB cable has enough resistance that you can get significant voltage drop on the cable when the Pi starts to draw current. I really don't understand why they used the crappy micro USB connector for power if the board can draw so much current - other SBC's use a proper barrel style connector for power.

FruitBoi, 2020-02-21:
I've been tinkering with it a bit, It's pretty hit or miss as to whether it works or not. I've found that the USB logo flashes for a second or two when connecting on OctoPrint, and then disappears. It only works when the logo on the LCD stays lit. As I said, it's really hit or miss.

FruitBoi, 2020-02-25:
Alright, I decided I'd try and use the official OctoPI image. I added the udev rule, installed pip, and installed the plugin, and reconfigured the steps provided in the readme, and it works now. No clue what I did wrong initially, glad it's working.

Mrnt, 2020-02-26:
Oh that's awesome. I was kind of stumped, so glad it's working!

Mrnt, 2020-05-04:
@Dmurph24 , @ammoniak , @FruitBoi - can I close this out? 

Note release 0.1.15 is actually detecting and using separate USB endpoints for SD uploads (rather than relying on hard coded endpoints). That may have previously caused some flakiness during SD upload and print.

closed 2020-08-24 by Mrnt