Mrnt/OctoPrint-FlashForge #8: [Request] Support turning the LED enclosure lights on/off, color changes

Mrnt, 2020-01-08:
Looks like most of the FlashForge made printers have LED illumination in the enclosure that can be controlled via USB to turn on/off and possibly brightness and color.

Requested by @twistedcustomdevel for a PowerSpec Ultra 3D

If anybody else wants this feature please chime in...

ammoniak, 2020-01-17:
I want that (and I already kind of use it). There are simple GCode commands, you can send from the terminal to set the color... A button to turn them on/off would be nice though.

boozecouncil, 2020-04-27:
This would be a cool addition. Lots of times I start a print to go overnight and the ability to turn off the lights when done would be nice. Or blink red if there's an error...

Mrnt, 2020-04-30:
@boozecouncil you can have the light change when a print job finishes or if cancelled I think by adding the g-code to custom g-code blocks in OctoPrint: Settings > GCODE Scripts and then add to the appropriate block eg:

`M146 r255 g0 b0`

could be added to the "After print job is cancelled" is block. 

But I agree it would be nice to have a light control in the "Control" tab.

twistedcustomdevel, 2020-04-30:
I am still interested in the feature and am willong to test when implemented.

Mrnt, 2020-05-13:
@twistedcustomdevel @ammoniak @boozecouncil I added a rudimentary control for turning the lights on, off and changing the color under the "Control" tab and added M146 to the default settings so that it is not auto uppercased by OctoPrint

twistedcustomdevel, 2020-07-09:
I just downloaded and updated my plugin this morning. I will test tonight for function.

Mrnt, 2020-08-24:
@twistedcustomdevel @ammoniak I'm going to close this. Please reopen if it is not working for you.

closed 2020-08-24 by Mrnt