Mrnt/OctoPrint-FlashForge #12: Doesn't work with Flashforge Ultra V2

kelkin, 2020-04-12:
This doesn't detect the Flashforge  / PowerSpec Ultra 2.0. Error is 

"Unexpected error while connecting to serial port: AUTO FlashForgeError: 'Found an unsupported FlashForge printer 'unknown' with USB ID: 0X00F6' @ comm.py:_openSerial:2691 (hook flashforge)"

Mrnt, 2020-04-16:
Thanks for reporting this - there are apparently many ID's for the PowerSpec Ultra! 

I just released a version that should recognize your printer. Can you test it out and if it works close the issue.

kelkin, 2020-04-19:
Yeah, I tried editing the config manually and got it to connect previously but wasn't successful in getting it to print. I downloaded your plugin and installed it on a fresh OctoPrint install. I got the error below, going to follow the instructions and try again.

Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: AUTO FlashForgeError: 'Unable to connect to FlashForge printer - permission error.

On OctoPi/Linux add the following line to

 /etc/udev/rules.d/99-octoprint.rules:

SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"

Then reboot your system for the rule to take effect.

' @ comm.py:_openSerial:2691 (hook flashforge)

Mrnt, 2020-04-19:
OK, so now it is recognizing the printer but not able to connect because of the default restrictions on USB access set by the OS, so you need to tell the OS to allow access to that specific USB device. Not sure what platform you are using but on linux/octopi you need to look for the file (or create one if it does not exist) eg:
`sudo nano /etc/udev/rules.d/99-octoprint.rules`

and add the following line:

`SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"`

Then save the file and close the editor. If you have to create the file then you might have to set the permissions to "rw-r--r--" using something like:
`sudo chmod 644 /etc/udev/rules.d/99-octoprint.rules`

Then restart and you should be able to connect.

kelkin, 2020-04-19:
Already did that, it connected when it rebooted. I tried to print a test file and it hung at 5%. Here's the full output:

Recv: MoveMode: READY
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:22 /0 T1:21 /0 B:12 /0
Recv: ok
Send: M119
Recv: CMD M119 Received.
Recv: Endstop: X-max: 0 Y-max: 0 Z-min: 0
Recv: MachineStatus: READY
Recv: MoveMode: READY
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:22 /0 T1:21 /0 B:12 /0
Recv: ok
Send: M119
Recv: CMD M119 Received.
Recv: Endstop: X-max: 0 Y-max: 0 Z-min: 0
Recv: MachineStatus: READY
Recv: MoveMode: READY
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:22 /0 T1:21 /0 B:12 /0
Recv: ok
Send: M119
Recv: CMD M119 Received.
Recv: Endstop: X-max: 0 Y-max: 0 Z-min: 0
Recv: MachineStatus: READY
Recv: MoveMode: READY
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:22 /0 T1:21 /0 B:12 /0
Recv: ok
Send: M119
Recv: CMD M119 Received.
Recv: Endstop: X-max: 0 Y-max: 0 Z-min: 0
Recv: MachineStatus: READY
Recv: MoveMode: READY
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:22 /0 T1:21 /0 B:12 /0
Recv: ok
Send: M119
Recv: CMD M119 Received.
Recv: Endstop: X-max: 0 Y-max: 0 Z-min: 0
Recv: MachineStatus: READY
Recv: MoveMode: READY
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:22 /0 T1:21 /0 B:12 /0
Recv: ok
Send: M119
Recv: CMD M119 Received.
Recv: Endstop: X-max: 0 Y-max: 0 Z-min: 0
Recv: MachineStatus: READY
Recv: MoveMode: READY
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:22 /0 T1:21 /0 B:13 /0
Recv: ok
Send: M119
Recv: CMD M119 Received.
[... 207 lines cut ...]
Send: G90
Recv: CMD M107 Received.
Recv: ok
Send: G28
Recv: CMD G90 Received.
Recv: ok
Send: M132 X Y Z A B
Recv: CMD G28 Received.
Recv: ok
Send: G1 Z50.00 F400
Recv: CMD M132 Received.
Recv: CMD G1 Received.
Recv: ok
Send: M119
Recv: ok
Send: M105
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: G161 X Y F3300
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Printing" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor

Mrnt, 2020-04-19:
What are you using to generate the gcode file? 
BTW, right now the plugin really only supports uploading a file directly to SD card preferably generated by FlashPrint. However, I'm pretty close to supporting printing files directly from Octoprint.

kelkin, 2020-04-19:
Oh! Sorry, didn't know that.  I'm using PowerPrint / Flashprint to slice.

kelkin, 2020-04-19:
I uploaded directly to the SD card a sample file it and it print, but it seems to hang and only send a few hundred bytes for the first few minutes. But it did print.
Thanks!

Mrnt, 2020-08-28:
@kelkin You may want to give the latest build a try - while not working on all printers, the plugin should now allow files to print by directly from OctoPrint by clicking the "Upload" button to load the print into OctoPrint then selecting and printing the file from the "Files" section of the OctoPrint screen. This does work on the PowerSpec Ultra I have but admittedly it is not the exact same version of the printer.

closed 2021-12-06 by Mrnt