Mrnt/OctoPrint-FlashForge #13: FlashForge Finder reports as an "Unsupported FlashForge printer"

Spillmaker, 2020-04-16:
```
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: AUTO FlashForgeError: 'Found an unsupported FlashForge printer 'unknown' with USB ID: 0X00EE' @ comm.py:_openSerial:2661 (hook flashforge)
```

Machine Type: Flashforge Finder
Firmware Version: 2.2.7.299 F2.12.2 20181203
Board Version: 2.x

Mrnt, 2020-04-16:
Thanks @Spillmaker I'll add it. There seems to be a few ID's for the Finder!

Mrnt, 2020-04-16:
I just published a new release which should support your printer. Please check and close this issue if it works.

Mrnt, 2020-04-20:
You will also need to tell the OS to allow access to that specific USB device. Not sure what platform you are using but on linux/octopi you need to look for the file (or create one if it does not exist) eg:
sudo nano /etc/udev/rules.d/99-octoprint.rules

and add the following line:

SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"

Then save the file and close the editor. If you have to create the file then you might have to set the permissions to "rw-r--r--" using something like:
sudo chmod 644 /etc/udev/rules.d/99-octoprint.rules

Then restart the OS and you should be able to connect.

Spillmaker, 2020-04-24:
Thanks! I've updated and added the line to the rules-file, and restarted. But now i get another more subtle errormessage:

```
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0x6c7247f0>, starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 115200
Unexpected error while writing to serial port: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_IO [-1])' @ comm.py:_do_send_without_checksum:3490
Changing monitoring state from "Detecting baudrate" to "Offline (Error: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_IO [-1])' @ comm.py:_do_send_without_checksum:3490)"
Connection closed, closing down monitor
Closing down send loop
```

This is Octopi on a Raspberry 2b

Mrnt, 2020-04-24:
Can you enable debugging for the plugin:

- Turn on debug messages for the plugin (Settings > Logging, under "Logging Levels" set octoprint.plugins.flashforge ([pic](https://github.com/Mrnt/OctoPrint-FlashForge/wiki/images/LoggingSettings.png)) to "DEBUG" and then click "Save") to help troubleshoot connection issues.
- After attempting to connect to the printer with debug messages turned on, review the log (Settings > Logging, octoprint.log) for clues.

And you can zip up the log file and paste upload it here.

Mrnt, 2020-05-04:
@Spillmaker I just finished working through a similar issue with another Finder v2 user and it turns out I needed to detect the USB connection endpoints (they were previously hard coded). 

Please try release v0.1.15 and let me know if that addresses the issue!

Spillmaker, 2020-05-05:
Connection is now successfull and i can (somewhat) control the printer. (Can only move motors once before i have to do some other movement, but i think that might me more of a octoprint issue, and not related directly to this issue) Now to find proper cura-settings to do a testprint from cura

closed 2020-05-05 by Spillmaker

Mrnt, 2020-05-05:
Great thanks for the update! You may have to home the axes via the Octoprint interface before you can move them.