Mrnt/OctoPrint-FlashForge #16: PowerSpec Ultra 2.0 Support?

bossrhino, 2020-04-21:
Will OctoPrint support the PowerSpec Ultra 2.0 printer?

https://www.microcenter.com/product/504453/ultra-20-dual-extruder-3d-printer

Mrnt, 2020-04-21:
I have one of these exact printers and use the plugin to upload files to it via OctoPrint.

With this plugin you should be able to use OctoPrint/OctoPi with the PowerSpec Ultra 2.0 (though we have discovered there are a few different releases of this printer). Right now you will have to slice the design using FlashPrint or the PowerSpec version of FlashPrint and then upload the resulting gcode file to the SD card via OctoPrint (rather than load the gcode file into OctoPrint and be able to view it printing line by line). However, I think I'm close to having OctoPrint be able to support printing line by line/direct print.

Does that answer your question?

bossrhino, 2020-04-24:
Yes it does.  So I have to use PowerSpec?  I can generate GCode and prints from Simplify3D.  Is that not supported?

Mrnt, 2020-04-24:
I have tried Cura which unfortunately adds some g-code at the beginning of its files that causes an error on these printers. 

You might be fine generating the file on Simlify3D as long as you upload it directly to the printer SD card rather than printing from OctoPrint (as mentioned above). 

If it doesn't work, paste the first 100 lines or so of the Simplify3D file in here so I can take a look - either way please let me know!

Mrnt, 2021-12-06:
Closing since PowerSpec Ultra 2.0 is supported via following workflows:

1. slice file on FlashPrint using FlashForge Dreamer settings
2. export file
3. upload file to OctoPrint using "Upload" button or upload to the printer's SD card via OctoPrint using  the "Upload to SD" button.

OR

1. configure custom printer settings on Cura for PowerSpec Ultra
2. slice file on Cura
3. export file
4. upload file to OctoPrint using "Upload" button or upload to the printer's SD card via OctoPrint using  the "Upload to SD" button.

OR

1. configure custom printer settings on Cura for PowerSpec Ultra and make it a network printer supported by OctoPrint
2. slice file on Cura
3. upload the file to the printer (OctoPrint) directly from Cura

closed 2021-12-06 by Mrnt