Mrnt/OctoPrint-FlashForge #17: Unable to communicate with Flashforge Finder

boozecouncil, 2020-04-26:
Hi -
I've installed OctoPrint on a Pi 3B, installed the OctoPrint-FlashForge plugin and configured it per the instructions but can't connect to my printer, a Flashforge Finder.

I get the following error:
```
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0x6a911c50>, starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 115200
Unexpected error while writing to serial port: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_IO [-1])' @ comm.py:_do_send_without_checksum:3490
Changing monitoring state from "Detecting baudrate" to "Offline (Error: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_IO [-1])' @ comm.py:_do_send_without_checksum:3490)"
Closing down send loop
Connection closed, closing down monitor
```
Initially I got the typical error saying to add a line to /etc/udev/rules.d/99-octoprint.rules which does not exist, but I added it to /etc/udev/rules.d/99-com.rules and rebooted, still get the same error.
I've commented it out in 99-com.rules and created the new /etc/udev/rules.d/99-octoprint.rules file, pasted text, saved, chmoded and rebooted, still get the same error.

I've updated all the packages on the Raspberry Pi, installed pip (both for python 2 & python 3) and rebooted, still get the same error.

dmesg shows that the printer is visible and usb works.

I've connected it to my windows laptop with the same USB cable and Flashprint connects fine and will easily move the extruder & bed.

The printer says its firmware is up to date.
[image]

octoprint.log file: https://pastebin.com/trLLTXHw
serial.log file: https://pastebin.com/pk8a67At

What am I missing here?

Thanks for your help and for putting this much effort into the plugin!

boozecouncil, 2020-04-26:
(title changed to hopefully help someone else find this solution)

Mrnt, 2020-04-26:
Sorry to hear about that'd thanks for supplying all the detail above! Odd that you have a Finder with a higher USB ID than mine with apparently a lower firmware version but dated more recently (mine reports USB ID 0x0002 but firmware V1.5 20170419), but there seems to be a lot of different versions of each FlashForge printer.
Can you do the following:

- Verify that you have the latest FlashPrint (what version?) and the Finder is updated to the latest firmware (I know you mentioned that - just want to confirm)
- Confirm you rebooted after installing the plugin and creating the dev rule
- Turn on debug logging for the plugin, create a fresh octopi.log (you may have to restart octopi) and then upload it here as a zip (or pastbin like you did already) - see [Troubleshooting](https://github.com/Mrnt/OctoPrint-FlashForge#troubleshooting) for how to enable the logging.

Mrnt, 2020-04-27:
Another thought - 
- was this a clean install of OctoPi?
- did you change any of the settings in:
  - OctoPrint Settings/Serial Connection/General
  - OctoPrint Settings/Serial Connection/Intervals & timeouts
  - OctoPrint Settings/Serial Connection/Firmware & protocol
  - OctoPrint Settings/Serial Connection/Behaviour

boozecouncil, 2020-04-27:
Thanks for the quick reply. Every time I tried to upgrade the firmware from the control panel it would fail with an error, which the internets told me was the printer's way of saying the firmware was up to date.
[image]

Turns out this is **wrong**. I contacted Flashforge about the printer firmware and they responded overnight with a new file. Flashed it and apparently this is a Finder Plus?
[image]
[image]

Now I have firmware version 2.7.2.366 F18 20181224
[image]

And to answer your questions above:

> Verify that you have the latest FlashPrint (what version?) and the Finder is updated to the latest firmware (I know you mentioned that - just want to confirm)

I am using the latest version of FlashPrint, 64 bit 4.2.0 both on OS X & Windows 10.

> Confirm you rebooted after installing the plugin and creating the dev rule

Confirmed, tried it multiple times (typical IT fix).

>Turn on debug logging for the plugin, create a fresh octopi.log (you may have to restart octopi) and then upload it here as a zip (or pastbin like you did already) - see Troubleshooting for how to enable the logging.

Done.
Serial.log: https://pastebin.com/abGAFdL9
Octoprint.log: https://pastebin.com/3cN3tMxs

>was this a clean install of OctoPi?

Yes, tried multiple times in case my microSD card was duff.

>did you change any of the settings in:
>OctoPrint Settings/Serial Connection/General
>OctoPrint Settings/Serial Connection/Intervals & timeouts
>OctoPrint Settings/Serial Connection/Firmware & protocol
>OctoPrint Settings/Serial Connection/Behaviour

Initially I changed all the settings per the plugin page, then tried a fresh install of OctoPi without changing anything, then again with changing the values. So it has been tried with the vanilla settings and also with the recommended settings on a clean install.

Thanks again!

Mrnt, 2020-04-28:
Thanks for all the detailed feedback - its helpful it rather looks like the length of time (about 5s) after opening the connection to the printer to actually writing something may be the problem, though I don't know yet what is causing that delay.

Another question - are you using a Raspberry camera or is there anything using USB besides the printer?

boozecouncil, 2020-04-28:
It is the Raspberry Pi camera attached to the flat ribbon cable connector. There is nothing else attached to the USB port of the Raspberry Pi. I've tried it with a thumbdrive connected to the printer and without, but still get the same error both ways.

boozecouncil, 2020-04-28:
[image]

boozecouncil, 2020-04-28:
Also a small explanation of this model from Flashforge support.

[image]

Mrnt, 2020-04-28:
OK great, I'll add a camera to the mix.

Got to love how flexible they seem to be with the model names and firmware.

boozecouncil, 2020-04-28:
> OK great, I'll add a camera to the mix.

Now that I think about it, I initially tried without the camera attached but got the same error.

> Got to love how flexible they seem to be with the model names and firmware.

“love” is one reaction (sarcasm noted)... seems like it would make it hard to support well.

Mrnt, 2020-04-30:
After delving into the USB protocol a little further, I have added some more debug messages that hopefully will help. Are you able to copy these two files over to the octoprint_flashforge folder to replace the current ones?
[octoprint_flashforge.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/4559020/octoprint_flashforge.zip)
If so can you clear out the octoprint.log, restart OctoPrint and try to make a connection and then upload the log as before?
Thanks for your help!

boozecouncil, 2020-04-30:
> If so can you clear out the octoprint.log, restart OctoPrint and try to make a connection and then upload the log as before?

Sure, will unzip, sftp it up and upload the logs momentarily.

boozecouncil, 2020-04-30:
Well I'll be damned. It connected!

[image]

Logs as requested.
serial.log: https://pastebin.com/X0dm78XR
octoprint.log: https://pastebin.com/21XK411B

boozecouncil, 2020-04-30:
Hmm, but doesn't seem to want to reconnect...

boozecouncil, 2020-04-30:
Yep, doesn't want to reconnect now. This was after a fresh reboot of everything (pi, server, printer). But if it connected once, no reason it shouldn't connect again...

serial.log https://pastebin.com/4CzJJrzy
octoprint.log https://pastebin.com/LD2FMVSd

Mrnt, 2020-04-30:
Can you make sure the Baudrate is set to Auto before trying to reconnect? 

Looking at the logs it appears OctoPrint is trying to read from the USB when it hasn't even sent anything yet. Not sure why it is doing that at the moment, but the communication requires that the host sends a hello to the printer before the printer will send anything back.

boozecouncil, 2020-04-30:
OK - noticed the baud rate was being left at 115200 after disconnecting the first time. Now if I set it to AUTO it reconnects without issue!

I don't know what you added to the two python files (haven't yet diff'd, don't know python in any case) but it worked!

serial.log https://pastebin.com/kGGWnwav
octoprint.log https://pastebin.com/VqZ4cQAn

Mrnt, 2020-04-30:
Awesome! Still need to figure out why it isn't connecting when the baud rate is other than Auto, but at least you can make some progress.

re the changes - the USB spec allows the device to indicate what type of functionality it supports (mass storage, audio, etc) and the endpoint (in and out) addresses to use for each service. Needless to say the Class, Subclass fields which are supposed to indicate the kind of device/service (and hence the purpose of each set of endpoint(s)) are set to 0xFF or 0x00... However, my mistake was to hard code the endpoint addresses and evidently on Finder V2 they decided to change the addresses. Now it will look for the first pair of in and out endpoints suitable for bulk transfer and (in the absence of any other useful information) will use them.

Thanks for helping me solve this - I think it may also be a problem for some of the other newer models.

boozecouncil, 2020-04-30:
While the connection issues have been solved, now I find I can't print.

I've got a .gx file sliced in Flashprint, when I attempt uploading to SD I get errors streaming to the SD. When I attempt to upload them print I get errors as well. Going to clear logs, restart the server and post logs for each.

Should I start a new thread for this, considering the original problem has been fixed?

Mrnt, 2020-05-01:
I think this may be related - it is possible they are using the other USB endpoints for uploading to the SD card even though the older printers use the same ones for communication and uploading.

Can you post your logs for the above and if possible try the following to provide additional data points:
- uploading a super simple model as gx file to the SD
- saving the (super simple) print as a ".g" file and upload to SD

and then upload the log. 

I still don't understand why the connection is not initiated correctly when you do not have "Auto" set for the Baudrate since it works fine on my end.

BTW if you are familiar with Wireshark and can easily set up USB sniffing (prob on Windows since Apple decided to break it on OS X) and look at how FlashPrint uploads to SD via USB you might be able to see if they are using a different endpoint from the main communication.

boozecouncil, 2020-05-01:
Flashprint has only ever output .gx files for me - do you mean rename it with “.g” extension?

I’m game for everything else (tomorrow) including some Wireshark USB packet sniffing.

Mrnt, 2020-05-01:
There should be a dropdown when you go to save (uh don't ask about the file name):
[image]

> I’m game for everything else (tomorrow) including some Wireshark USB packet sniffing.

Awesome! I haven't done it under Windows (so I'm not sure I can help you set it up) but it worked pretty well under OS X before they made it a real pain to enable it.

Mrnt, 2020-05-01:
I went ahead and released the changes you tested already because they are possibly relevant to other machines.

boozecouncil, 2020-05-01:
Failed upload to SD of simple print with .gx file, octoprint.log: https://pastebin.com/EjgWqjvC
Failed upload to SD of simple print with .g file, octoprint.log: https://pastebin.com/8pk7rwUV

Gonna have a go with Wireshark / USBPcap now.

Mrnt, 2020-05-01:
It looks like it uploaded the file but never got a response to the M29 command which is what is sent to signal to the printer that the upload is complete:
```
2020-05-01 15:16:34,284 - octoprint.plugins.flashforge - DEBUG - Sent: 100.00% 131553/131553
2020-05-01 15:16:34,286 - octoprint.plugins.flashforge - DEBUG - FlashForge.sendcommand() M29
2020-05-01 15:16:34,288 - octoprint.plugins.flashforge - DEBUG - FlashForge.writeraw() called by thread SD Uploader
2020-05-01 15:16:34,291 - octoprint.plugins.flashforge - DEBUG - FlashForge.readraw() called by thread: SD Uploader, timeout: 10000
2020-05-01 15:16:35,630 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:M105, cmd:M105
...
2020-05-01 15:16:44,293 - octoprint.plugins.flashforge - DEBUG - FlashForge.readraw() error: LIBUSB_ERROR_TIMEOUT [-7]
```

Whereas you should see something like:

```
2020-05-01 08:16:39,357 - octoprint.plugins.flashforge - DEBUG - Sent: 100.00% 363786/363786
2020-05-01 08:16:39,357 - octoprint.plugins.flashforge - DEBUG - FlashForge.sendcommand() M29
2020-05-01 08:16:39,358 - octoprint.plugins.flashforge - DEBUG - FlashForge.writeraw() called by thread SD Uploader
2020-05-01 08:16:39,359 - octoprint.plugins.flashforge - DEBUG - FlashForge.readraw() called by thread: SD Uploader, timeout: 10000
2020-05-01 08:16:39,860 - octoprint.plugins.flashforge - DEBUG - FlashForge.readraw() CMD M29 Received. | Done saving file. | ok | 
2020-05-01 08:16:39,860 - octoprint.plugins.flashforge - DEBUG - FlashForge.sendcommand() got an ok
...
```

Because the plugin never receives a response to the M29 (file upload finished) command it timeouts the read request and then the timeout is probably resulting in OctoPrint disconnecting from the printer. So it may be the printer just sat there expecting something different while the file was uploaded (for example to a different USB endpoint) or there is something about the actual uploading handshaking that it does not like (though it was happy enough at the beginning when the M28 command was sent to start the upload). If you can capture a FlashPrint upload with Wireshark that would be helpful - the printers I have tested send `M29 Received. \r\n Done saving file. \r\n ok` as a response when the file upload has finished.

boozecouncil, 2020-05-01:
Finally got Wireshark & USBPcapure to play nicely. Here I connected to the printer in Flashprint, uploaded a simple file, let it get going then canceled the print and disconnected.

[flashforge.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/4565022/flashforge.zip)

boozecouncil, 2020-05-01:
> Awesome! I haven't done it under Windows (so I'm not sure I can help you set it up) but it worked pretty well under OS X before they made it a real pain to enable it.

It was pretty easy under Windows 10 if you do things the normal way instead of trying to be slick.

Just download the 32 or 64 bit installer and select to install USBPcap during installation. Don't misinterpret the instructions to mean you need to install USBPcap separately, then try to install the portable version of Wireshark (dot org not dot com). That dog won't hunt.

Uninstall all that mess, then grab the 64 bit full installer, let it install everything, reboot and you're rocking like Dokken.

Mrnt, 2020-05-01:
Thats useful to know! There are definitely more resources available for Windows for stuff like this but unfortunately I gave up on Windows several years ago - my last straw was a HP all in one touch screen desktop as an "easy to use family machine" which was really painful having used tablet devices. 

Thanks for the pcap!

boozecouncil, 2020-05-01:
Well truth be told I still have my 12-year-old MacBook that I typically use even though it is rapidly failing, but since I'm fortunate to be working from home because of (_gestures at everything_) I'm using my work laptop, which is Windows.

Mrnt, 2020-05-01:
OK, I think it is indeed using a different endpoint for the upload:

[image]

0x04 instead of 0x01. 

I have some things to attend to but I will upload a new version here for you to try later today.

Mrnt, 2020-05-01:
Yes the current "situation" requires much patience as one provides in home teaching service [_unproductive rant about the state of education removed_], etc.

I like that under the covers the MacBook is much like Linux, so using command line tools is pretty painless while moving between embedded and development platforms. And also the UI generally seems more intuitive though in recent versions it does seems like they are trying to make configuring system preferences as obtuse as Windows. Oh and there is not so much of the vendor finger pointing eg why my Dell business grade laptop does not sleep properly - Dell: it's the OS, Microsoft: you need to talk to Dell, install the right drivers, etc

boozecouncil, 2020-05-01:
Wholeheartedly agreed that OS X is much nicer now that it's a skinned BSD-ish system.

I generally budget for a minimum of 2 days after receiving a new-to-me or reimaged Windows machine to fix all the driver issues, missing programs, utilities, etc... the driver mismatches or incompatibilities are the worst, though. My HP business grade laptop didn't have functioning keys to increase or decrease screen brightness until I went into the device manager and uninstalled the keyboard(!) which forced a reboot, when the correct driver was automagically installed.

Mrnt, 2020-05-02:
Can you give the following a whirl. It now detects the second set of endpoints and tries to use them for uploading to the SD card
[octoprint_flashforge2.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/4567916/octoprint_flashforge2.zip)

boozecouncil, 2020-05-04:
> Can you give the following a whirl. It now detects the second set of endpoints and tries to use them for uploading to the SD card
> [octoprint_flashforge2.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/4567916/octoprint_flashforge2.zip)
Uploaded, this time remembered to mv the old files to new names so I wouldn't overwrite them, restarted Octoprint and it seems to be working, I streamed a simple file to the SD card and it's started printing.
[image]

Webcam needs a better mount, it's just laying on the bed. Longer cables on the way from Amazon.
[image]

Temp is tracking well.
[image]

I think this is a winner winner chicken dinner! Thanks Mrnt.

boozecouncil, 2020-05-04:
[image]
Teeny tiny lil keychain cube printed successfully! Closing this one out. Thanks again @Mrnt !

closed 2020-05-04 by boozecouncil