Mrnt/OctoPrint-FlashForge #18: Feature request: Flashforge Adventurer 3 Support

lhassaine, 2020-04-30:
Great plugin! Do you think the Flashforge Adventurer 3 would work?

Mrnt, 2020-04-30:
I think it might - the product ID will need to be added to the code in order for it to connect. 

If you install the plugin and try to connect, it will attempt to find FlashForge printers and indicate the product ID in the OctoPrint Terminal tab if it does not recognize the printer as supported. If you feel comfortable editing the code you can add it to the list here: https://github.com/Mrnt/OctoPrint-FlashForge/blob/e781b2823342f4edb8bcd59d0e107a86b01993ff/octoprint_flashforge/__init__.py#L20
If it works you can post the ID here and I will add it to the next release. 

If you are not comfortable editing the code I can give you a version to test with if you provide the ID.

Related to this, I'm working to resolve an issue with another model that may play into this one, so if you go ahead and add the ID to the code but still can't connect then let me know anyway.

Da-Stu13, 2020-05-08:
Hi,
Just wondering where would I find the product ID and/or where did you find the other ones.
Thanks

Mrnt, 2020-05-08:
The easiest way to find the IDs if you are using OctoPrint and have installed this plugin:

1. connect the printer to the computer/Raspberry Pi running OctoPrint
2. switch to the "Terminal" tab in OctoPrint
3. make sure the "Serial Port" and "Baudrate" in OctoPrint are set to "Auto"
4. click "Connect" in OctoPrint and you should see some messages appear in the Terminal window, including the vendor and device ID's

Otherwise, if you have the printer hooked up to a Mac: 

1. go to the Apple icon in the menu bar and select "About This Mac"
2. select "System Report..."
3. then look under Hardware section, USB and in the right side of the screen there will be a list of USB devices including the details for each
[image]

It has been a while since I used Windows, so not sure how to do it on there.

If you can find them, then put them in here and I'll add them to the plugin.

Da-Stu13, 2020-05-08:
Hi,
The Adventurer 3 only has a power port, ethernet and USB. Would it still work?
Thanks

Mrnt, 2020-05-08:
So you would run OctoPrint on a RaspberryPi or old PC and connect it to the Adventurer via the USB port and the Raspberry Pi/ PC would be connected to the ethernet/wifi. You would then control the printer using the OctoPrint web interface running on the RaspberryPi/PC.

I'm not too familiar with the Adventurer, but since it has ethernet already I believe that you can already control it remotely using the Polar Cloud but I cannot speak to the pros and cons of using that over Octoprint - how are you looking to use OctoPrint with the Adventurer?

wolfbiteaus, 2020-05-12:
Hi, I might be reading this wrong, but as I read it, its like connecting to a usb printer which the FlashForge Adventure is not. only ethernet, wifi, & usb stick (which never seems to work)
But saying that usb 2 ethernet works, is that relavent to this addon or not?
I've added a dump of the usb2eth connection below
I apologise if this is not relevant.
If it is relevent I can do further testing if required for you.

May 13 05:33:25 wolfbite-1000H avahi-daemon[643]: Leaving mDNS multicast group on interface enx00e04c03296d.IPv6 with address fe80::957f:b2e4:4dcc:f953.
May 13 05:33:25 wolfbite-1000H dhclient[1378]: receive_packet failed on enx00e04c03296d: Network is down
May 13 05:33:25 wolfbite-1000H avahi-daemon[643]: Withdrawing address record for fe80::957f:b2e4:4dcc:f953 on enx00e04c03296d.
May 13 05:33:25 wolfbite-1000H NetworkManager[601]: <info>  [1589319205.2390] device (enx00e04c03296d): state change: ip-config -> unmanaged (reason 'removed', sys-iface-state: 'removed')
May 13 05:33:25 wolfbite-1000H upowerd[970]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2/2-2:1.0
May 13 05:33:25 wolfbite-1000H NetworkManager[601]: <info>  [1589319205.2722] dhcp4 (enx00e04c03296d): canceled DHCP transaction, DHCP client pid 1378
May 13 05:33:25 wolfbite-1000H NetworkManager[601]: <info>  [1589319205.2724] dhcp4 (enx00e04c03296d): state changed unknown -> done
May 13 05:33:25 wolfbite-1000H NetworkManager[601]: <info>  [1589319205.2880] manager: NetworkManager state is now DISCONNECTED
May 13 05:33:25 wolfbite-1000H upowerd[970]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2
May 13 05:33:25 wolfbite-1000H NetworkManager[601]: <info>  [1589319205.4176] devices removed (path: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2/2-2:1.0/net/enx00e04c03296d, iface: enx00e04c03296d)
May 13 05:34:02 wolfbite-1000H kernel: [  783.916277] usb 2-2: new full-speed USB device number 3 using uhci_hcd
May 13 05:34:02 wolfbite-1000H kernel: [  784.117387] usb 2-2: New USB device found, idVendor=0bda, idProduct=8150
May 13 05:34:02 wolfbite-1000H kernel: [  784.117398] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 13 05:34:02 wolfbite-1000H kernel: [  784.117405] usb 2-2: Product: USB 10/100 LAN
May 13 05:34:02 wolfbite-1000H kernel: [  784.117411] usb 2-2: Manufacturer: REALTEK
May 13 05:34:02 wolfbite-1000H kernel: [  784.117417] usb 2-2: SerialNumber: 0605
May 13 05:34:02 wolfbite-1000H kernel: [  784.131255] rtl8150 2-2:1.0: eth0: rtl8150 is detected
May 13 05:34:02 wolfbite-1000H mtp-probe: checking bus 2, device 3: "/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2"
May 13 05:34:02 wolfbite-1000H NetworkManager[601]: <info>  [1589319242.4790] manager: (eth0): new Ethernet device (/org/freedesktop/NetworkManager/Devices/4)
May 13 05:34:02 wolfbite-1000H mtp-probe: bus: 2, device: 3 was not an MTP device
May 13 05:34:02 wolfbite-1000H systemd-udevd[1395]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
May 13 05:34:02 wolfbite-1000H kernel: [  784.237891] rtl8150 2-2:1.0 enx00e04c03296d: renamed from eth0
May 13 05:34:02 wolfbite-1000H NetworkManager[601]: <info>  [1589319242.6402] devices added (path: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2/2-2:1.0/net/enx00e04c03296d, iface: enx00e04c03296d)
May 13 05:34:02 wolfbite-1000H NetworkManager[601]: <info>  [1589319242.6467] device added (path: /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2/2-2:1.0/net/enx00e04c03296d, iface: enx00e04c03296d): no ifupdown configuration found.
May 13 05:34:02 wolfbite-1000H NetworkManager[601]: <info>  [1589319242.6485] device (enx00e04c03296d): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
May 13 05:34:02 wolfbite-1000H kernel: [  784.332713] IPv6: ADDRCONF(NETDEV_UP): enx00e04c03296d: link is not ready
May 13 05:34:02 wolfbite-1000H NetworkManager[601]: <info>  [1589319242.6659] device (enx00e04c03296d): carrier: link connected
May 13 05:34:02 wolfbite-1000H NetworkManager[601]: <info>  [1589319242.6960] keyfile: add connection in-memory (4de768e3-5232-31a4-8a4a-5cdf0472f5f2,"Wired connection 2")
May 13 05:34:02 wolfbite-1000H upowerd[970]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2/2-2:1.0
May 13 05:34:02 wolfbite-1000H upowerd[970]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2
May 13 05:34:02 wolfbite-1000H NetworkManager[601]: <info>  [1589319242.7187] settings: (enx00e04c03296d): created default wired connection 'Wired connection 2'
May 13 05:34:03 wolfbite-1000H NetworkManager[601]: <info>  [1589319243.7642] device (enx00e04c03296d): carrier: link connected
May 13 05:34:03 wolfbite-1000H NetworkManager[601]: <info>  [1589319243.7707] device (enx00e04c03296d): state change: unavailable -> disconnected (reason 'carrier-changed', sys-iface-state: 'managed')
May 13 05:34:03 wolfbite-1000H NetworkManager[601]: <info>  [1589319243.7779] policy: auto-activating connection 'Wired connection 2'
May 13 05:34:03 wolfbite-1000H NetworkManager[601]: <info>  [1589319243.7927] device (enx00e04c03296d): Activation: starting connection 'Wired connection 2' (4de768e3-5232-31a4-8a4a-5cdf0472f5f2)
May 13 05:34:03 wolfbite-1000H NetworkManager[601]: <info>  [1589319243.7951] device (enx00e04c03296d): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
[... cut ...]
May 13 05:34:05 wolfbite-1000H ModemManager[645]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-2': not supported by any plugin
May 13 05:34:06 wolfbite-1000H dhclient[1412]: DHCPDISCOVER on enx00e04c03296d to 255.255.255.255 port 67 interval 3 (xid=0xc91e1535)
May 13 05:34:09 wolfbite-1000H dhclient[1412]: DHCPDISCOVER on enx00e04c03296d to 255.255.255.255 port 67 interval 8 (xid=0xc91e1535)
May 13 05:34:17 wolfbite-1000H dhclient[1412]: DHCPDISCOVER on enx00e04c03296d to 255.255.255.255 port 67 interval 13 (xid=0xc91e1535)

Mrnt, 2020-05-14:
@lhassaine, @Da-Stu13 I had to realized the Adventurer does not support USB to host connections. Unfortunately right now the Adventurer will not work with this plugin as the plugin is using USB for the connection.

@wolfbiteaus You are correct - after looking at the spec I realize the Adventurer does not have a USB to host connection. The idea of a connection via USB to ethernet on the Adventurer is interesting but I am not sure how the Plugin would know to try and communicate via that USB port since it will presumably show up as a USB to Ethernet adapter (not as a FlashForge printer) when polling the USB ports. I am guessing it doe not currently show up as a port in OctoPrint?

wolfbiteaus, 2020-05-15:
I'm not a expert with this stuff (but know how to google :)
Found https://www.lucid-control.com/accessing-usb-io-module-network-linux-socat/
Is this something useful for you? or I log to octoprint? or something not relevent?

Side note gpx now updated :)

wolfbiteaus, 2020-05-15:
If not interested or not relevant to you, just say so & i'll shutup :)
But I tried a few things to see if I could work it/something out.
now seeing /dev/tty?? under auto where I wasnt before

sudo socat /dev/ttyS0,b115200,raw,echo=0 TCP:192.168.111.192:8899
2020/05/15 04:00:24 socat[2040] E tcgetattr(5, 0x7ede5060): Inappropriate ioctl for device

https://stackoverflow.com/questions/11634725/socat-virtual-serial-port-not-recognized
socat PTY,link=/dev/ttyS10 PTY,link=/dev/ttyS11
and link your programs to ttyS10 and ttyS11.

that worked but when accessing from octoprint get error
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: AUTO IOError: 'GPX plugin not able to discover AUTO port and/or baudrate. Please choose specific values for them.' @ comm.py:_openSerial:2691 (hook GPX)
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: /dev/ttyS11 OSError: '[Errno 13] Permission denied: '/dev/ttyS11'' @ comm.py:_openSerial:2691 (hook GPX)

pi@octopi:~ $ ls -la /dev/tty*
-rw-r--r-- 1 root root          0 May 15 04:00 /dev/ttyS0
lrwxrwxrwx 1 root root         10 May 15 04:03 /dev/ttyS10 -> /dev/pts/1
lrwxrwxrwx 1 root root         10 May 15 04:03 /dev/ttyS11 -> /dev/pts/2
-rw-r--r-- 1 root root          0 May 15 04:00 /dev/ttyS2

pi@octopi:~ $ ls -la /dev/pts/*
crw--w---- 1 pi   tty  136, 0 May 15 04:13 /dev/pts/0
crw--w---- 1 root tty  136, 1 May 15 04:03 /dev/pts/1
crw--w---- 1 root tty  136, 2 May 15 04:03 /dev/pts/2
c--------- 1 root root   5, 2 Feb 14  2019 /dev/pts/ptmx

pi@octopi:~ $ sudo chmod 777 /dev/pts/1
pi@octopi:~ $ sudo chmod 777 /dev/pts/2
pi@octopi:~ $ ls -la /dev/pts/*
crw--w---- 1 pi   tty  136, 0 May 15 04:15 /dev/pts/0
crwxrwxrwx 1 root tty  136, 1 May 15 04:03 /dev/pts/1
crwxrwxrwx 1 root tty  136, 2 May 15 04:03 /dev/pts/2
c--------- 1 root root   5, 2 Feb 14  2019 /dev/pts/ptmx

trying octoprint now gives 
Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: /dev/ttyS11 Timeout: 'Timeout' @ comm.py:_openSerial:2691 (hook GPX)

wolfbiteaus, 2020-05-15:
had a brain fart moment, google socat & octoprint.
Seem various comments & work arounds. but I think I'm too dumb for most of it.
close and ignore if need be, its starting to hurt my head :)

Side note: OctoPrint SHOULD note its ONLY for usb printers in the headings :(

Julzilla, 2020-10-03:
Any updates on this project? Willing to test if needed.

DaShoe, 2020-10-05:
We established that it is not possible until Octoprint release support over Ethernet. I have not checked recently so I’ll take a look😀

Mrnt, 2020-10-05:
I might be able to add it. 

Questions for you guys:

- My printer does not support connections using the hostname, ie http://myprinterhostname.local does not work - it has to be the IP address eg http://192.168.1.123   Does your printer connection by hostname or does it have to be IP address?
- Since it is not practical to scan the user's network for the printer, the user will have to enter the printer address manually. Any thoughts on how/where in OctoPrint the user would enter the network address of the printer?

DaShoe, 2020-10-06:
My adventurer 3 has currently got a fixed ip
Is 192.168.1.89:8899 as far as I know this is the only way to connect wirelessly without using a cloud service. Due to it having a fixed up you could possibly input the it in the initial setup.

Julzilla, 2020-10-06:
> I might be able to add it.
>
> Questions for you guys:
>
> * My printer does not support connections using the hostname, ie http://myprinterhostname.local does not work - it has to be the IP address eg [...]

It is possible to get samba access to the Adventurer 3 through CrashForge **https://github.com/pressreset/CrashForge-samba**

With this you would be able to edit '/etc/hostname' in a text editor on windows (Notepad++) and give the machine a hostname. I can try this for you and report back if you like?

A-Human-13, 2021-03-03:
Any new info? can do some testig if u need.

humor4fun, 2021-03-08:
I'm also interested in a solution here. Since the Adventurer 3 has a native interface over ethernet for receiving print jobs, perhaps there's a way to simply send octoprint jobs over to that web/api interface for the AD3? (such that the two devices are not actually connected except for being on the same network segment)

kfsone, 2021-09-15:
There's a cross-platform client application for uploading files to an Adventurer III/Voxel:

https://andybradford.dev/2020/01/12/using-the-monoprice-voxel-with-ultimaker-cura/
https://github.com/andycb/AdventurerClientDotNet

sohsmith, 2022-01-24:
There's a python library that works for sending prints to an Adventurer III/Voxel in the tdptk package (https://github.com/johndoe31415/tdptk.git). I'm currently using tdptk to send Cura generated jobs to my Monoprice Voxel, it claims to be able to convert gcode to gx complete with bitmap and metadata but I haven't bothered to try it, just producing files with .g and the right prefix/postfix gcode gets the print done. The interface module is a single file module (tdptk/FlashForgeProtocol.py) that looks to be straightforward to use and not even too horrible to reverse engineer a protocol spec from. The author might even have the spec.

brendann993, 2022-04-26:
I've had a quick look and it appears that the adventurer 4 (and likely the adventurer 3) use the same command syntax e.g. ~ in front of the command and responds in the same way as what this plugin does but instead of communicating via USB it does so over TCP e.g. a virtual serial port connected to TCP endpoint. I don't know enough about python to add the ability to use tcp as the connection.

Animizio, 2022-09-15:
hey guys, anything new to this?

patstave, 2023-02-09:
Also really interested in this

ravinder-Olivier, 2023-03-05:
> There's a cross-platform client application for uploading files to an Adventurer III/Voxel:
> 
> https://andybradford.dev/2020/01/12/using-the-monoprice-voxel-with-ultimaker-cura/ https://github.com/andycb/AdventurerClientDotNet

The adventurer client.net library could be piggy backed off of to use with octopi. I'll be testing this over the next few months.