Mrnt/OctoPrint-FlashForge #21: Communication timeout while printing

jumpingmushroom, 2020-05-05:
Hi! I just wanted to first thank you for your hard working in getting this plugin up and running. The fact that I actually can communicate with the printer at all with Octoprint now is just excellent :)

So, I've however been having some issues with the Flashforge finder and using octoprint with your plugin. This is the error I receive, and it happens pretty consistently on all prints after a short time.

> Send: M27
> Recv: CMD M27 Received.
> Recv: SD printing byte 28185/2807826
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 0 Y-max: 0 Z-max: 0
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: BUILDING_FROM_SD
> Recv: MoveMode: MOVING
> Recv: ok
> Send: M105
> Recv: CMD M105 Received.
> Recv: T0:210 /210 B:0 /0
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 0 Y-max: 0 Z-max: 0
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: BUILDING_FROM_SD
> Recv: MoveMode: MOVING
> Recv: ok
> Send: M27
> Recv: CMD M27 Received.
> Recv: SD printing byte 28240/2807826
> Recv: ok
> Send: M117 1% L=-/-
> Recv: CMD M117 Received.
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Send: M117 ETA: -
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Changing monitoring state from "Printing from SD" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
> Connection closed, closing down monitor

Is there anything I can tweak to maybe resolve this issue? I've gone through your README several times and made sure all settings are defined correctly, and done the troubleshooting, but I can't really figure out what's happening.
I noticed it says that it can happen if you have long moves, but this happens even on short move prints with small models.

Hope someone could provide some input on what's going on here. Thanks.

Mrnt, 2020-05-05:
I notice this `M117 1% L=-/-` is being sent to the printer which I think is supposed to be putting text onto the display, however I am not sure that will work. What software are you using to generate the g-code file?

jumpingmushroom, 2020-05-05:
I see. I don't think there's any way to send text to the display on the finder, so I agree. I'm using the official software from Flashforge called Flashprint to generate the GX file, which I upload using upload to SD.

Mrnt, 2020-05-05:
Generating files with FlashPrint and uploading them the SD card via OctoPrint with the plugin should work. So is FlashPrint adding those lines to the file?

jumpingmushroom, 2020-05-05:
I just opened the GX file and tried to find M117, but there's no reference to it there. I could upload the GX if that would help in figuring out what's going on?

Mrnt, 2020-05-05:
Sure, you should be able to upload it here as a zip file. What printer, firmware version (from the Terminal tab on Octoprint when you connect or from the "About" screen on the printer display), Flashprint version are you using?

Mrnt, 2020-05-05:
Also double check that  that when you create files using FlashPrint that you have "FlashForge Finder" selected: in FlashForge go to the menu > Print > Machine Type

jumpingmushroom, 2020-05-05:
Alright, so I've added the GX file as a zip file. This is the one that failed with the above mentioned message.
[Square_Xiaomi_Aquara_temperature.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/4581741/Square_Xiaomi_Aquara_temperature.zip)

The printer is a Flashforge Finder. This one: 
[image]
I know there's also a red type, but I'm thinking they're just different depending on region.

The firmware version is: V1.5 20170419
> Recv: CMD M115 Received.
> Recv: Machine Type: Flashforge Finder
> Recv: Machine Name: My Finder
> Recv: Firmware: V1.5 20170419
Flashprint is version is 4.2.0 64-bit

Machine type is set to the correct machine:
[image]

Hope this helps out a little. Let me know if you need me to do find any more info.

Mrnt, 2020-05-05:
Sorry, I'm not fully awake yet. Thinking about this further, when printing from the SD card you should only see the commands M27 (get print progress), M105 (get temps), M119 (get printer status) being sent to the printer. Have you added any other plugins? Is the GPX plugin active?

jumpingmushroom, 2020-05-05:
Well, good suggestion. So I tried to disable all plugins I've installed, and only left the required ones. I did not have the GPX plugin active as I did a clean install of octoprint after finding your plugin.

After disabling as many plugins as possible, and restarting the service I tried a new print. This is the entire log of that session:

> Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0x6f39bdf0>, starting monitor
> Starting baud rate detection...
> Changing monitoring state from "Opening serial port" to "Detecting baudrate"
> Trying baudrate: 115200
> Send: M601 S0
> Recv: CMD M601 Received.
> Recv: Control Success.
> Recv: ok
> Changing monitoring state from "Detecting baudrate" to "Operational"
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 1 Y-max: 1 Z-max: 1
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: READY
> Recv: MoveMode: READY
> Recv: ok
> Send: M27
> Recv: CMD M27 Received.
> Recv: Done printing file
> Recv: ok
> Send: M115
> Recv: CMD M115 Received.
> Recv: Machine Type: Flashforge Finder
> Recv: Machine Name: My Finder
> Recv: Firmware: V1.5 20170419
> Recv: SN: 69A68971
> Recv: X: 140  Y: 140  Z: 140
> Recv: Tool Count: 1
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 1 Y-max: 1 Z-max: 1
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: READY
> Recv: MoveMode: READY
> Recv: ok
> Send: M105
> Recv: CMD M105 Received.
> Recv: T0:210 /210 B:0 /0
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 1 Y-max: 1 Z-max: 1
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: READY
> Recv: MoveMode: READY
> Recv: ok
> Send: M105
> Recv: CMD M105 Received.
> Recv: T0:210 /210 B:0 /0
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 1 Y-max: 1 Z-max: 1
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: READY
> Recv: MoveMode: READY
> Recv: ok
> Send: M105
> Recv: CMD M105 Received.
> Recv: T0:210 /210 B:0 /0
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 1 Y-max: 1 Z-max: 1
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: READY
> Recv: MoveMode: READY
> Recv: ok
> Send: M105
> Recv: CMD M105 Received.
> Recv: T0:210 /210 B:0 /0
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 1 Y-max: 1 Z-max: 1
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: READY
> Recv: MoveMode: READY
> Recv: ok
> Send: M105
> Recv: CMD M105 Received.
> Recv: T0:206 /0 B:0 /0
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 1 Y-max: 1 Z-max: 1
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: READY
> Recv: MoveMode: READY
> Recv: ok
> Send: M23 0:/user/Outdoor_Aqara_Sensor_Case.gx
> Recv: CMD M23 Received.
> Recv: File opened: 0:/user/Outdoor_Aqara_Sensor_Case.gx Size: 3326779
> Recv: File selected
> Recv: ok
> Send: M119
> Recv: CMD M119 Received.
> Recv: Endstop: X-max: 1 Y-max: 1 Z-max: 0
> Recv: Status: S:0 L:0 J:0 F:1
> Recv: MachineStatus: BUILDING_FROM_SD
> Recv: MoveMode: HOMING
> Recv: ok
> Send: M27
> Recv: CMD M27 Received.
> Recv: SD printing byte 722/3312267
> Changing monitoring state from "Operational" to "Starting print from SD"
> Recv: ok
> Changing monitoring state from "Starting print from SD" to "Printing from SD"
> Send: M117 0% L=-/-
> Recv: CMD M117 Received.
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Send: M119
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
> Changing monitoring state from "Printing from SD" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
> Connection closed, closing down monitor

Seems the same issue appears again.

What could be sending CMD M117? I'm thinking if we find that, we might know what's happening. I'll dig through some of the plugins and code to see what I can find. But feel free to come with suggestions.

Mrnt, 2020-05-05:
Its puzzling - I took a quick look through the OctoPrint code itself and did not see anything that would send it. Did you restart OctoPrint after disabling all the plugins you installed?

jumpingmushroom, 2020-05-05:
Yes, disabled all the plugins and restarted the octoprint service completely. I found this https://github.com/FormerLurker/Octolapse/issues/289, but I'm not entierly sure it's related. I do run a timelapse at the same time, so on next print I'll disable that as well (currently completing the print I started ;) ).

jumpingmushroom, 2020-05-06:
Ok, so some reading later I found that this plugin [DisplayLayerProgress Plugin](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) seems to be the culprit. After disabling this (even though it advices against it due to it being needed to detect fan speed, layer height etc.), the connection continues and i never lose connection during a print. However now I can't see any data on the dashboard.

I'll check out the settings of this plugin to hopefully figure out if there's a way to disable M117 from being sent, while still being able to use the plugin.

Thanks for your help and hope this thread can help others as well if they stumble upon the same situation :)

Mrnt, 2020-05-06:
I think what I'll do is filter out any commands that the printer will not accept or adversely affect printing, that way I think these plugins will still work but the printer will just ignore them.
Would you be willing to test this for me? I can either point you to a separate branch or upload files here for you to test with - what would you like?

jumpingmushroom, 2020-05-07:
Yes of course I'll test it for you. I found out by the way that there's a big notice on the DisplayLayerProgress plugin that M117 might not work or even crash the connection, so yeah, it's safe to say that one is the culprit.

I did end up setting up octopi instead and started with just your plugin, and it worked great. I do however want to use these plugins that seem to do the whole M117 command, so testing your plugin with those would be interesting.

I installed your plugin through the plugin manager, so if you provide a file I can use for either the "from an uploaded archive" or the "from URL", that would make it easiest.

Mrnt, 2020-05-07:
OK made a new branch which you can try with DisplayLayerProgress plugin. You can install it using the following URL:
https://github.com/Mrnt/OctoPrint-FlashForge/archive/Dremel_and_DisplayLayer.zip
I think there are some other issues with DisplayLayerProgress - looking in the Octopi log when doing an SD upload it gave an error that I am not sure is related to my plugin.

jumpingmushroom, 2020-05-12:
I'm so sorry for not giving you an update on this, but I ended up moving my octoprint to OctoPI on a raspbery, so things took a little longer. I saw that you pushed the updated through the repo and I've updated it right away. Looks like it works great :) Now I can run DisplayLayerProgress and the dashboard without issues. Thanks for your great work!

closed 2020-05-12 by jumpingmushroom