Mrnt/OctoPrint-FlashForge #24: Register in plugin repository?

foosel, 2020-05-11:
Hey @Mrnt! Great to see something like this to finally come together! 

I'd love to see this getting registered on the official plugin repository at plugins.octoprint.org once you deem it stable enough. That way more people could find it easier. Or what do you think? :)

Mrnt, 2020-05-12:
Thanks @foosel, I am flattered for the invite! I think it is a great idea. I have been trying to get enough traction so that I have a few "beta" testers to make sure it is basically solid before finding out how to get registered. 

It seems fairly stable at this point - I guess one only hears when something is broken :). However, right now it only supports printing via SD and I didn't want to get a deluge of complaints about not being able to print line by line/layer by layer direct from OctoPrint (which I am currently trying to implement). Do you think it is useful enough as it is to add it to the official repository?

foosel, 2020-05-12:
> Do you think it is useful enough as it is to add it to the official repository?

I would say so, yes. I'd suggest to add a big fat disclaimer to the listing that it currently only supports SD upload and be prepared to close a bunch of tickets in that regard. Also make sure that it can update itself for when you push fixes and such. But other than that, considering that it's printing and one of those requests that have been going on for ages, I'd say it would be a good idea to register it ASAP and get more testers.

garyriet, 2020-05-12:
I loaded your plugin on two octopie connected to two dreamers and its working pretty good!
Thanks
Gary

Mrnt, 2020-05-16:
@foosel I think that I have the plugin ready to put in the repository but when I test the Github Pages locally following the setup [here](https://plugins.octoprint.org/help/registering/) I am seeing an error:
```
Liquid Exception: undefined method `[]' for nil:NilClass in /_layouts/plugin_list.html
jekyll 3.8.5 | Error:  undefined method `[]' for nil:NilClass
```
Which seems to have been introduced in [this](https://github.com/OctoPrint/plugins.octoprint.org/commit/b1b02a8ffc13761195e135253aca129d946550ca) commit.

foosel, 2020-05-18:
Yep, that's my fault, I'm looking into it 😅

foosel, 2020-05-18:
Fixed now, should be fine after a pull from origin.

Mrnt, 2020-05-18:
Thanks! I just created a pull request that hopefully is ok :smile:

closed 2020-05-19 by Mrnt