Mrnt/OctoPrint-FlashForge #25: Flashforge Creator 3 IDEX v2

lukedog5990, 2020-05-12:
Please support when you can. 

holding off buying the hardware for octo print till i have something that'll connect the creator 3

Mrnt, 2020-05-12:
If you already have the creator 3 can you provide the USB vendor ID (probably 0x2b71) and the USB Product ID (it will also be a 4 digit hex number like 0x1234). 

If you setup OctoPrint on your desktop computer you should be able to test the plugin with your printer and find the ID's that way.

Or if you have the printer hooked up to a Mac: 

1. go to the Apple icon in the menu bar and select "About This Mac"
2. select "System Report..."
3. then look under Hardware section, USB and in the right side of the screen there will be a list of USB devices including the details for each
[image]

It has been a while since I used Windows, so not sure how to do it on there.

If you can find them, then put them in here and I'll add them to the plugin. 

Thanks!

lukedog5990, 2020-05-12:
The creator 3 has no usb connection. Other than the usb port for thumb
drives.

It connects over WIFI or Ethernet port

Is there still a way to get product and vendor ID

Mrnt, 2020-05-14:
It is so different from the Creator Max! Yep you are right no USB slave port. It looks like right now it is in the same boat as the FF Adventurer and cannot be connected to OctoPrint.

Can you give describe how you would like to use the printer with OctoPrint (vs say Polar Cloud)?