Mrnt/OctoPrint-FlashForge #27: Flashforge Guider II

jayceekeys, 2020-05-14:
Hi,

Would it be possible to add the Flashforge Guider II. Also ... would this plugin expose data such as the printer status - printing / paused / complete / time remaining?

Many thanks,

Jay :)

Mrnt, 2020-05-14:
It looks like the Guider II supports direct USB connection so if you are able to provide the USB vendor ID and Product ID I can add it to the plugin. 

You should be able to find them by looking at the hardware details for the printer when it is connected via USB to your computer. The Vendor ID should be 0x2b71 and the Product ID would be a similar 4 digit number.

The plugin will allow you to see status and pause/continue/cancel upload to SD to print. Time remaining does work - sort of.

jayceekeys, 2020-05-15:
[image]

Hi Mrnt,

Thankyou so much - I REALLY appreciate all your hard work and for adding the Guider IIS.

Many thanks,

Jay :)

Mrnt, 2020-05-16:
Jay, thanks for that! I have added the ID to the latest release if you want to give it a try. Let me know how it works!

jayceekeys, 2020-05-17:
Hey buddy,

My Raspberry Pi 4 arrived today with the new sd cards !

I sorted the drive image and downloaded your plugin. I've checked all the settings you detail are correct, but am getting the following error ... is it normal and how do I get to the file to make the required changes it says?

Thanks Jay :)

Changing monitoring state from "Offline" to "Error: Connection error, see Terminal tab"
Unexpected error while connecting to serial port: AUTO FlashForgeError: '>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

Unable to connect to FlashForge printer - permission error.

If you are using OctoPi/Linux add permission to access this device by editing file:

 /etc/udev/rules.d/99-octoprint.rules

and adding the line:

SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"

You can do this as follows:

1) Connect to your OctoPi/Octoprint device using ssh

2) Type the following to open a text editor:

sudo nano /etc/udev/rules.d/99-octoprint.rules

3) Add the following line:

SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"

4) Save the file and close the editor

5) Verify the file permissions are set to "rw-r--r--" by typing:

ls /etc/udev/rules.d/99-octoprint.rules

6) Reboot your system for the rule to take effect.

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

' @ comm.py:_openSerial:2691 (hook flashforge)

garyriet, 2020-05-17:
You need to add that line to the file. I had to do the same thing

jayceekeys, 2020-05-17:
Hey Gary !

Thankyou so much for the info .... I’m genuinely not an idiot lol .... despite many years working in electronics I’ve never actually used a Rasberry Pi .... this is literally my first day (always used other microprocessors).

How do I edit the file - it sounds stupid but do I have to use SSH or can I just add a monitor and keyboard and mouse and get straight into the file system ?

Thankyou so much,

Jay :)

garyriet, 2020-05-17:
Jay,
No problem. I designed and ran super computers for thr army and didn’t want any Linux variant in my house. Now I got 7 pi’s lmao
Yea you can hook up a monitor or download putty to ssh in

Mrnt, 2020-05-17:
@jayceekeys I think it is because OctoPrint is not running as root (which is fine), and I guess by default the Pi restricts access to all USB devices so explicit permissions have to be added to allow access. I don't think  I can do much to resolve this step in the plugin eg by adding the permissions automatically because I don't think the plugin would have permission to edit the system config...  

If you are using Windows I think you will need to download and install putty to ssh into the pi. If you are using a Mac, there is already an ssh client installed that you access via the Terminal.

I guess I should add some instructions on using SSH to complete this step?

jayceekeys, 2020-05-17:
Hey Guys,

So I managed to get in and edit that file entry. It’s now connected to the printer !!!! 

Initial checks seemed to be that the control of lights and each axis was responding, however I think the up on the Z is actually going down and vice versa. The X and Y seem to work for a short movement but then don’t seem to respond after the first move.

One of the main reasons was to get some form of interface between Simplify3D and the Flashforge which I think with an extra plugin I can achieve? The other BIG thing is to read the printer status from my home automation system (HomeSeer). This would allow me to have Sonos announcements in the house for print complete - or even ask Alexa how long is left on my print (I’ve got a developer account with Amazon and have linked my home automation system to the Amazon devices. There’s a plug-in for HomeSeer that works with Octoprint and reads the printer status. Even if I can control the printer from Octoprint or even Simplify3D it would still be cool to be able to monitor the printer status from my home automation system. It could trigger events and voice announcements.

I started a short test print from the SD card on the printer direct from it’s screen to see what Octoprint could see. The temperature stats all seem to be logging and the terminal window shows machine status as “BUILDING_FROM_SD”, but the main state of the printer in the front stays as “Operational” is this normal ? There are no print times visible either.

Cheers,

Jay :)

Mrnt, 2020-05-17:
Hey Jay, glad you got it up and running!

> however I think the up on the Z is actually going down and vice versa.

you can fix that in Settings > Printer Profiles > click the edit icon next to the printer profile/Default Printer > Axes tab > click the checkbox "Invert Control" next to "Z"

> The X and Y seem to work for a short movement but then don’t seem to respond after the first move.

I see that too sometimes. I think you have to click the Home button to get the printer to calibrate the axes.

> interface between Simplify3D and the Flashforge 

I have not used Simplify3D maybe @garyriet can speak to that.

> read the printer status from my home automation system

You are way ahead of me :) There are a lot of plugins for OctoPrint - I would not be surprised if you can find something for that.

>  SD card on the printer direct from it’s screen

Hmm you are right - when building directly from the printer's screen OctoPrint does not enable the Pause/Cancel buttons or display print progress - I guess I should see if I can fix that.
When you upload to the SD card from within Octoprint the controls and progress should work -
print times are a bit iffy when printing from the SD card - OctoPrint should receive print progress from the printer in the form of x bytes printed/y bytes total. I think it tries to guesstimate print time based on how long it has taken so far to print the x bytes, however those times may not appear immediately.

jayceekeys, 2020-05-17:
Hey,

Thankyou so much for all your help and for being so quick to reply - if I get this working there’s a beer heading your way for sure !!!

I’ve clicked the Z invert option, however the home button doesn’t work and the down still makes the z axis go up.

I’m nervous about even trying to initiate a print from the raspberry pi SD card at this point (I’m guessing thats the SD that the screen refers to as the Guider has built in memory and not an SD) as the printer would just crash into the bed and ruin the surface at this stage.

Cheers,

Jay :)

jayceekeys, 2020-05-17:
Ps I think you replied before I had edited my original post .... it has more detail on !

Mrnt, 2020-05-17:
Oops sorry!

Ok so on the Control tab:
- make sure to click the home button for X/Y and Z before trying to control the respective axes.
- make sure the “1” or “0.1” (as in mm movement per step) buttons are selected for  X/Y and Z. This should stop it moving very far on each click of the arrow buttons.
- when making changes to the settings screens, be sure to scroll down and click Save before closing the Settings screen.

I hear you on being nervous on kicking off a print from OctoPrint. If you can get the controls working then maybe you can try printing something super simple created in FlashPrint and uploaded via OctoPrint.

Another potential issue is where the printer thinks the X/Y origin is. Usually it’s in the home location but some printers treat the center of the build plate as the origin - that’s more of an issue for when you configure Cura/Simplify3D.

jayceekeys, 2020-05-17:
Hi,

I’ve checked all the settings and made sure that save was pressed.

The home buttons have no affect whatsoever. I would expect them to send a message to home each axis but no movement even when the printer is mid axis.

The jog buttons sort of work - the z began to work in the right direction but after a bit resulted in any instruction (up or down) making the bed go up.

When the axis are responding you can only jog the axis one step (eg left) before the button no longer responds. You can then jog it right - again one step and it then stops and no longer will go any further. You can jog one nudge left one nudge right in repetition but no further.

The y axis acts the same.

I did bite the bullet and upload a tiny file .gx file to the screen just to see if it would at least home and start heating up. It didn’t. There was no response and eventually the terminal (after sending a load of code) stated the communication had timed out waiting for a response from the printer. 

Cheers,

Jay

Mrnt, 2020-05-18:
Hi Jay,

Re uploading a file to the printer - did you use the "Upload to SD Card" option? The "Upload" loads the file into the OctoPrint host and subsequent printing from within OctoPrint by clicking the print icon next to a file, tries to send the file to the printer command by command which currently will not work. The older FlashForge printers (and I'm guessing the newer ones such as yours) have a micro SD card in a socket on the logic board which functions as the printer's memory. When you select the option "Upload to SD Card" the plugin will attempt to upload it to this card/memory on the printer.

I did notice that if you click the X, Y, Z step buttons too quickly after hitting a Home button (while the printer is still moving) that my printers drop the connection to OctoPrint, so you may or may not be seeing some of that.

To help debug this:

- Please restart the printer and OctoPi so we can start a connection from a clean state. 
- Can you enable debugging for the plugin, instructions are as follows:
Turn on debug messages for the plugin (Settings > Logging, under "Logging Levels" set [octoprint.plugins.flashforge](https://github.com/Mrnt/OctoPrint-FlashForge/wiki/images/LoggingSettings.png) to "DEBUG" and then click the "+" sign next to it, then click "Save") to help troubleshoot connection issues.
- After enabling debugging, make a connection to the printer and try homing, clicking the X, Y step buttons (allow any movement to finish before clicking another button).
- Download octoprint.log from the Settings > Logging page and upload it here.

jayceekeys, 2020-05-19:
Hi,

If I use the Upload to SD Card option, it opens a Windows browser to define which drive to save the file to. As I understand, the Guider IIS does not use an internal SD card on the motherboard, it has onboard memory instead. I imagine that this is going to be a big issue in terms of using the plugin?

I'm not even too worried about printing from Octoprint ..... I just want to be able to read the printer status, so it's not a deal breaker as such. Let me know if you still want the logging info - I assume that you may not need it if the printer cannot be controlled in respect of printing?

Cheers,

Jay

jayceekeys, 2020-05-19:
Hi,

I've attached a log in case it is of use to you. The time that I did the test you asked for was today (19/05/20 at 13:11pm)

Cheers,

Jay 
[
[octoprint.log.txt](https://github.com/Mrnt/OctoPrint-FlashForge/files/4650297/octoprint.log.txt)
](url)

Mrnt, 2020-05-20:
> If I use the Upload to SD Card option, it opens a Windows browser to define which drive to save the file to. As I understand, the Guider IIS does not use an internal SD card on the motherboard, it [...]

That should be a window to search your computer for a file to upload - you should not be seeing a window showing what is on the Guider. I think you may be right that the Guider has maybe a memory chip instead on an internal SD card however I am fairly certain that it is considered internal/local memory just like the SD card is on other FlashForge printers:
[image]
In the manual it looks like there is 8GB of internal memory and the option to use a USB stick with the printer. When you select "Upload to SD" from Octoprint the file should be transferred to that internal memory.

Thanks for the log file - it looks like OctoPrint is communicating OK with the printer and it appears the printer recognizes and accepts the move commands - whether it moves is another matter! One thing I noticed that is a bit odd (and I have not seen before) and may explain what is happening is that the printer responded with `CMD M601 Received. | Control failed. | ok | ` when you reconnected to it with OctoPrint at 13:12. So maybe that is an indicator that the printer will communicate to show status, but will not allow control of movement, etc.  **Do you simultaneously have an ethernet connection to the printer via FlashPrint?**

I think at the very least you should be able to monitor temps, possibly progress using OctoPrint. I'm frustrated that you cannot get it to control the printer too do the "Upload to SD".

jayceekeys, 2020-05-20:
Hi!

Firstly apologies as I don’t know your name yet and it’s seems so abrupt to just say hi.

You sir are and absolute bloody genius !

Using the upload to sd card button as described by you allows me to print perfectly from .gx from flashprint or from .gcode form Simplify3D. I am also getting full printer status and time remaining.

I’ve installed the plugin in HomeSeer my automation system and it’s reading full printer data !!!! I should now be able to integrate it with Alexa and also my Sonos system to give me status updates and query the printer asking Alexa !!!

I’ve yet to try the controls as I suspect they may still be a little weird, but have even managed to use the internal printer camera stream to go straight into Octoprint so everything is in one place!!

I will update as soon as I test more !

Beer being delivered to your PayPal shortly !!

Many thanks you’re a legend,

Jay :)

Mrnt, 2020-05-20:
Oh that's awesome - thanks for the update! 

Yes, please provide additional feedback as you test - I will try to update the ReadMe with supported functionality by printer model at some point so other people know what to expect.

Gavin

jayceekeys, 2020-05-20:
[image]

Mrnt, 2020-05-20:
Nice - I really need to set up a camera! I have been too busy working on the plugin to tweak my printer :-(

jayceekeys, 2020-05-23:
Hey G,

I’ve been testing and the print to SD works fine. The controls are suffering from the identical problems as on the recent Finder 2.0 issue. I imagine they share the same code.

One thing I did notice was that when I send g-code from Simplify3D via the Octoprint, the time remaining estimate goes crazy - not a little out which I would always understand, but it tells me that I’ve got as long as 8 years left to wait for the print to finish ! It was a 4 hour print!
[image]

Mrnt, 2020-05-25:
J - that is indeed a long print 😄 ! Did it finish the print ok and OctoPrint return to an idle state?

Can you turn on debugging for the plugin if you have not already done so: 
Turn on debug messages for the plugin (Settings > Logging, under "Logging Levels" set [octoprint.plugins.flashforge](https://github.com/Mrnt/OctoPrint-FlashForge/wiki/images/LoggingSettings.png)
to "DEBUG" and then click the "+" sign next to it, then click "Save") to help troubleshoot connection issues.

Next time you print from SD download the octoprint.log from the Settings>Logging screen then upload it here as a zip file.

Mrnt, 2020-05-27:
Thanks for confirmation on the controls not working (like the Finder 2). I think I almost have a solution for that - Question: if it was an option in the printer profile to enable a work around, would you find the setting?

Mrnt, 2020-05-29:
@jayceekeys Hi J -  I have added some functionality to the development branch to support this printer via a checkbox under the Axes tab of the printer profile.
You can install the development branch using the Plugin Manager > Get More... > and under "...from URL" enter:
https://github.com/Mrnt/OctoPrint-FlashForge/archive/devel.zip

Can you give it a try and let me know if it works?

jayceekeys, 2020-05-30:
Hey buddy,

Sorry for being quiet after your last message to me. My Guider IIS has faults since it arrived so went back to Sweden this week for investigation. Really hoping they turn it around quickly so I can get it back and start work with it again but they’re not sure what the fault is caused by so there might be a delay before I can get the testing done.

Ps the Finder issue is identical to what I was getting so imagine it will be the exact same problem :)

Cheers,

Jay