Mrnt/OctoPrint-FlashForge #30: Dremel 3d45

garyriet, 2020-05-21:
Question, will the plugin work withe 3D45? Just picked up a brand new one for 500.

Mrnt, 2020-05-21:
That seems like a good deal!

The plugin does not have the USB product ID for the 3D45 right now, although it does have the one for the 3D20. You can either edit the code in [__init__.py](https://github.com/Mrnt/OctoPrint-FlashForge/blob/895814d2f1fcdd961c39985c3cd158a128dafb2e/octoprint_flashforge/__init__.py#L18-L21) so that 
```
0x8889: "Dremel IdeaBuilder 3D20"
```
becomes 
```
0x8889: "Dremel IdeaBuilder 3D20", 0x<newproductIDhere>: "Dremel IdeaBuilder 3D45"
```
or upload the ID and I can add it. 

If you feel comfortable adding the code, it would be helpful because you could go ahead and test with it and see what works.

garyriet, 2020-05-21:
I dotn feel comfortable changing code. Lmao.  I’ll look into 
Thanks 
Gary

https://patents.justia.com/inventor/gary-rietmann

garyriet, 2020-05-21:
Going tuna fishing tonight, I look at it over the weekend. Its in the middle of a print right now.

Mrnt, 2020-05-21:
Fun! 

Ha - just looked at your link, I think you can do this ;)

garyriet, 2020-05-21:
https://www.linkedin.com/in/gary-rietmann-4054007/

LMAO Yea I got it. Going thru bad divorce that mad me crack up. Thanks for that

garyriet, 2020-05-24:
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0x69144bf0>, starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 115200
Send: M601 S0
Recv: CMD M601 Received.
Recv: Control Success.
Recv: ok
Changing monitoring state from "Detecting baudrate" to "Operational"
Send: M119
Recv: status: ready
Recv: temperature: 25
Recv: platform_temperature: 21
Recv: chamber_temperature: 19
Recv: extruder_target_temperature: 0
Recv: buildPlate_target_temperature: 0
Recv: error_code: 200
Recv: door_open: 1
Recv: remaining: 0
Recv: totalTime: 0
Recv: firmware_version: v3.0_R02.09.04
Recv: jobstatus:
Recv: progress: 0
Recv: layer: 50
Recv: jobname:
Recv: elaspedtime: 0
Recv: networkBuild: 0
Recv: filament_type: ECO-ABS
Recv: usbBuild: 0
Recv: fanSpeed: 0
Recv: message: success
Recv: ok
Send: M27
Recv: CMD M27 Received.
Recv: SD printing byte 0/0
Recv: ok
Send: M115
Recv: CMD M115 Received.
Recv: machine_type: DREMEL 3D45 IDEA BUILDER
Recv: firmware_version: v3.0_R02.09.04
Recv: api_version: 1.0.2-alpha
Recv: SN: 803036493
Recv: ethernet_ip: 192.168.1.79
Recv: ethernet_connected: 1
Recv: wifi_ip:
Recv: wifi_connected: 0
Recv: error_code: 200
Recv: message: success
Recv: usb_connected: 1
Recv: ok
Send: M146 r255 g255 b255
Recv: CMD M146 Received.
Recv: ok
Send: M119
Recv: status: ready
Recv: temperature: 25
Recv: platform_temperature: 21
Recv: chamber_temperature: 19
Recv: extruder_target_temperature: 0
Recv: buildPlate_target_temperature: 0
Recv: error_code: 200
Recv: door_open: 1
Recv: remaining: 0
Recv: totalTime: 0
Recv: firmware_version: v3.0_R02.09.04
Recv: jobstatus:
Recv: progress: 0
Recv: layer: 50
Recv: jobname:
Recv: elaspedtime: 0
Recv: networkBuild: 0
Recv: filament_type: ECO-ABS
Recv: usbBuild: 0
Recv: fanSpeed: 0
Recv: message: success
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:25/0 B:21/0 C:19 E:0ok
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M119
Recv: status: ready
Recv: temperature: 25
Recv: platform_temperature: 21
Recv: chamber_temperature: 19
Recv: extruder_target_temperature: 0
Recv: buildPlate_target_temperature: 0
Recv: error_code: 200
Recv: door_open: 1
Recv: remaining: 0
Recv: totalTime: 0
Recv: firmware_version: v3.0_R02.09.04
Recv: jobstatus:
Recv: progress: 0
Recv: layer: 50
Recv: jobname:
Recv: elaspedtime: 0
Recv: networkBuild: 0
Recv: filament_type: ECO-ABS
Recv: usbBuild: 0
Recv: fanSpeed: 0
Recv: message: success
Recv: ok

garyriet, 2020-05-24:
Sweet it works!
"Dremel": {0x8889: "Dremel IdeaBuilder 3D20", 0x888d: "Dremel IdeaBuilder 3D45"},

garyriet, 2020-05-24:
I still have to set the other settings for OctoPrint

Mrnt, 2020-05-25:
Nice! It’s a pretty chatty little thing isn’t it (looking at the response to M119)

eduncan911, 2020-08-05:
Sweet!  What's the status of this plugin?

I have two 3D45s loaned from work for face shields, and a single 3D20 that the daughter won in a contest to connect.

Always wondered, can a single OctoPrint instance control 3 printers connected to the same RPi?

Would this plugin work for 3 printers connected to the same RPi (e.g. no global vars)?

I also have some UP Boards (25x faster per core than a RPi, and native x64 Intel CPU for real Linux) that I can use if more horsepower is needed.

eduncan911, 2020-09-10:
There is a PR already open for this issue: https://github.com/Mrnt/OctoPrint-FlashForge/pull/9

Just needs some final testing.  I am going to try this over the next few days/week, as I finish testing the 3D20 changes of late.

eduncan911, 2020-09-10:
So about that Camera in the 3D45... It appears to be a an M-JPEG stream.  I found this URL when directly accessing the printer:

http://<ip-of-printer>:10123

It immediately brings up a default web server landing page of "M-JPEG streamer":

[image]

Why they want the extra overhead of running an embedded Java, and encoding individual JPEG images, is insane.  First thing that pops into my head is, "how unsecure is this free-standing java web server on a printer with no security?"  Good thing I run these printers on a dedicated Untrusted VLAN for IoT crap.  I would never let this thing on my regular wireless.

Clicking on the Stream, clearly shows a simple `<img src="/?action=stream" />` HTML code that one could embed to show the camera.

[image]

And it streams the JPEG images, with just that img!

As a matter of fact, hitting the url directly brings up a browser-window of what i think is around 800x640 resolution video of the camera feed: http://<IP of printer>:10123/?action=stream.  Note no "HTTPS", just plaintext HTTP - and no security.

That URL is also compatible with VideoLAN video streaming:

[image]

The final interesting fact is, we have camera controls!

[image]

---

So about OctoPrint and cameras... It got me thinking, can it stream the streaming URL?

YES IT CAN!!!

[image]

So in summary, when we get the 3D45 working with this plugin, we can clearly document how one would add the Camera to their octoprint instance by saying something like:

> To display the camera, go into OctoPrint's settings for Webcam & Timelaspe, and enter the URL: http://172.20.40.101:10123/?action=stream

Nice!

Mrnt, 2020-09-10:
@eduncan911 Nice job on the camera! 

This made me chuckle:
` Good thing I run these printers on a dedicated Untrusted VLAN for IoT crap. I would never let this thing on my regular wireless.`
Yep, me too, along with the wifi security cams of questionable security...

Ve2mrx, 2021-03-19:
I have a Dremel 3D45 if you need to test/troubleshoot.

Ve2mrx, 2021-03-19:
USB ID: Bus 001 Device 003: ID 2a89:888d

eduncan911, 2021-03-19:
@Ve2mrx Take a look at this PR: https://github.com/Mrnt/OctoPrint-FlashForge/pull/9#issuecomment-715574187

That's basically where we stand.  The plugin needs to be tested.

Note, right now there are some "work arounds" for the 3D20 built into the codebase.  The 3D45 seemed much more forgiving of regular Gcode in my previous tests, whereas the 3D20 has caused several lockups and reboots that were worked around (and now works great with).

Hopefully we don't need to do that.

Also, someone needs to pull down the codebase and get it updated with `master` branch, in that PR.

bvaerewyck, 2021-12-07:
I have access to a Dremel 3D45 if this gets picked up again.

nbetcher, 2024-12-01:
Is there any way to rekindle this project? I'd really like to use OctoPrint with my Dremel 3D45 without limitations. According to the README, Dremel 3D45 support is basically non-existent. [Here are some important changes that could be merged](https://github.com/Mrnt/OctoPrint-FlashForge/compare/master...powpingdone:OctoPrint-FlashForge:master).

Plus, (and I thought I found this on here), but [there's comprehensive documentation on the G-Code for it now](https://docs.google.com/document/d/1Of4ZUU13UWfF1-2vZIfyzsz1yCOvCj9v9FT0aroKhuU/edit?usp=sharing). ([Mirror in case it's pulled](https://docs.google.com/document/d/e/2PACX-1vSccKjJ91wNJhLefybW0w1Z9BGBWA6DzrJwa473_sqnaLIabLZRZo1MNQvgzbX3nXXuyuRuD0KbCEXe/pub), but also cleaner to read since it's "Published for Web")