Mrnt/OctoPrint-FlashForge #32: Dremel 3D20 time-outs on/after streaming to SD

RenskeB, 2020-05-24:
First: thank you so much!!

During/ after streaming the file to SD the connection times-out.
Set communication time-out to 60 seconds.
Haven't touched anything during streaming.

This happens when I use upload to SD and when I send directly from CURA.

Yesterday I had the same problem, after turning the printer off and on and reconnecting, it streamed and started printing. Just that once. Can't get it to start (even the same file) again. 

[image]

jimprince, 2020-05-25:
I've had this a couple times. Once was with the same file that I was able to print twice, had the error once and had to restart the printer, and then had another successful print. 

Out of curiosity, how large is the file you're trying to send?

jimprince, 2020-05-25:
BTW, if you're able to turn on debugging and then download a copy of octopi.log after you've had the error, that would go a long way in helping to fin the issue. You can find how to turn on debugging and download log files here:  [https://github.com/Mrnt/OctoPrint-FlashForge#troubleshooting](https://github.com/Mrnt/OctoPrint-FlashForge#troubleshooting)

Mrnt, 2020-05-25:
@RenskeB as @jimprince suggested (thanks Brad) capturing a log of the plugin with debugging turned on and uploading it as a zip would be very helpful.

A couple of thoughts/questions:
- Are you using a Raspberry Pi if so what model?
- Make sure you do not have something else connected  to the printer at the same time, eg Dremel Digilab or Dremel Print Cloud connected via WiFi.
- Make sure the usb connection between your Raspberry Pi is good (cable, plugged in properly, etc)
- Make sure the Raspberry Pi power source is adequate and not causing the Pi to restart or throttle.

RenskeB, 2020-05-26:
@jimprince @Mrnt 

Thanks for your replies! Debugging was on, also for FlashForge, I can post them later (It is just repeatedly trying to write the same command to the Dremel? Not sure if I'm understandigthe logfiles correctly.), but I think @jimprince provided me with the correct solution already. When a file (with the same name?) already exists on the Dremel, it will not start (upload) the print (unfortunately I cannot check this, it doesn't show me the contents of the 'SD', but of course I can add time&date to the filename to solve this). 

I know this has nothing to do with the Plugin, but maybe, could you drop me a hint for Cura? It is giving me a headache. I've installed the Dremel as a FFF printer and sending a file using the plugin works just fine (when I take above into account). However, it refuses to wait until the printhead reaches the correct temperature. I know (sorry!) this question shouldn't be here, but maybe you have a nice and quick solution for this. If not, that is, of course, no problem. Very happy with the plugin! Thank you so much!!!!

RenskeB, 2020-05-26:
> @RenskeB as @jimprince suggested (thanks Brad) capturing a log of the plugin with debugging turned on and uploading it as a zip would be very helpful.
> A couple of thoughts/questions:
>
> Are you using [...]

It's a Pi3, model B, cables & power source have been fine for about a year for the Prusa. The only tricky thing I can think of is that I'm running two instances of octopi (I know, no support), but that seems to work fine (with fresh file names).

Mrnt, 2020-05-27:
@RenskeB OK sounds like you may have found the problem with the uploads (the -9 is a new error to me) thanks to @jimprince.

Regarding the heating issue, there are a two temperature wait commands - one for the extruders and one for the bed which may help if you set them up in the end of the "begin" G-Code block in Cura - see the [Wiki](https://github.com/Mrnt/OctoPrint-FlashForge/wiki/G-Code-Dictionary#m6---wait-for-extruder).

Mrnt, 2020-10-02:
@RenskeB is uploading to the SD card directly or from Cura working OK for you now? Can we close this?