Mrnt/OctoPrint-FlashForge #34: Cannot Connect To FlashForge Guider II

maesoph, 2020-07-04:
Firmware:  1.4.6.167 g21 20200521
Program:  OctoPrint 1.4.0
Computer:  Raspberry Pi 4B  (that's all they had at Target)

My apologies if this is operator error.

[octoprint.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/4873407/octoprint.log)

Mrnt, 2020-07-06:
Thanks for the log! No, not user error - thanks to your log I have now added the USB ID for the Guider II. Please download ver 0.1.19 and give that a test. 

Note, per the ReadMe, the one time adjustment that you will likely have to make re USB permissions.

maesoph, 2020-07-06:
Thanks for the lightning response!  Looking forward to OctoLapse.  It's hard to get the attention of today's youth.  Need the WOW factor of OctoLapse for sure.

Mrnt, 2020-07-06:
> It's hard to get the attention of today's youth.
Ha tell me about it.

Once you get it connected can you test if the homing, arrow controls work on the OctoPrint "Controls" tab - they don't seem to on the Guider IIs and the Finder II...

maesoph, 2020-07-06:
Arrow keys are inconsistent and seem to only work for one bump in a given direction until changing the amount 1, 10, 100...

Home doesn't work.  

Also not getting my webcam to work.  It might be the $21 Chinese webcam though.  Added the -y option and set size to 1920 X 1080 but still getting "Webcam stream not loaded".  

Once you get it connected can you test if the homing, arrow controls work on the OctoPrint "Controls" tab - they don't seem to on the Guider IIs and the Finder II...

Mrnt, 2020-07-07:
It looks like the Guider II has the same issue as the Guider IIs, Finder II in that it does not understand the relative movement command (so steps do not work) or have the ability to home individual axes. I'm working on a fix for this.

I don't think the webcam issue is related to the plugin however ;-)

maesoph, 2020-07-07:
I'm forgetting about the web cam for now. Will likely get a real one before continuing. 

Getting nothing but time-outs when trying to print the standard 20mm xyz cube.

This is not my priority project so if you're too busy for hand holding, that's fine. LOL

Mrnt, 2020-07-08:
Note that you have to use the "Upload to SD" button - the direct print from OctoPrint method where you load the file into OctoPrint and then select it from the list of files on the left and print it does not work right now. Basically the printer does not respond until its finished moving, OctoPrint doesn't send anything until it receives an OK response and the printer times out the connection if it does not hear a command from the host within about 4 seconds. Good times.... ;-)

maesoph, 2020-07-08:
Upload to SD is working fine.  But does this imply that OctoLapse won't work?

Mrnt, 2020-07-08:
Oh you may be right. I have not looked at OctoLapse to see whether it takes images based on time or whether it depends on OctoPrint knowing what layer is currently being printed. I am working on the direct print method, but please understand I have no support from FlashForge so everything is by trial and error...

maesoph, 2020-07-09:
I am definitely looking for the layer based octolapse which does not work with Upload To SD from what I understand but regular time based lapse still works fine but time  based lapse can be done with a regular GoPro or whatever.  This isn't a priority project for me - maybe I'll fight with it later......  Working on the "World's Fastest Ping Pong Ball Cannon" until it's done....

Mrnt, 2020-08-31:
Can try the latest build and:

1. connect to the Guider
2. allow control of the X, Y, Z axes INCLUDING the Home buttons
3. upload a file to the SD card and verify that upload works, that it starts to print and you can pause, cancel the print

maesoph, 2020-09-01:
Thanks for the notification, but I've had to put this on hold.  I've got too many geeky irons in the fire and have to back off on some of them.  Here's my latest geek project:
St. John Pico Balloon AROUND THE WORLD!!! DONE!!!

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
St. John Pico Balloon AROUND THE WORLD!!! DONE!!!

 |

 |

 |

 
 

   
   - connect to the Guider

Mrnt, 2020-09-01:
That's really cool! Congrats!

elnazh68, 2021-07-11:
Hi. I have tried a couple of times to connect Octorprint (on Raspberry Pi) to my Guider 2 printer and didn't have any success, although I tried many options. Does anyone have a step by step guide on how to connect them together? Thanks.