Mrnt/OctoPrint-FlashForge #37: Octoprint 1.4.1

rmhalvorsen, 2020-08-05:
Upgrading to octoprint 1.4.1 breaks usb connection, cannot connect to printer (Finder2.0) anymore.

Connection State: Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)

brendanm720, 2020-08-06:
I can confirm that this has broken the plugin.  Here's the console readout.

It doesn't matter what I change the baud rate to.

Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"

brendanm720, 2020-08-06:
I'm probably going to downgrade back to my original version.

brendanm720, 2020-08-07:
I finally got the Serial on the Finder 2.0 to work again with OctoPrint 1.3.12 and OctoPrint-Flashforge 0.1.17.  

0.1.18 and 0.1.19 seemed to connect to the serial but it returned the results of the commands as bytes which something was not expecting.  Please see the attached log for more info.

[serial.log.2020-08-07_15-01-17.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5043362/serial.log.2020-08-07_15-01-17.log)

TheMagicDick, 2020-08-08:
Ditto on the connection problem on 1.4.2. 

I did get it to work again be rolling back to octoprint 1.3.12 and octoprint-flashforge 1.17

I also had to disconnect my printer from my polar cloud. I going to try to solve that by using the polar cloud plugin that also only works on older octoprint versions

Mrnt, 2020-08-12:
Sorry for the delay in responding, have been busy. Will take a look.

trejan, 2020-08-14:
Fix is at https://github.com/Mrnt/OctoPrint-FlashForge/pull/38

Mrnt, 2020-08-14:
Thank you @trejan for doing this - merged into a new release. 

@TheMagicDick @brendanm720 @rmhalvorsen Please try with ver 0.1.20

geirme, 2020-08-15:
I tried to upgrade to 1.4.2 now after first updating the FlashForge plugin from 0.1.19 to 0.1.20

First the installation of the plugin failed with the message "TabError: inconsistent use of tabs and spaces in indentation"
While trying to dive into the __init.py__ file to se if any tabs should be replaced with white spaces instead suddenly i found a new update, versjon 0.1.21 and tried that one.

Magic then happened and it all works perfect now. I also finally get the autoconnect feature introduced in 1.4.1 working so when starting up octoprint it now automatically connects to my FlasForge (Finder 2.0) printer directly!

(I can now happly go to bed as the clock has passed 03:00 AM...)

trejan, 2020-08-15:
Sorry. The tab/spaces was my mistake. Mrnt's version should be fixed now.

geirme, 2020-08-15:
That explains the sudden raise in version number. Well, it seems to be working fine!

eduncan911, 2020-08-18:
New install (on an x86 device, no RPi).  First time using OctoPrint, and this plugin!

* Dremel 3D20
* OctoPrint 1.4.2
* OctoPrint-FlashForge 0.1.21

Wiped and installed twice now; however, same error as OP.  

I am installing per README, and followed the `master` zip file approach, as well as just searching Get more and installing that way.

Should I be specifying individual tags to download from within the Plugin Manager?