Mrnt/OctoPrint-FlashForge #41: Issue connecting

Fred583, 2020-08-19:
So following the steps, I did encounter the permission error, so I followed the instructions on the octopi terminal. Once I restarted, I get this error from the octopi terminal:

Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: unknown@115200, unknown@250000, unknown@230400, unknown@57600, unknown@38400, unknown@19200, unknown@9600
Trying port unknown, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Unexpected error while writing to serial port: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_IO [-1])' @ comm.py:_do_send_without_checksum:3593
Changing monitoring state from "Detecting serial connection" to "Offline (Error: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_IO [-1])' @ comm.py:_do_send_without_checksum:3593)"
Closing down send loop

Mrnt, 2020-08-19:
Can you turn on debug messages for the plugin (Settings > Logging, under "Logging Levels" set [octoprint.plugins.flashforge](https://github.com/Mrnt/OctoPrint-FlashForge/wiki/images/LoggingSettings.png) to "DEBUG" and then click the "+" sign next to it, then click "Save"), then try connecting again and then upload octopi.log here and I will take a look.

Mrnt, 2020-08-19:
BTW what printer and OS (Raspbian/etc) are you using?

Fred583, 2020-08-19:
I am using a Dremel 3D20 printer with the Octoprint image on a micro sd card in a Raspberry Pi Zero W.

Hold on, it just connected. Lemme give it a try.

Fred583, 2020-08-19:
Ok, its connecting then disconnecting, saying: Offline (Error: Too many consecutive timeouts, printer still connected and alive?)

The terminal says:
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 0 port/baudrate candidates: 
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 0 port/baudrate candidates: 
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 0 port/baudrate candidates: 
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 0 port/baudrate candidates: 
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: BOSCH Dremel 3D Printer@115200, BOSCH Dremel 3D Printer@250000, BOSCH Dremel 3D Printer@230400, BOSCH Dremel 3D Printer@57600, BOSCH Dremel 3D Printer@38400, BOSCH Dremel 3D Printer@19200, BOSCH Dremel 3D Printer@9600
Trying port BOSCH Dremel 3D Printer, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0xa88be7f0>, starting monitor
Recv: CMD M601 Received.
Recv: Control Success.
Recv: ok
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: M119
Recv: CMD M119 Received.
Recv: Endstop: X-max: 1 Y-max: 1 Z-min: 0
Recv: MachineStatus: READY
Recv: MoveMode: READY
Recv: ok
Send: M27
Recv: CMD M27 Received.
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: BOSCH Dremel 3D Printer@115200, BOSCH Dremel 3D Printer@250000, BOSCH Dremel 3D Printer@230400, BOSCH Dremel 3D Printer@57600, BOSCH Dremel 3D Printer@38400, BOSCH Dremel 3D Printer@19200, BOSCH Dremel 3D Printer@9600
Trying port BOSCH Dremel 3D Printer, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0xa910cc70>, starting monitor
Recv: CMD M601 Received.
Recv: Control Success.
Recv: ok
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: M119
Recv: CMD M119 Received.
Recv: Endstop: X-max: 1 Y-max: 1 Z-min: 0
Recv: MachineStatus: READY
Recv: MoveMode: READY
Recv: ok
Send: M27
Recv: CMD M27 Received.
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M115
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M119
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: BOSCH Dremel 3D Printer@115200, BOSCH Dremel 3D Printer@250000, BOSCH Dremel 3D Printer@230400, BOSCH Dremel 3D Printer@57600, BOSCH Dremel 3D Printer@38400, BOSCH Dremel 3D Printer@19200, BOSCH Dremel 3D Printer@9600
Trying port BOSCH Dremel 3D Printer, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0xa922d6f0>, starting monitor
[... 19 lines cut ...]
Changing monitoring state from "Offline" to "Detecting serial connection"
[... cut ...]
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M119
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor

Fred583, 2020-08-19:
It was already on debug, here is the log.
[octoprint.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5099264/octoprint.log)

Mrnt, 2020-08-19:
Ah it looks like the same issue as https://github.com/Mrnt/OctoPrint-FlashForge/issues/40 since. that was also a Dremel 3D20. Looks like I might need to do some special casing for this printer...

Fred583, 2020-08-19:
I just saw #40.

eduncan911, 2020-08-19:
renamed that other issue to make it easier to search on next time.  ;)

Mrnt, 2020-08-20:
@Fred583 no worries! 

Closing as a duplicate of #40

closed 2020-08-20 by Mrnt