Mrnt/OctoPrint-FlashForge #46: Dremel 3D45 printer head trying to go past home position

visualvalhalla, 2020-08-31:
I'm new to octoprint, but have experience in programming IoT devices.  I have a 3D45 and looked at tryign out the octoprint plugin for it.  Have it installed on Raspberry Pi 4 and tested with bacic Fan on/off move head commands.  All seem to be fine until I tried to upload a gcode output from the Dremel 3D slicer and attempt to print.  The print head immediately tries to go past the home position and I have to shut the printer off to prevent further damage.  Tried it multiple times with the same results.
I've attached the full session log

[ocotprint_printhead_error.txt](https://github.com/Mrnt/OctoPrint-FlashForge/files/5148034/ocotprint_printhead_error.txt)

I am not familiar with gcode, but why is octoprint sending the print head to the -X direction?
Send: G1 F1500 X0.163 Y22.179 E0.04145
Recv: CMD G1 Received.
Recv: ok
Send: G1 X-.604 Y22.171 E0.08163
Recv: CMD G1 Received.
Recv: ok
Send: G1 X-1.403 Y22.134 E0.12353
Recv: CMD G1 Received.

If the coordinates are referenced off the center position of the bed, I did not see the head move to the middle.

visualvalhalla, 2020-08-31:
User error - maybe.

I see only Upload to SD is support.  Closing issue

Mrnt, 2020-08-31:
@visualvalhalla I have added support for printing directly from OctoPrint though it is new and may not work for all printers and almost certainly will require your help to debug.
If you are willing to test maybe we can get it to work. You will need to turn on debugging for the plugin - see Troubleshooting section.
Also can you try:

1. making a fresh connection
2. use the OctoPrint move controls to move the head to some random position
3. go to the OctoPrint "Terminal" tab and enter the command `G28` and report back what happens

Mrnt, 2020-08-31:
Also if you feel ok sharing the file you were printing, can you upload that here too?

visualvalhalla, 2020-08-31:
Attached gcode sliced from this stl: https://www.thingiverse.com/thing:2197230

[D3_Latice_torture_test_2.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/5148186/D3_Latice_torture_test_2.zip)

visualvalhalla, 2020-08-31:
rebooted printer, connected Octo to printer.
Fan on/off works.  Tapping LEFT and DOWN arrows moved the print head in the expected directions

Send: G28
Recv: CMD G28 Received.
Recv: ok
Recv: status: ready
Recv: temperature: 48
Recv: platform_temperature: 31
Recv: chamber_temperature: 30
Recv: extruder_target_temperature: 0
Recv: buildPlate_target_temperature: 0
Recv: error_code: 200
Recv: door_open: 0
Recv: remaining: 0
Recv: totalTime: 0
+++++++++========++++++
Nothing happened to the print head

visualvalhalla, 2020-08-31:
I have debug logging turned on for octoprint.plugins.flashforge but I don't see any log files

Mrnt, 2020-08-31:
It will be in the octopi.log

visualvalhalla, 2020-08-31:
[octoprint.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5148223/octoprint.log)

Debug on.  G28

visualvalhalla, 2020-08-31:
[octoprint (1)_attempt to print.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5148226/octoprint.1._attempt.to.print.log)
Right after that, I attempted to print to print the gocde that was attached earlier

Mrnt, 2020-08-31:
re the G28 issue - just to double check, you:

1. used the X, Y buttons to move the head to some random position 
2. went to the terminal and entered G28 and the head did not return to home?

If you try `G28 X Y` instead, does that work?

visualvalhalla, 2020-08-31:
Yes, I tried combinations of X Y and Z.... head did not move at all

Mrnt, 2020-08-31:
And:
The head is not already at home - definitely at some random location?
What about `G28 X` or `G28 Y`?

visualvalhalla, 2020-08-31:
[octoprint _G28_test.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5148305/octoprint._G28_test.log)

Confirm the head is in the middle of the plate.  No movement observed

visualvalhalla, 2020-08-31:
Would sniffing the USB comms between a PC and 3D45 help?

Mrnt, 2020-08-31:
Are you able to print from the Digilab Slicer directly? ie without uploading the file from the slicer to the SD card? If so then sniffing the connection should show the slicer sending the file command by command and the response. That might be helpful. If it just uploads it to the SD card, then not useful.

I pulled up the Digilab Slicer but I do not have a printer for it to connect with - does it have a control panel to move the head around directly, maybe make it "Home"? If so then that would be useful to sniff.

visualvalhalla, 2020-08-31:
I've not tried the direct USB route yet to print. I use network connection to the printer.  The Dremel slicer does not have controls other than print.

Mrnt, 2020-08-31:
The center of the plate is the origin (ie, X,Y coordinates 0,0) and I feel like what is happening is that it does not home and therefore assumes the current head position is 0,0 so when it starts printing using the `G1 X-_some number_` it just starts moving to the right and bangs against the side instead of moving to the center and then printing from there. 

Another thing you can try is:
Close the connection
Power printer off then on
Connect OctoPi to the printer
Send `M114` in the Terminal - note the positions it returns
Send `G28 X` then `G28 Y` in the Terminal
Send `M114` in the Terminal - note what it returns - there should be new values for X, Y

Mrnt, 2020-08-31:
> I've not tried the direct USB route yet to print. I use network connection to the printer. The Dremel slicer does not have controls other than print.

When you do the direct network connection can you tell if it uploads the file to SD or does it seem like it is sending command by command?

visualvalhalla, 2020-08-31:
Send: M114
Recv: CMD M114 Received.
Recv: X:34.0742 Y:-51.3811 Z:-5 E0:0
[...]
Send: G28 X
Recv: CMD G28 Received.
Recv: ok
Recv: status: ready
Send: M114
Recv: CMD M114 Received.
Recv: X:34.0742 Y:-51.3811 Z:-5 E0:0

visualvalhalla, 2020-08-31:
Send: M114
Recv: CMD M114 Received.
Recv: X:64.0742 Y:-51.3811 Z:-5 E0:0
Recv: B:0
Recv: ok
Send: G91
Recv: CMD G91 Received.
Recv: ok
Send: G1 X-10 F6000
Recv: CMD G1 Received.
Recv: ok
Send: G90
Recv: CMD G90 Received.
Recv: ok
Send: G91
Recv: CMD G91 Received.
Recv: ok
Send: G1 X-10 F6000
Recv: CMD G1 Received.
Recv: ok
Send: G90
Recv: CMD G90 Received.
Recv: ok
Send: G91
Recv: CMD G91 Received.
Recv: ok
Send: G1 X-10 F6000
Recv: CMD G1 Received.
Recv: ok
Send: G90
Recv: CMD G90 Received.
Recv: ok
Send: G91
Recv: CMD G91 Received.
Recv: ok
Send: G1 Y10 F6000
Recv: CMD G1 Received.
Recv: ok
Send: G90
Recv: CMD G90 Received.
Recv: ok
Send: M114
Recv: CMD M114 Received.
Recv: X:34.0742 Y:-41.3811 Z:-5 E0:0
Recv: B:0
Recv: ok

Mrnt, 2020-08-31:
> Recv: CMD M114 Received.
> Recv: X:34.0742 Y:-51.3811 Z:-5 E0:0

Hmm I don't think thats even close to the right numbers for "home" on that printer.

> Recv: CMD M114 Received.
> Recv: X:64.0742 Y:-51.3811 Z:-5 E0:0

How did you get the X coordinate to increase? That would mean that it would have to think it moved the head to the righ (since "Home" was X:34.0742 Y:-51.3811 Z:-5 E0:0)?

visualvalhalla, 2020-08-31:
Sorry. That wasn't at home position.   Repeating test

visualvalhalla, 2020-08-31:
Head at home position:
Recv: CMD M114 Received.
Recv: X:100.962 Y:71.089 Z:-5 E0:0

Mrnt, 2020-08-31:
I need to call it a night I'm afraid. If you want to experiment, I made a dictionary of g-code to try and document what is supported and what they do. It's in the Wiki here: https://github.com/Mrnt/OctoPrint-FlashForge/wiki/G-Code-Dictionary

Like the plugin, it's a work in progress based on some scant docs someone found and trial and error...

visualvalhalla, 2020-08-31:
I appreciate the hard work you put in.  Let's get the printer working

visualvalhalla, 2020-09-01:
I'm on a FB Dremel 3D printer group.  There were a couple Dremel employees in the group and we had a conversation about this issue.  It appears that the Windows driver for 3D45 is broken and Dremel has decided to discontinue support for the USB port.  
As such, I am not able to capture the USB comms between DigiLabs and the printer.