Mrnt/OctoPrint-FlashForge #48: OctoPrint goes offline while printing

zenixnet, 2020-09-03:
Since updating to 0.2.2 OctoPrint will go offline a few minutes after printing has started. The print will continue, but I can't get into OctoPrint or SSH or anything. I just have to unplug and reboot. I'm only running this plugin on it, everything else is as it comes. Only started happening after updating to 0.2.2.

Attaching a log. Most recent event was around 2020-09-03 15:45:11.

Printer is PowerSpec Ultra 3D, uploading to SD card. 

[octoprint.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5169897/octoprint.log)

Mrnt, 2020-09-03:
Sorry to hear that! I don't see any debug messages from the plugin, but I do see a message that OctoPrint cannot connect to the internet (though that should have no effect on the plugin). 

Can you:

1. Turn on debugging for the plugin (see Troubleshooting section of Readme) & delete octoprint.log
2. Restart OctoPrint
3. Do another SD Upload
4. Upload octoprint.log

Mrnt, 2020-09-04:
I think I see what's happening. I'll have a new build up shortly, meanwhile you may have to go back to pre 0.2 to get reliable SD uploads.

zenixnet, 2020-09-13:
Naturally after I post this ticket it stops happening... Gonna keep trying. Did it like 4 times in a row before posting, now nothing after a few prints.

Mrnt, 2020-09-14:
I just took another look at the log file and noticed something odd with the timestamps:
```
2020-09-03 15:45:18,980 - octoprint.plugins.flashforge - INFO - Starting SDCard upload from SquashFeet.gx to SquashFeet.gx
2020-09-03 15:39:53,459 - octoprint.startup - INFO - ******************************************************************************

```
there's a block of debug messages out of sync, not sure why that would happen.

zenixnet, 2020-09-14:
I was going to upload the log of the last print so you could at least see how it went but Octoprint isn't showing any log files to download. 
[image]

Mrnt, 2020-09-14:
If you delete the logs I think you have to restart OctoPrint to get it to create new logs.

Another though - are you using a Raspberry Pi? If so, check that you do not have SD card corruption or power supply voltage issues.

zenixnet, 2020-09-29:
Updated to 0.23 and it printed fine, then my thermistor went bad. Got that fixed and have printed twice successfully on 0.24. So I'm gonna assume at this point whatever you did fixed it, but if not I'll post back.

Mrnt, 2020-09-29:
> Edit 2: Undo that last edit, my webcam fell off my printer, knocked my Pi off the table and came unplugged.

Ouch. I hear you. I thought I had my webcam perfectly positioned until I sent the extruder header to the edge of the platform and heard a weird banging noise and lost the picture...

closed 2020-12-16 by zenixnet