Mrnt/OctoPrint-FlashForge #51: Can't connect to Flashforge Finder

devslashlol, 2020-09-24:
Hi! I'm using a regular USB cable (USB-A to USB-A) to connect from my Raspberry Pi with OctoPi/print to my Flashforge Finder (it's less than a year old, so I'm not sure how to tell if it is a v1 or v2). 
Anyway, Octoprint does not recognize the device. I turned on debugging, and I get this: 

2020-09-24 07:41:34,560 - octoprint.util.comm - INFO - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2020-09-24 07:41:34,566 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-09-24 07:42:23,508 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.flashforge level to 10
2020-09-24 07:42:23,562 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 07:42:23,579 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:23,581 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 07:42:23,583 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 07:42:23,586 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:23,676 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 07:42:23,682 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:23,683 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 07:42:23,684 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 07:42:23,685 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,154 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 07:42:26,170 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,173 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 07:42:26,175 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 07:42:26,177 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,247 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"
2020-09-24 07:42:26,268 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 07:42:26,275 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,276 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 07:42:26,277 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 07:42:26,278 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,280 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 0 port/baudrate candidates: 
2020-09-24 07:42:26,281 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
2020-09-24 07:42:26,288 - octoprint.util.comm - INFO - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2020-09-24 07:42:26,292 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-09-24 07:42:26,424 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 07:42:26,431 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,432 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 07:42:26,433 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 07:42:26,435 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,502 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 07:42:26,509 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,510 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 07:42:26,511 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 07:42:26,512 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,696 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,698 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,700 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,701 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,703 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,704 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,706 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,707 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,709 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,710 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,712 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,713 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,715 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,716 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,718 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,719 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 07:42:26,766 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 07:42:26,774 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 07:42:26,776 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 07:42:26,777 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 07:42:26,777 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002

Mrnt, 2020-09-24:
Thanks for the detailed debugging - if it is no more than a year old, it is probably a Finder v2, either way it should be detected.

- Which model Pi is this?
- Do you have it connected to the pi via a USB hub?

Other thoughts:
- is the printer turned on?
- is the cable good/plugged in properly?
- is the printer detected on your regular computer with the same cable?

devslashlol, 2020-09-24:
Hi! thanks for getting back to me so fast. This is a brand new Pi, so I think 4B? I do not have it connected to hub. The Pi goes straight into the USB A of the finder. There is a USB "B" port as well. Not sure if it matters which one I use? 
The printer is on
The cable is indeed plugged in. The printer is only connected to the Pi. I wll unplug it, and do a check again to see if issues are the same (if so, then it's not the printeR)

devslashlol, 2020-09-24:
ok,I unplugged the ff entirely and I get this
2020-09-24 08:41:21,080 - octoprint.util.comm - INFO - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2020-09-24 08:41:21,086 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-09-24 08:41:21,153 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 08:41:21,160 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 08:41:21,160 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 08:41:21,160 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 08:41:21,161 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 08:41:21,242 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 08:41:21,249 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 08:41:21,249 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-24 08:41:21,250 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2109, USB ID: 0X3431
2020-09-24 08:41:21,251 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 08:41:21,445 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,446 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,448 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,449 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,451 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,452 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,453 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,454 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,456 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,457 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,459 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,460 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,462 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,463 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,465 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,466 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-24 08:41:21,522 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-24 08:41:21,529 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-24 08:41:21,530 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003

devslashlol, 2020-09-24:
So it seems like it's a bit of a red herring (original stuff)

devslashlol, 2020-09-24:
from FF manual: 
[image]

devslashlol, 2020-09-24:
I am using "13"

devslashlol, 2020-09-24:
perhaps I need to use 12?

Mrnt, 2020-09-24:
Yep - 12 is the usb connection for the printer to act as a slave which in this case it is since the computer/pi is the usb “host”. The other port allows the printer to act as a host to a usb (slave or client) device such as a usb flash drive

devslashlol, 2020-09-24:
Ah ! ok. I ordered a USB-a to B cable. I think we can close this ticket. It might be a good note for your documentation if you feel so inclined. :) thank you for your help, I will re-open if I still have issues.

Mrnt, 2020-09-24:
Sounds good. I wonder whether the usb A port on the printer would deliver enough power to power the pi by connecting it to the pi’s usb C port. Probably dodgy since I think pi 4 takes 700ma or so and regular usb A ports IIRC rated 500ma max

Mrnt, 2020-09-24:
Looks like you got it. I wonder whether the usb A port on the printer would deliver enough power to power the pi by connecting it to the pi’s usb C port

Mrnt, 2020-09-24:
Thanks for the suggestion - I just added a note about usb b.

devslashlol, 2020-09-24:
one last comment - when I added the plugin, I was not able to add a new profile.I had to uninstall the plugin to allow me to add a profile (the 'add profile' button basically stopped working)

devslashlol, 2020-09-24:
(should I open a new issue?)

Mrnt, 2020-09-24:
Yes, looks like there is an issue - something new :( go ahead and create a new issue for it

devslashlol, 2020-09-25:
Hi - ok, got my usb B connected: now I'm getting the following:

FlashForgeError: Found printer but there is a connection problem - reboot printer.
2020-09-25 10:54:34,626 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 0 port/baudrate candidates: 
2020-09-25 10:54:34,627 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
2020-09-25 10:54:34,634 - octoprint.util.comm - INFO - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2020-09-25 10:54:34,638 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-09-25 10:54:34,649 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-25 10:54:34,657 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-25 10:54:34,658 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-25 10:54:34,659 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2B71, USB ID: 0X0007
2020-09-25 10:54:34,660 - octoprint.util.comm - ERROR - Error while retrieving additional serial port names from hook flashforge
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 194, in serialList
    candidates += hook(candidates)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_flashforge/__init__.py", line 171, in get_additional_port_names
    self.detect_printer()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_flashforge/__init__.py", line 132, in detect_printer
    raise flashforge.FlashForgeError("Found printer but there is a connection problem - reboot printer.")
FlashForgeError: Found printer but there is a connection problem - reboot printer.
2020-09-25 10:54:34,979 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,981 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,982 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,984 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,985 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,986 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,988 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,989 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,991 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,992 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,996 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:34,998 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:35,000 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:35,001 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:35,003 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:35,004 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 10:54:35,067 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2020-09-25 10:54:35,093 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0002
2020-09-25 10:54:35,095 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X1D6B, USB ID: 0X0003
2020-09-25 10:54:35,096 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown' with Vendor ID: 0X2B71, USB ID: 0X0007
2020-09-25 10:54:35,097 - octoprint.util.comm - ERROR - Error while retrieving additional serial port names from hook flashforge
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 194, in serialList
    candidates += hook(candidates)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_flashforge/__init__.py", line 171, in get_additional_port_names
    self.detect_printer()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_flashforge/__init__.py", line 132, in detect_printer
    raise flashforge.FlashForgeError("Found printer but there is a connection problem - reboot printer.")
FlashForgeError: Found printer but there is a connection problem - reboot printer.
2020-09-25 10:54:35,205 - octoprint.server.preemptive_cache - INFO - ... done in 9.49s

devslashlol, 2020-09-25:
I reboot the printer, ... restarted octopi/print...

devslashlol, 2020-09-25:
I checked all my settings per the wiki/homepage, and they are set as 
[image]

devslashlol, 2020-09-25:
[image]

Mrnt, 2020-09-25:
Sorry, I think the last version broke the ability to make the initial connection after clean install and hence the lack of feedback on what to do next (enable USB permissions). I'm about to release a fix for that and the printer profile issue you identified.

Thanks for your patience.

devslashlol, 2020-09-25:
no worries! Thanks for being so quick to respond!

Mrnt, 2020-09-25:
I just put up the new version (v0.2.4) if you want to give it a try. In theory OctoPrint will notify you that it is available, but that can sometimes take a few hours. You can just go an reinstall it and I think that will get you the latest version.

devslashlol, 2020-09-25:
ok, trying now. It did install the latest by uninstalling and reinstalling

devslashlol, 2020-09-25:
what is the USB permissions thing? Can I read more about what it is I am doing, or why this is necessary? (I'm just curious)

devslashlol, 2020-09-25:
it worked! Thank you!

devslashlol, 2020-09-25:
Also - I see the finder respond to using the "Upload from SD Card", but it never sets the temperature or actually starts the print - but the finder's display seems to indicate that it is ready to start printing. If you have any insight, not really a bug - just trying to sort it out. :)

Mrnt, 2020-09-25:
The USB permissions thing is that octoprint does not seem to have read/write access to USB devices by default, so you have to explicitly add it (per the instructions that the plugin displays in the terminal window).

> Also - I see the finder respond to using the "Upload from SD Card", but it never sets the temperature or actually starts the print - but the finder's display seems to indicate that it is ready to start printing. If you have any insight, not really a bug - just trying to sort it out. :)

What kind of file are you uploading to the SD card? Is it something sliced with FlashPrint?

Mrnt, 2020-09-25:
> The USB permissions thing is that octoprint does not seem to have read/write access to USB devices by default, so you have to explicitly add it (per the instructions that the plugin displays in the terminal window).

TBH I didn't spend a lot of time looking at why this is, but I'm assuming that only root has read/write access by default and other users (eg octoprint) have to be granted access explicitly. The dev rule grants permission to any user/process on the device.

Mrnt, 2020-10-02:
This should be resolved. Combination of wrong USB connector and bug not allowing for printer detection resolved in 0.2.24

closed 2020-10-02 by Mrnt