Mrnt/OctoPrint-FlashForge #55: 0.2.4 can't process simplify3d gcode file when printing from SD Card

devslashlol, 2020-09-25:
Hi,
I used Upload to SD card, then printed from that - I get these kinds of messages in the log
2020-09-25 14:43:37,492 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): dropping command
2020-09-25 14:43:37,493 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:G1, cmd:G1 X1.482 Y-9.400 E1.5032

The extruder *sounds* like it's grinding or skipping steps or something. I also see this:
2020-09-25 16:23:46,385 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:0 Y-max:0 Z-max:0 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-09-25 16:23:46,387 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:46,389 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:46,391 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M27 Received. | SD printing byte 0/1000 | ok | 
2020-09-25 16:23:46,392 - octoprint.plugins.flashforge - DEBUG - buffering: CMD M27 Received.
2020-09-25 16:23:46,394 - octoprint.plugins.flashforge - DEBUG - buffering: Not SD printing
2020-09-25 16:23:46,395 - octoprint.plugins.flashforge - DEBUG - buffering: ok
2020-09-25 16:23:46,398 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:46,400 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:46,402 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:46,403 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:47,364 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:M27, cmd:M27
2020-09-25 16:23:47,364 - octoprint.plugins.flashforge - DEBUG - is_sd_printing()
2020-09-25 16:23:47,366 - octoprint.plugins.flashforge - DEBUG - write() called by thread comm.sending_thread
2020-09-25 16:23:47,367 - octoprint.plugins.flashforge - DEBUG - write() M119
~M27
2020-09-25 16:23:47,376 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:0 Y-max:0 Z-max:0 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-09-25 16:23:47,377 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:47,380 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:47,382 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M27 Received. | SD printing byte 0/1000 | ok | 
2020-09-25 16:23:47,383 - octoprint.plugins.flashforge - DEBUG - buffering: CMD M27 Received.
2020-09-25 16:23:47,384 - octoprint.plugins.flashforge - DEBUG - buffering: Not SD printing
2020-09-25 16:23:47,384 - octoprint.plugins.flashforge - DEBUG - buffering: ok
2020-09-25 16:23:47,386 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:47,387 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:47,388 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:47,388 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:48,359 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:M105, cmd:M105
2020-09-25 16:23:48,360 - octoprint.plugins.flashforge - DEBUG - is_sd_printing()
2020-09-25 16:23:48,363 - octoprint.plugins.flashforge - DEBUG - write() called by thread comm.sending_thread
2020-09-25 16:23:48,364 - octoprint.plugins.flashforge - DEBUG - write() M105
2020-09-25 16:23:48,367 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:M27, cmd:M27
2020-09-25 16:23:48,369 - octoprint.plugins.flashforge - DEBUG - is_sd_printing()
2020-09-25 16:23:48,374 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M105 Received. | T0:28 /0 B:0/0 | ok | 
2020-09-25 16:23:48,374 - octoprint.plugins.flashforge - DEBUG - buffering: CMD M105 Received.
2020-09-25 16:23:48,376 - octoprint.plugins.flashforge - DEBUG - buffering: T0:28 /0 B:0/0
2020-09-25 16:23:48,377 - octoprint.plugins.flashforge - DEBUG - buffering: ok
2020-09-25 16:23:48,380 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:48,383 - octoprint.util.comm - INFO - Externally triggered heatup detected
2020-09-25 16:23:48,395 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:48,397 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:48,399 - octoprint.plugins.flashforge - DEBUG - write() called by thread comm.sending_thread
2020-09-25 16:23:48,400 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:48,401 - octoprint.plugins.flashforge - DEBUG - write() M119
~M27
2020-09-25 16:23:48,411 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:0 Y-max:0 Z-max:0 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-09-25 16:23:48,412 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:48,414 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:48,417 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M27 Received. | SD printing byte 0/1000 | ok | 
2020-09-25 16:23:48,418 - octoprint.plugins.flashforge - DEBUG - buffering: CMD M27 Received.
2020-09-25 16:23:48,420 - octoprint.plugins.flashforge - DEBUG - buffering: Not SD printing
2020-09-25 16:23:48,421 - octoprint.plugins.flashforge - DEBUG - buffering: ok
2020-09-25 16:23:48,423 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:48,426 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:48,429 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:48,430 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:49,371 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:M27, cmd:M27
2020-09-25 16:23:49,373 - octoprint.plugins.flashforge - DEBUG - is_sd_printing()
2020-09-25 16:23:49,376 - octoprint.plugins.flashforge - DEBUG - write() called by thread comm.sending_thread
2020-09-25 16:23:49,378 - octoprint.plugins.flashforge - DEBUG - write() M119
~M27
2020-09-25 16:23:49,386 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:0 Y-max:0 Z-max:0 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-09-25 16:23:49,388 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:49,389 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
[... 5 lines cut ...]
2020-09-25 16:23:49,408 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:49,411 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:49,413 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:50,375 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:M27, cmd:M27
2020-09-25 16:23:50,376 - octoprint.plugins.flashforge - DEBUG - is_sd_printing()
2020-09-25 16:23:50,378 - octoprint.plugins.flashforge - DEBUG - write() called by thread comm.sending_thread
2020-09-25 16:23:50,380 - octoprint.plugins.flashforge - DEBUG - write() M119
~M27
2020-09-25 16:23:50,389 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:0 Y-max:0 Z-max:0 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-09-25 16:23:50,390 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:23:50,392 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:23:50,395 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M27 Received. | SD printing byte 0/1000 | ok | 
[... cut ...]
2020-09-25 16:23:51,378 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:M27, cmd:M27
2020-09-25 16:23:51,379 - octoprint.plugins.flashforge - DEBUG - is_sd_printing()
2020-09-25 16:23:51,382 - octoprint.plugins.flashforge - DEBUG - write() called by thread comm.sending_thread
2020-09-25 16:23:51,387 - octoprint.plugins.flashforge - DEBUG - write() M119

I use Simplify 3D to slice, and the gcode files directly  out of that always work if I load them via a usb flash drive at the unit directly.

Mrnt, 2020-09-25:
Please upload the file so I can test with it, thanks!

devslashlol, 2020-09-26:
[xyzCalibration_cube.gcode.txt](https://github.com/Mrnt/OctoPrint-FlashForge/files/5285898/xyzCalibration_cube.gcode.txt)

devslashlol, 2020-09-26:
it wouldn't let me upload a plain gcode file.

Mrnt, 2020-09-26:
If you upload to the address card using “Upload to SD” the file should go straight to the SD card and you should just see printing progress type messages in the log which looks like what you have. The plugin does not parse the uploaded file and alter it. It is a bit of a red flag that it says the total bytes it is/was printing is 1000. If you look further back in the log you should see the actual upload happening with  progress in %. Make sure that got to 100%. There will be an M29 command at the end of the upload which should get a non error response.

Mrnt, 2020-09-26:
If you can upload a fresh OctoPrint.log captured during an SD upload that doesn’t print it would be helpful.

devslashlol, 2020-09-26:
I see the files completing and M29 looks ok
[image]

devslashlol, 2020-09-26:
every time I see this in the log, it's completes

devslashlol, 2020-09-26:
I should note, the printer never actually does anything - the hot end doesn't heat up, nothing. Just sits there. 
The display shows what appears to be a print job, but ... just sits.

devslashlol, 2020-09-26:
mo' log
2020-09-25 16:21:33,996 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-09-25 16:21:33,997 - octoprint.plugins.flashforge - DEBUG - Sent: 99.00% 248832/250896
2020-09-25 16:21:33,998 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-09-25 16:21:33,999 - octoprint.plugins.flashforge - DEBUG - Sent: 99.00% 249856/250896
2020-09-25 16:21:34,001 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-09-25 16:21:34,002 - octoprint.plugins.flashforge - DEBUG - Sent: 99.00% 250880/250896
2020-09-25 16:21:34,003 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-09-25 16:21:34,004 - octoprint.plugins.flashforge - DEBUG - Sent: 100.00% 250896/250896
2020-09-25 16:21:34,005 - octoprint.plugins.flashforge - DEBUG - sendcommand() M29
2020-09-25 16:21:34,006 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-09-25 16:21:34,007 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: FlashForge.SD_Uploader, timeout: 10000
2020-09-25 16:21:34,052 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M29 Received. | Done saving file. | ok | 
2020-09-25 16:21:34,052 - octoprint.plugins.flashforge - DEBUG - sendcommand() got an ok
2020-09-25 16:21:34,054 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,062 - octoprint.plugins.flashforge - DEBUG - enable_keep_alive(True)
2020-09-25 16:21:34,063 - octoprint.plugins.flashforge - DEBUG - rewrite_gcode(): gcode:M23, cmd:M23 0:/user/finder_gcode/xyzCalibration_cube.gcode
2020-09-25 16:21:34,063 - octoprint.plugins.flashforge - DEBUG - is_sd_printing()
2020-09-25 16:21:34,064 - octoprint.util.comm - INFO - Externally triggered heatup detected
2020-09-25 16:21:34,068 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:21:34,069 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:21:34,070 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:21:34,071 - octoprint.plugins.flashforge - DEBUG - write() called by thread comm.sending_thread
2020-09-25 16:21:34,072 - octoprint.plugins.flashforge - DEBUG - write() M23 0:/user/finder_gcode/xyzCalibration_cube.gcode
M119
2020-09-25 16:21:34,092 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M23 Received. | File opened: /data/xyzCalibration_cube.gcode Size: 0 | File selected | ok | 
2020-09-25 16:21:34,092 - octoprint.plugins.flashforge - DEBUG - buffering: CMD M23 Received.
2020-09-25 16:21:34,093 - octoprint.plugins.flashforge - DEBUG - buffering: File opened: /data/xyzCalibration_cube.gcode Size: 0
2020-09-25 16:21:34,094 - octoprint.plugins.flashforge - DEBUG - buffering: File selected
2020-09-25 16:21:34,095 - octoprint.plugins.flashforge - DEBUG - buffering: ok
2020-09-25 16:21:34,096 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:21:34,101 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:21:34,103 - octoprint.printer.standard.job - INFO - Print job selected - origin: sdcard, path: 0:/user/finder_gcode/xyzCalibration_cube.gcode
, owner: None, user: None
2020-09-25 16:21:34,109 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:21:34,110 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:21:34,111 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:21:34,179 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,180 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,182 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,183 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,185 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,187 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,189 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,190 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,192 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,193 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,196 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,197 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,271 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,272 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,274 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,276 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,278 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,279 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,281 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-09-25 16:21:34,283 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
[... cut ...]
2020-09-25 16:21:51,929 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-09-25 16:21:51,930 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-09-25 16:21:51,932 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M27 Received. | SD printing byte 0/1000 | ok | 
2020-09-25 16:21:51,933 - octoprint.plugins.flashforge - DEBUG - buffering: CMD M27 Received.
2020-09-25 16:21:51,935 - octoprint.plugins.flashforge - DEBUG - buffering: SD printing byte 0/1000

devslashlol, 2020-09-26:
This might be a silly question, but is this the internal SD card? The finder has two usb ports, so I can only assume that this SD card is internal to the unit...

Mrnt, 2020-09-26:
Yes there's an sd card embedded on the control board of most of these printers so when you connect using FlashPrint and upload a file it goes onto that internal SD card. On the display on the printer you can print files off the internal SD card or sleet the USB drive if You have one plugged in.

devslashlol, 2020-09-27:
[xyzCalibration_cube.gcode.txt](https://github.com/Mrnt/OctoPrint-FlashForge/files/5288437/xyzCalibration_cube.gcode.txt)
[octoprint (10).log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5288438/octoprint.10.log)

here are files from this morning. log file is freshly new and shows the same behavior.

Mrnt, 2020-09-27:
Great, thanks for that. The file upload appears to be successful and the printer responds correctly to the M29 command which makes it look like it saved the file:
`readraw() returns: CMD M29 Received. | Done saving file. | ok |`
However clearly something is not quite right because when the printer is asked to print the file using the M23 command, it responds with:
`readraw() returns: CMD M23 Received. | File opened: /data/xyzCalibration_cube.gcode Size: 0 | File selected | ok | `
whereas my printer (not a Finder v2) responds with:
`readraw() returns: CMD M23 Received. | File opened: 0:/user/xyzCalibration_cube.gcode Size: 250896 | File selected | ok |
`
Which makes me think that the file was in fact not transferred properly or not saved properly on the SD card. It could of course also be a firmware difference unfortunately.

Have you added any plugins besides the FlashForge one?

Mrnt, 2020-09-27:
And you have not filled up the memory on the SD card in the printer?

Mrnt, 2020-09-27:
Looking again at the log it appears you have several plugins installed. Perhaps try disabling all of the ones you added except the FlashForge one and then try uploading to SD again. Depending on the function of the plugin, some plugins will manipulate the file before it is uploaded to the SD card (I believe DisplayLayer does this) so it would be good to eliminate any of them as a possible cause.

devslashlol, 2020-09-28:
The FF memory is certainly not full. however, it does not show the 'xyz.gcode' file at all. I do have a number of plugins installed. I will report back.

devslashlol, 2020-09-28:
as an aside, if I try to stream the gcode to the finder directly, the printer responds, but makes horrible grinding noises and grossly over-extrudes.

devslashlol, 2020-09-28:
seems to be working, although not sure which plugin it might be having issues with. I think the one that presents a 'preheat' button might be it, but that's total conjecture.

devslashlol, 2020-09-28:
I was able to do one print, but after I cancelled it, it didn't seem to work again. Power cycling, and trying again

devslashlol, 2020-09-28:
Ok, strange - xyz.gcode prints, but xy.gcode (a tile holder, much larger file) does not.  I tried re-slicing the same model with what I thought were the same factory settings in S3D, and it .. doesn't seem to work. I do see the xyz.gcode on the onboard flash now. So this is weird. Let me play with it a little bit and see if I can get something repeatable for you to chew on. Thanks for working with me!

devslashlol, 2020-09-28:
...aaand now it's not working at all. :/

devslashlol, 2020-09-28:
not sure if this matters, but the port on the connection seems to be changing - when it connected before, it said 1:3, then 1:7 now 1:8...
[image]

devslashlol, 2020-09-28:
I think I figured it out - if the file is in a folder in octoprint, it doesn't load correctly. If I "Upload to SD" from the root, it works. If I try from a folder (in this case, "finder_test", it does not work. I think before I was changing directories unwittingly: 
i.e. This print never worked:
2020-09-27 19:42:29,160 - octoprint.plugins.flashforge - INFO - Starting SDCard upload from finder_gcode/tiles3.gcode to finder_gcode/tiles3.gcode

this print did work:
2020-09-27 21:05:02,298 - octoprint.plugins.flashforge - INFO - Starting SDCard upload from tiles3.gcode to tiles3.gcode

...likely because there is no folder on the SD card called "finder_gcode" ?

devslashlol, 2020-09-28:
...also, it seems that if you get the file to the printer, then the folder doesn't matter - although it is not clear if it transfers in a new file, or just uses the existing file. Anyway, I think I understand the problem. I went and deleted the xyz.gcode from the printer directly, and tried to load from the finder_tests folder in octoprint, and it did not print, as expected. what's interesting is that the printer display indicates that it has a job, and has the right name - but just never goes anywhere. It does 'home', but then that's it.

devslashlol, 2020-09-28:
[octoprint (16).log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5289901/octoprint.16.log)

my most recent log if it is helpful. this does not have a faulty load, but a successful one.

Mrnt, 2020-09-28:
Oh yes the file path is totally screwed up. On my printers the upload path should always be `/user/` eg `/user/xyzCalibration.gcode`. It looks like the Finder 2 is expecting the upload paths to be `/data/` eg `/data/xyzCalibration.gcode`. I don't quite understand yet how we ended up with subfolders in the path name eg `/user/finder_gcode/xyzCalibration_cube.gcode`.

Mrnt, 2020-09-28:
Not sure whether it is relevant - but what web browser are you using for the 'Upload to SD'?

Mrnt, 2020-09-28:
> not sure if this matters, but the port on the connection seems to be changing - when it connected before, it said 1:3, then 1:7 now 1:8...
image

This will happen if you reboot the printer which you sometimes have to do if the connection isn't closed properly - eg for some reason the plugin did not get all of the response from the printer before the connection was closed. It reconnects on the Pi/computer and ends up taking a new port number (I'm guessing the USB stack is not sure if the previous port is still in use).

Mrnt, 2020-09-28:
I think I will do a special version for you to try, but please tell me how you are able to get a subfolder name into the filename when using 'Upload to SD' because I cannot reproduce it using Safari, Chrome or Firefox (I don't have access to a Window box right now).

devslashlol, 2020-09-28:
Shoot, I'm sorry - I should have shown those steps. I'm just making a folder in Octoprint here: 
[image]

devslashlol, 2020-09-28:
[image]

devslashlol, 2020-09-28:
also, am using chrome

Mrnt, 2020-09-28:
It looks like OctoPrint prepends the local folder name to the filename of the file being uploaded. 

Ok I made some changes to the dev branch to address that and the fact that the Finder 2 seems to use "/data" as the target path on the SD card (instead of "/user" . You can force OctoPrint to install the dev branch - go to Settings > Plugin Manager > click "Get More" > in the box "... from URL" enter: https://github.com/Mrnt/OctoPrint-FlashForge/archive/devel.zip
and it should download the latest version of the devel branch. Restart OctoPrint before testing.

devslashlol, 2020-09-30:
Hi, sorry I haven't tried this yet with the development version. Since I got the main one to work, I've been using that (but see the other issue I just opened re: not being able to reconnect after the print is completed. also, the serial connection seems to stall as the print proceeds... just somewhere in the middle, it says 20/1000 bytes (no matter how many bytes it actually is) and then never communicates again.

devslashlol, 2020-10-01:
ok, trying it now

devslashlol, 2020-10-01:
ok, now nothing streams correctly (even from root)

devslashlol, 2020-10-01:
I had a bit of a snafu when starting, in that the M21 command from the machine was sitting there for a while, and maybe I confused it. I will power cycle and try again

devslashlol, 2020-10-01:
2020-10-01 07:22:29,344 - octoprint.filemanager.analysis - INFO - Starting analysis of local:finder_gcode/xyz.gcode
2020-10-01 07:22:29,349 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,352 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=4800 --speed-y=4800 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/finder_gcode/xyz.gcode
2020-10-01 07:22:29,396 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,405 - octoprint.plugins.flashforge - INFO - Starting SDCard upload from finder_gcode/xyz.gcode to data/xyz.gcode
2020-10-01 07:22:29,519 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,522 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,524 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,526 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,530 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,533 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,535 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,539 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,541 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,545 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,546 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,549 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,550 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,551 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,553 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,555 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,557 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,559 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,560 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,562 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,564 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,566 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:29,569 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:30,396 - octoprint.plugins.flashforge - DEBUG - write() called by thread FlashForge.Keep_Alive
2020-10-01 07:22:30,396 - octoprint.plugins.flashforge - DEBUG - write() M119
2020-10-01 07:22:30,398 - octoprint.plugins.flashforge - DEBUG - readraw() TIMEOUT
2020-10-01 07:22:30,399 - octoprint.plugins.flashforge - DEBUG - readraw() returns: 
2020-10-01 07:22:30,400 - octoprint.plugins.flashforge - DEBUG - enable_keep_alive(False)
2020-10-01 07:22:30,400 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-10-01 07:22:30,403 - octoprint.plugins.flashforge - DEBUG - sendcommand() M104 S0 T0
2020-10-01 07:22:30,405 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-10-01 07:22:30,406 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: FlashForge.SD_Uploader, timeout: 2000
2020-10-01 07:22:30,407 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:1 Y-max:1 Z-max:1 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-10-01 07:22:30,409 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: FlashForge.SD_Uploader, timeout: 2000
2020-10-01 07:22:30,417 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M104 Received. | ok | 
2020-10-01 07:22:30,417 - octoprint.plugins.flashforge - DEBUG - sendcommand() got an ok
2020-10-01 07:22:30,418 - octoprint.plugins.flashforge - DEBUG - sendcommand() M104 S0 T1
2020-10-01 07:22:30,418 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-10-01 07:22:30,419 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: FlashForge.SD_Uploader, timeout: 2000
2020-10-01 07:22:30,432 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M104 Received. | ok | 
2020-10-01 07:22:30,432 - octoprint.plugins.flashforge - DEBUG - sendcommand() got an ok
2020-10-01 07:22:30,433 - octoprint.plugins.flashforge - DEBUG - sendcommand() M140 S0
2020-10-01 07:22:30,433 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-10-01 07:22:30,435 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: FlashForge.SD_Uploader, timeout: 2000
2020-10-01 07:22:30,446 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M140 Received. | ok | 
2020-10-01 07:22:30,446 - octoprint.plugins.flashforge - DEBUG - sendcommand() got an ok
2020-10-01 07:22:30,447 - octoprint.plugins.flashforge - DEBUG - sendcommand() M28 250929 0:/data/xyz.gcode
2020-10-01 07:22:30,448 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
[... cut ...]
2020-10-01 07:22:31,153 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:31,155 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:31,158 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:31,160 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:31,161 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 07:22:31,163 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()

devslashlol, 2020-10-01:
I don't think it likes the /data bit.

devslashlol, 2020-10-01:
connection details: 
Performing autodetection with 7 port/baudrate candidates: FlashForge Finder 3D Printer, port:1:26@115200, FlashForge Finder 3D Printer, port:1:26@250000, FlashForge Finder 3D Printer, port:1:26@230400, FlashForge Finder 3D Printer, port:1:26@57600, FlashForge Finder 3D Printer, port:1:26@38400, FlashForge Finder 3D Printer, port:1:26@19200, FlashForge Finder 3D Printer, port:1:26@9600
Trying port FlashForge Finder 3D Printer, port:1:26, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0xab482070>, starting monitor
Send: M601 S0
Recv: CMD M601 Received.
Recv: Control Success.
Recv: ok
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: M115
Recv: CMD M115 Received.
Recv: Machine Type: Flashforge Finder
Recv: Machine Name: My 3D Printer
Recv: FIRMWARE_NAME: FlashForge VER: v2.7.2.366 F2.12 20191224
Recv: SN: SNFFF226299
Recv: X: 140 Y: 140 Z: 140
Recv: Tool Count:1
Recv: ok
Send: M21
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M20
Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M146 r255 g255 b255
Recv: CMD M146 Received.
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:26 /0 B:0/0
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:26 /0 B:0/0
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:26 /0 B:0/0
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:26 /0 B:0/0
Recv: ok
Send: M105
Recv: CMD M105 Received.
Recv: T0:26 /0 B:0/0
Recv: ok

I noticed it has a timeout up front, which seems weird. Timeouts are all specified as in the config page (5,2,2 etc)

Mrnt, 2020-10-01:
re M21 - I just looked again log file you uploaded and it seems the printer never responded to the M21 command at connect. That would cause the OctoPrint to wait for a response and eventually generate a timeout at which point it sends an M105 (get temperature) command to see if the printer is still there. I cannot tell if the command does anything on any of the FlashForge printers so maybe the answer is to just stop OctoPrint from sending it. Probably likewise for the M20 command which the printer is supposed to reply with the list of files on the SD card. The older printers at least respond to the command (even though they don't provide a list of files), so no timeout occurs.

re SD card uploads - with the dev version does it never start the upload process? Does the printer not respond to the sendcommand() M28 250929 0:/data/xyz.gcode?

Mrnt, 2020-10-01:
I just pushed an update to the dev branch to eliminate the M20, M21 commands. Please reinstall from the dev URL (as described above) - that should eliminate the communication timeout messages immediately after connecting. Can't tell yet whether that has an impact on the SD Upload issue.

devslashlol, 2020-10-01:
Ok, that seemed to resolve the timeouts (m20/21), but the streaming still isn't working. It has 'broken' between the 'stable' 2.4 and the dev 2.4... so I can't print anything at all at least Upload from SD-wise

devslashlol, 2020-10-01:
Here is what the log says:
2020-10-01 15:00:14,593 - octoprint.plugins.flashforge - DEBUG - is_sd_printing()
2020-10-01 15:00:14,597 - octoprint.plugins.flashforge - DEBUG - write() called by thread comm.sending_thread
2020-10-01 15:00:14,601 - octoprint.plugins.flashforge - DEBUG - write() M105
2020-10-01 15:00:14,612 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M105 Received. | T0:48 /0 B:0/0 | ok | 
2020-10-01 15:00:14,612 - octoprint.plugins.flashforge - DEBUG - buffering: CMD M105 Received.
2020-10-01 15:00:14,613 - octoprint.plugins.flashforge - DEBUG - buffering: T0:48 /0 B:0/0
2020-10-01 15:00:14,615 - octoprint.plugins.flashforge - DEBUG - buffering: ok
2020-10-01 15:00:14,618 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-10-01 15:00:14,620 - octoprint.util.comm - INFO - Externally triggered heatup detected
2020-10-01 15:00:14,623 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-10-01 15:00:14,626 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-10-01 15:00:14,628 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-10-01 15:00:14,695 - octoprint.plugins.flashforge - DEBUG - write() called by thread FlashForge.Keep_Alive
2020-10-01 15:00:14,696 - octoprint.plugins.flashforge - DEBUG - write() M119
2020-10-01 15:00:14,705 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:1 Y-max:0 Z-max:1 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-10-01 15:00:14,706 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-10-01 15:00:14,708 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-10-01 15:00:16,698 - octoprint.plugins.flashforge - DEBUG - write() called by thread FlashForge.Keep_Alive
2020-10-01 15:00:16,699 - octoprint.plugins.flashforge - DEBUG - write() M119
2020-10-01 15:00:16,708 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:1 Y-max:0 Z-max:1 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-10-01 15:00:16,710 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-10-01 15:00:16,710 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: comm._monitor, timeout: 2000
2020-10-01 15:00:16,844 - tornado.application - ERROR - Uncaught exception GET /downloads/logs/octoprint.log (::ffff:192.168.1.173)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 1592, in _execute
    result = yield result
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1133, in run
    value = future.result()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/concurrent.py", line 261, in result
    raise_exc_info(self._exc_info)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/gen.py", line 1147, in run
    yielded = self.gen.send(value)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2528, in get
    yield self.flush()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 1000, in flush
    return self.request.connection.write(chunk, callback=callback)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 451, in write
    self._pending_write = self.stream.write(self._format_chunk(chunk))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 424, in _format_chunk
    "Tried to write more data than Content-Length")
HTTPOutputError: Tried to write more data than Content-Length
2020-10-01 15:00:16,846 - tornado.general - ERROR - Cannot send error response after headers written

Mrnt, 2020-10-01:
Not sure what is going on there - did some of the log not paste in correctly? We should at least see the M28 command being issued.

devslashlol, 2020-10-01:
more log:
2020-10-01 14:58:46,962 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:46,967 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:46,976 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:46,982 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:46,985 - octoprint.filemanager.analysis - INFO - Starting analysis of local:finder_gcode/xyz.gcode
2020-10-01 14:58:46,987 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:46,989 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=4800 --speed-y=4800 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/finder_gcode/xyz.gcode
2020-10-01 14:58:46,996 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,012 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,048 - octoprint.plugins.flashforge - INFO - Starting SDCard upload from finder_gcode/xyz.gcode to data/xyz.gcode
2020-10-01 14:58:47,093 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,096 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,098 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,100 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,104 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,106 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,108 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,111 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,114 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,117 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,118 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,120 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,122 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,124 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,126 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,128 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,129 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,131 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,133 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,136 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,138 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,139 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:47,142 - octoprint.plugins.flashforge - DEBUG - get_extension_tree()
2020-10-01 14:58:48,537 - octoprint.plugins.flashforge - DEBUG - write() called by thread FlashForge.Keep_Alive
2020-10-01 14:58:48,538 - octoprint.plugins.flashforge - DEBUG - write() M119
2020-10-01 14:58:48,547 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M119 Received. | Endstop: X-max:1 Y-max:0 Z-max:1 | MachineStatus: READY | MoveMode: READY | Status: S:1 L:0 J:0 F:0 | ok | 
2020-10-01 14:58:48,548 - octoprint.plugins.flashforge - DEBUG - readline() called by thread comm._monitor
2020-10-01 14:58:48,548 - octoprint.plugins.flashforge - DEBUG - enable_keep_alive(False)
2020-10-01 14:58:48,552 - octoprint.plugins.flashforge - DEBUG - sendcommand() M104 S0 T0
2020-10-01 14:58:48,554 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-10-01 14:58:48,555 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: FlashForge.SD_Uploader, timeout: 2000
2020-10-01 14:58:48,564 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M104 Received. | ok | 
2020-10-01 14:58:48,564 - octoprint.plugins.flashforge - DEBUG - sendcommand() got an ok
2020-10-01 14:58:48,564 - octoprint.plugins.flashforge - DEBUG - sendcommand() M104 S0 T1
2020-10-01 14:58:48,566 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-10-01 14:58:48,567 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: FlashForge.SD_Uploader, timeout: 2000
2020-10-01 14:58:48,577 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M104 Received. | ok | 
2020-10-01 14:58:48,577 - octoprint.plugins.flashforge - DEBUG - sendcommand() got an ok
2020-10-01 14:58:48,578 - octoprint.plugins.flashforge - DEBUG - sendcommand() M140 S0
2020-10-01 14:58:48,579 - octoprint.plugins.flashforge - DEBUG - writeraw() called by thread FlashForge.SD_Uploader
2020-10-01 14:58:48,580 - octoprint.plugins.flashforge - DEBUG - readraw() called by thread: FlashForge.SD_Uploader, timeout: 2000
2020-10-01 14:58:48,590 - octoprint.plugins.flashforge - DEBUG - readraw() returns: CMD M140 Received. | ok | 
2020-10-01 14:58:48,590 - octoprint.plugins.flashforge - DEBUG - sendcommand() got an ok
2020-10-01 14:58:48,591 - octoprint.plugins.flashforge - DEBUG - sendcommand() M28 250929 0:/data/xyz.gcode
[... cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 451, in write
    self._pending_write = self.stream.write(self._format_chunk(chunk))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 424, in _format_chunk
    "Tried to write more data than Content-Length")
HTTPOutputError: Tried to write more data than Content-Length
2020-10-01 15:00:16,846 - tornado.general - ERROR - Cannot send error response after headers written

devslashlol, 2020-10-01:
the shmutz at the end might not be your plugin

Mrnt, 2020-10-01:
Just changed it back to using /user/ as the upload path if you want to try reinstalling dev

devslashlol, 2020-10-01:
ok, doing so.

devslashlol, 2020-10-01:
ok, that fixed the Upload to SD thing. and the timeouts are gone. Let's see if the print carries through.. it usually stops reporting/auto reporting about 24 'bytes' in.. and it never reports the actual number of bytes.. it's always x/1000

devslashlol, 2020-10-01:
any tinkering with timeouts? the auto reports seem to work fine if there is no print job going, but once the print job gets going, the reports seem ... slow.

Mrnt, 2020-10-01:
Do you mean the print status when printing from the SD card? That looks to be set at a default of 1s using the "SD status interval (polling)" setting.

devslashlol, 2020-10-01:
Yes, I'm just saying that it stops auto updating after a while. In the case of the print I am running right now, the update stopped happening before the temp even got to 200C (the print is well under way at this point)
[image]

devslashlol, 2020-10-01:
[image]

devslashlol, 2020-10-01:
I hit the 'print' button after it appeared to have stalled out to see if it would unstick, but to no avail

Mrnt, 2020-10-01:
Did the printer actually stop or is it OctoPrint no longer getting updates?

devslashlol, 2020-10-01:
Octoprint is no longer getting updates. The printer goes along happily once I do the Upload to SD card (with your latest refactor of the directory)

Mrnt, 2020-10-01:
Can you upload the complete log that you have right now so I can see the whole context?

devslashlol, 2020-10-01:
[octoprint (6).log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5315252/octoprint.6.log)

Mrnt, 2020-10-02:
The printer seems to stop responding but the weird thing is the plugin does not generate a timeout waiting for a response from the printer.

[6 more replies]