Mrnt/OctoPrint-FlashForge #58: Can't upload to SDCard anymore.

reuter, 2020-10-15:
Updated to the latest and now I can't upload to SDCard.
Tried from Cura as well as direct in the Octoprint interface.

Flashforge Finder V2

[octoprint (3).log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5381842/octoprint.3.log)

Mrnt, 2020-10-15:
Sorry!! Thanks for the log - I think I see what happened. In trying to add error checking with user messages to the Upload to SD I made the testing too aggressive and it failed. Not sure why it worked ok for me. 

I also found an issue when uploading to SD card from Cura where it sends an M23 command with an incorrect path.

Just uploaded version 0.2.26 which should address these two issues.

reuter, 2020-10-16:
Tested and working again. Thanks.

closed 2020-10-16 by reuter