Mrnt/OctoPrint-FlashForge #61: Direct Printing Not working

reuter, 2020-10-29:
Trying to print direct rather than upload to SD Card.

It warms the nozzle and some filament feeds out. The bed never moves. The print head does move a little.

The terminal output shows a lot of 
"Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves."

and after cancelling the job, I have to manually shut off the Nozzle temp.

FlashForge Finder V2
Attached Logs.
[octoprint (1).log](https://github.com/Mrnt/OctoPrint-FlashForge/files/5455734/octoprint.1.log)

Mrnt, 2020-10-30:
It looks like there are a couple of things going on:

1. the M109 command (which gets converted to M6 "wait for hot end temperature") is not returning "ok" because it will only do that when the extruder gets to temp, so after 30s OctoPrint thinks the connection is stalled and generates a M105 then merrily carries on without waiting for the extruder to finish warming up. Maybe try changing "Communication timeout" to however many seconds it takes for the hot end to get to temperature from room temperature and see if that helps.

2. the slicer generated G0 (instead of G1) for the move commands and it looks like the printer is not returning anything (causing OctoPrint to generate a timeout) but that may be because the printer is being asked to move when it hasn't finished waiting for the extruder to heat up.

Please test if the G0 command is supported in the Terminal tab by entering something like `G90` and then `G0 X0` which should make the head move left from home to the center of the plate

reuter, 2020-10-30:
I had a water leak in the basement where the printer is. So, right now I have a bunch of fans blowing in there. I'll test this on Sunday night when I can get back in and it's not a wind tunnel.

Mrnt, 2020-10-30:
You know I put in a test for when the printer is waiting for the hot end based on the status it returns - on the older Finder it reports "WAIT_ON_TOOL" to indicate that it is waiting for the extruder to heat up. I have the plugin poll it for temperature while it is waiting and return the data to OctoPrint so that OctoPrint sees something happening. It looks like Finder V2 may always report "MachineStatus: READY | MoveMode: READY" which is not very helpful...

Oxize, 2021-05-16:
@Mrnt i though that the Finder Series didnt work on direct printing, and that we always had to use Print to SD. Something changed ? I got a Bresser Saurus which is inside an FlashForge Finder II. If i start a print, it warms up and the heads start moving, but doesnt really know where he is, and bangs into de site of the heatbed.