Mrnt/OctoPrint-FlashForge #62: Various errors when trying to print / connect

zenixnet, 2020-11-06:
Started having this issue today. I've rebooted everything so many times.

The most common errors are:
```
Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)

There was an error while trying to start a print job. Error: Unable to upload to SD card - printer is busy.
Offline (Error: FlashForgeError: 'USB Error readraw() (LIBUSB_ERROR_PIPE [-9])' @ comm.py:_readline:2916)

State: Offline (Error: FlashForgeError: 'USB Error write() (LIBUSB_ERROR_TIMEOUT [-7])' @ comm.py:_do_send_without_checksum:3593)
```
I turned the debug on for the plugin, cleared my logs, and started again. Here are all my logs: [Logs.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/5502788/Logs.zip)

Mrnt, 2020-11-10:
Sorry - just noticed I made a typo in the release number which is why it never seemed to actually update to the latest release.

However, that does not explain the errors you are seeing.

Looking through the log I spotted:
```
CMD M28 Received. | out of space | ok | 
```
So it looks like the SD card may be full? So after uploading to the printer, the printer starts responding with endless `unknown command: ""` messages presumably because it tried to open a file that was partially uploaded - not really sure.

Try making some space on the printer SD card by deleting some files using the printer's touch screen and see if that resolves the problem. Unfortunately there is no way to list and delete files on the printer's SD card via the USB port using OctoPrint - the firmware just does not seem to support it.

zenixnet, 2020-11-12:
I still had more than half space available for the built in storage, and my sd card only had a few files. Deleted a few files from the internal and formatted the SD and haven't had an issue yet.

Mrnt, 2020-12-16:
@zenixnet if it is working for you can we close this issue now?

zenixnet, 2020-12-16:
Opps, my bad