Mrnt/OctoPrint-FlashForge #64: Cannot connect to Dreamer NX single extruder

Kellimz92, 2020-12-16:
I have a Raspbery Pi 4 4GB with the OctiPi image to run octoprint. When I attempted to connect the printer it seems to connect briefly then is disconnect and never reconnects. I have attached some output and logs along with printer info.

I did see the permissions issue and followed instructions to get that working but there still seems to be an access error in the logs.

Thanks in advance!
[OctoPrint_flashforge_logs_KZ.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/5702893/OctoPrint_flashforge_logs_KZ.zip)

Mrnt, 2020-12-16:
It looks like you set up the permissions correctly because the plugin was able to open the USB connection to the printer and detect the communication endpoints:

```
2020-12-16 13:06:29,179 - octoprint.plugins.flashforge - DEBUG - __init__()
2020-12-16 13:06:29,181 - octoprint.plugins.flashforge - DEBUG - claimed USB interface
2020-12-16 13:06:29,182 - octoprint.plugins.flashforge - DEBUG -  setting number: 0x00, class: 0xff, subclass: 0xff, protocol: 0xff, #endpoints: 4
2020-12-16 13:06:29,183 - octoprint.plugins.flashforge - DEBUG -   found endpoint type LIBUSB_TRANSFER_TYPE_BULK at address 0x01, max packet size 64
2020-12-16 13:06:29,184 - octoprint.plugins.flashforge - DEBUG -   found endpoint type LIBUSB_TRANSFER_TYPE_BULK at address 0x81, max packet size 64
2020-12-16 13:06:29,185 - octoprint.plugins.flashforge - DEBUG -   found endpoint type LIBUSB_TRANSFER_TYPE_BULK at address 0x83, max packet size 64
2020-12-16 13:06:29,186 - octoprint.plugins.flashforge - DEBUG -   found endpoint type LIBUSB_TRANSFER_TYPE_BULK at address 0x03, max packet size 64
2020-12-16 13:06:29,186 - octoprint.plugins.flashforge - DEBUG -   cmd_endpoint_out 0x01, cmd_endpoint_in 0x81
2020-12-16 13:06:29,187 - octoprint.plugins.flashforge - DEBUG -   sd_endpoint_out 0x03, sd_endpoint_in 0x83
```

It then starts sending the "hello" message:

```
2020-12-16 13:06:29,202 - octoprint.plugins.flashforge - DEBUG - write() M601 S0
~M119
```

but nothing is coming back from the printer within 2 seconds:

```
2020-12-16 13:06:31,205 - octoprint.plugins.flashforge - DEBUG - readraw() TIMEOUT
2020-12-16 13:06:31,206 - octoprint.plugins.flashforge - DEBUG - readraw() returns:
```

and so on for subsequent attempts.

Can you try rebooting the Pi (to clear out any USB issues) and the printer.

If that fails you may need to sniff the communication between the printer and FlashPrint on your computer to see if there is something special that happens when connecting to the Dreamer NX.

Mrnt, 2020-12-16:
Just had another thought since it appears other people are able to connect to the Dreamer NX - make sure that the printer is not connected via WiFi at the same time to something else and that the display on the front of the printer is on its "home screen"

[image]

KeltE, 2020-12-16:
Is there any way I can help?
I have a working Dreamer Nx with Octoprint RPI 4 on it.

Kellimz92, 2020-12-16:
This may be it, I did have Wifi enabled and connected. Once the current print is done I will diable it. Thank you for the info.

Kellimz92, 2020-12-16:
Do you have wifi on at the same time?

KeltE, 2020-12-16:
I just tried and also worked with Wifi running.

Kellimz92, 2020-12-16:
Interesting, I just turned Wifi off and it connected instantly. I was always on the host screen so I am not sure that was it.

KeltE, 2020-12-16:
I couldn't just connect when Flashprint was connected to Wifi. Otherwise, the printer's Wifi may be turned on.

Kellimz92, 2020-12-16:
One last question for anyone with a dreamer, what are your profile settings. I really struggled with this one but I think I have them right. The Axes tab is the one I am most concerned about. Thanks in advance!!

Mrnt, 2020-12-16:
Thanks Egon for helping!

> I couldn't just connect when Flashprint was connected to Wifi. Otherwise, the printer's Wifi may be turned on.

That makes sense because the printer can only communicate with one host at a time, so if FlashPrint is already connected via WiFi then the printer will be unable to respond to USB communications.

KeltE, 2020-12-16:
[image]

closed 2021-12-07 by Mrnt