Mrnt/OctoPrint-FlashForge #65: Flash-forge dreamer (dual extruder) issues connecting!

Gamereric21, 2021-01-09:
-Error connecting to printer
There was an error while trying to connect to your printer. Error: Connection error, see Terminal tab

Terminal:

`<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

' @ comm.py:_open_serial:2784
Changing monitoring state from "Offline" to "Opening serial connection"
Changing monitoring state from "Opening serial connection" to "Error: Connection error, see Terminal tab"
Changing monitoring state from "Error: Connection error, see Terminal tab" to "Offline (Error: Connection error, see Terminal tab)"
Unexpected error while connecting to serial port unknown device, port:1:5, baudrate 38400 from hook flashforge: FlashForgeError: '
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

Unable to connect to FlashForge printer - permission error.

If you are using OctoPi/Linux add permission to access this device by editing file:
 /etc/udev/rules.d/99-octoprint.rules

and adding the line:
SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"

You can do this as follows:
1) Connect to your OctoPi/Octoprint device using ssh
2) Type the following to open a text editor:
sudo nano /etc/udev/rules.d/99-octoprint.rules
3) Add the following line:
SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"
4) Save the file and close the editor
5) Verify the file permissions are set to "rw-r--r--" by typing:
ls -al /etc/udev/rules.d/99-octoprint.rules
6) Reboot your system for the rule to take effect.

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

' @ comm.py:_open_serial:2784`

I ssh'ed into it via terminal, added that rule, and double checked that the permissions are set to rw-r--r--, and for some reason it still doesn't want to connect.

The dreamer is not on wifi, connected to the raspberry pi with the usb cable that came with it, and its on the home screen!

Can you help me please lol!

Mrnt, 2021-01-09:
It sounds like the Pi is correctly connected or you would not get the message you are seeing. 

The message is generated before the Pi has got as far as connecting to the Dreamer - at this point the plug-in is just trying to access the USB port to make a connection to printer and is prevented by default OS settings.

Although it sounds like you did this, can you verify:
- You checked the rule file you made is in the right place
- The file has the correct contents, no typos
- The permissions on the file are as specified
- You rebooted the Pi after making the file and setting its permissions

If the above is all true, are you still seeing the message posted above? If so then it sounds like the file is not being read for some reason.

Can you also indicate your Pi setup:
-  Are you using the latest stock octopi image?
-  Anything else installed on the Pi?
-  Pi model?

If you are still getting the above issue - then  can you upload the rule file.

Gamereric21, 2021-01-09:
Im using the 64 bit version as someone suggested I use it (internet issues, it apparently helped them fix them!)

I have a few extensions installed, but otherwise, no.

Model 3B!

if when you check it, it says -rw-r--r--, does that mean it was inputted correctly?  Im also not sure how to send it.  Do you want a screenshot of the file in termial?

Mrnt, 2021-01-10:
You can upload the file here, though you would need to fetch it from the Pi with sftp. If you can’t figure out how to do that just copy it’s contents from your ssh client and paste them here.

Gamereric21, 2021-01-10:
Sorry about the delay in responding, I was out away from my desk.  

> pi@octopi:~ $ sudo nano /etc/udev/rules.d/99-octoprint.rules
[sudo] password for pi: 

  GNU nano 3.2          /etc/udev/rules.d/99-octoprint.rules                    

SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666^x^X

                                [ Read 1 line ]
^G Get Help  ^O Write Out ^W Where Is  ^K Cut Text  ^J Justify   ^C Cur Pos
^X Exit      ^R Read File ^\ Replace   ^U Uncut Text^T To Spell  ^_ Go To Line

I saved the output as a file, and copied the info necessary!  You can see the commands to access the file, the file info, and the options given!

> pi@octopi:~ $ ls -al /etc/udev/rules.d/99-octoprint.rules
-rw-r--r-- 1 root root 57 Jan  9 01:02 /etc/udev/rules.d/99-octoprint.rules
pi@octopi:~ $ 

This is what happens when I try the tester command!  As you can see it is -rw-r--r--, which I believe is correct

Mrnt, 2021-01-10:
does it actually look like:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"
```
or do you have the ^x^X at the end like what you pasted above?

Gamereric21, 2021-01-10:
Yes, it had the x^X^, idk why lol

I removed it, and im going to try it again!

Gamereric21, 2021-01-10:
I have no clue what I did, but as of a few hours ago, im not seeing the errors popping up >

State: Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)

Terminal is still saying this for some reason:

Unable to connect to FlashForge printer - permission error.

If you are using OctoPi/Linux add permission to access this device by editing file:
 /etc/udev/rules.d/99-octoprint.rules

and adding the line:
SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"

You can do this as follows:
1) Connect to your OctoPi/Octoprint device using ssh
2) Type the following to open a text editor:
sudo nano /etc/udev/rules.d/99-octoprint.rules
3) Add the following line:
SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"
4) Save the file and close the editor
5) Verify the file permissions are set to "rw-r--r--" by typing:
ls -al /etc/udev/rules.d/99-octoprint.rules
6) Reboot your system for the rule to take effect.

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

' @ comm.py:_open_serial:2784
Could not open port unknown device, port:1:4, baudrate 19200, skipping
Trying port unknown device, port:1:4, baudrate 9600
Unexpected error while connecting to serial port unknown device, port:1:4, baudrate 9600 from hook flashforge: FlashForgeError: '
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

Unable to connect to FlashForge printer - permission error.

If you are using OctoPi/Linux add permission to access this device by editing file:
 /etc/udev/rules.d/99-octoprint.rules

and adding the line:
SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"

You can do this as follows:
1) Connect to your OctoPi/Octoprint device using ssh
2) Type the following to open a text editor:
sudo nano /etc/udev/rules.d/99-octoprint.rules
3) Add the following line:
SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"
4) Save the file and close the editor
5) Verify the file permissions are set to "rw-r--r--" by typing:
ls -al /etc/udev/rules.d/99-octoprint.rules
6) Reboot your system for the rule to take effect.

<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

' @ comm.py:_open_serial:2784
Could not open port unknown device, port:1:4, baudrate 9600, skipping
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"

This is what I updated the rule to:

 GNU nano 3.2          /etc/udev/rules.d/99-octoprint.rules                    

**SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666**

                                [ Read 1 line ]
^G Get Help  ^O Write Out ^W Where Is  ^K Cut Text  ^J Justify   ^C Cur Pos
^X Exit      ^R Read File ^\ Replace   ^U Uncut Text^T To Spell  ^_ Go To Line

Hope this helps!

sorry about the delay, I had to go away from my computer again lol

Mrnt, 2021-01-10:
You need that last quote mark after 666 ie:
```
SUBSYSTEM=="usb", ATTR{idVendor}=="2b71", MODE="0666"
```
do you have that?

Gamereric21, 2021-01-10:
I must have deleted that with the x^X^

Back to the drawing (cough: ssh) board!

Gamereric21, 2021-01-10:
This is what terminal says now.  I have auto enabled on both!

Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 0 port/baudrate candidates: 
Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."

Any idea on what to do from here?

Gamereric21, 2021-01-10:
LMAO I forgot to turn the printer on this time!

IT SAYS ITS CONNECTED!!!

Gamereric21, 2021-01-10:
It looks like I'm printing ok!  I had to modify the communication error time, and set it to 30 seconds, because for some reason it kept disconnecting, and then restarting, but otherwise its doing amazing so far!

I tested it, and all the movement controls work great, though the Z axis needs to be flipped on my end.  Gotta fix that soon!

Printing a camera mount as a test!

Mrnt, 2021-01-10:
Awesome! You can invert the Z axis in the settings by going to Settings>Printer Profiles:
either create a printer profile for the Dreamer or if you only will use the Pi with the Dreamer then edit the "default" profile, select the "Axes" tab and click the checkbox "invert control" next to "Z" and don't forget to click "Confirm".

Gamereric21, 2021-01-10:
For some reason whenever I print to SD, it makes a weird clicking noise, and in the mid part of the print, its always stringy.  Im uploading a GX file from FlashPrint, so I have no clue why when I print from a local SD card, it prints a ton better than octoprint

https://www.reddit.com/r/3Dprinting/comments/ku5ku6/dreamer_printbed_making_annoying_noise_octoprint/?utm_source=share&utm_medium=web2x&context=3, I recoded, and posted it here on reddit!

lmk what you think is happening, and if there are any layer / movement settings I should try changing!

Mrnt, 2021-01-10:
Just to verify - are you sending your prints to OctoPrint using the "Upload" button (and then selecting the file in OctoPrint and printing, or are you using the "Upload to SD" button?

Gamereric21, 2021-01-10:
Upload to sd I believe!

Sent by a real poopypie

Mrnt, 2021-01-11:
"Upload to SD" should be just the same as sending the file from FlashPrint to the printer - are you creating the file in FlashPrint or something else?

Gamereric21, 2021-01-11:
Im double checking I did that, and running a print again!  Ill check back if it has the same issues!

Is there any way to run octolapse on this, since you can't move the printer mid print

- I assume its injecting G-code to the printer, but flashforge isn't an option!

Mrnt, 2021-01-11:
I have not used OctoLapse, but I assume as you said it injects gcode to move the print head out of the way before taking a picture. In which case it will only work if you use the "Upload" button, which uploads the file to OctoPrint which will pass the file to the printer line by line via the FlashForge plugin. I call this "direct printing" from OctoPrint vs uploading to the SD card via OctoPrint, and if you look at the README I note that this is a relatively new feature that is not entirely stable. I have had pretty good luck with it on the Finder and PowerSpec but please note its not 100% reliable and results for other printers  are very variable.

Gamereric21, 2021-01-11:
I have a few files that I pressed upload to sd card.  If I pressed the print button near the left edge, is that "direct printing", or print from sd card?

Also, there is alot of info that I have to manually put in, and I dont know where id find alot of that info.  Do you know of a spec sheet that has like retract info, and the x/y/z coordinates for home?

Mrnt, 2021-01-12:
The left button uploads the files to the OctoPi's memory (ie the flash card on the Raspberry Pi), and they will then appear in the list of files in the OctoPi ui. When you print any of them via the OctoPi ui you are printing DIRECTLY from OctoPi, command by command. This is the part that is not terribly robust because of the lack of support by FlashForge and it does not appear to work on some newer printers though the Dreamer should be ok.

When you use the right button you are uploading directly to the SD card in the printer and you only have the ability to start, pause, cancel the print via the OctoPi ui. You will not be able to go to the GCode Viewer tab and see progress or use any OctoPi plugins that rely on access to the gcode.

The basic config settings for the printer will be on the spec page on FlashForge's site. There is an active FlashForge community on Facebook and reddit which might give you hints for tweaks on the Dreamer. I don't have any specific settings for it.

Gamereric21, 2021-01-23:
I am unable to reliably print from octoprint due to 2 issues:

When uploading from SD card, I get a "File content error" (something like that), and it looks like octoprint somehow screws up the file.  If I try saving the file that I uploaded to sd, and put it in manually, the printer syas print complete when it starts, and doesn't do any printing or heating, even though it was sliced in FlashPrint

When printing directly, it works alot better, but theres 2 issues I have.

- as seen here, around the 10th-20th layer, it jumps, and theres a huge gap, leading to consistent failed prints
 
[image]

- The Z axis is also for some reason alot noisier when printing directly.  It clicks, and makes a really irritating noise when it moves, but it doesn't seem to do that when printing from my physical SD card!

Let me know what settings I could change / if you know any possible fixes!

Mrnt, 2021-03-11:
Sorry for the delay in getting back to you. 
re upload to SD: If you have a sample file that you cannot upload to the SD card can you upload it here (you may have to zip it first) and provide the error message that you see when upload fails.

re layer skipping issue: I have occasionally seen something similar but found that I can print the exact same file again and it will work, so still trying to figure out what might be happening there.

re noisy z axis: I think it may be related to the above.

What are you using to slice to the files?