Mrnt/OctoPrint-FlashForge #73: Any ideas of making all other Flashforge products fully compatible with Octoprint

Oxize, 2021-05-16:
Since Flashforge/Bresser Users have to use the Upload to SD to print their stuff, you miss alot functionality. I really would like to have the G-code Viewer, so you can exactly see what the extruder is doing. 

But if we arent using Upload to SD, the whole printer doesnt know where he is, and wacks its extruder head everywhere in the heatbead like a drunken Irish sailor.

Would be so nice to print directly from OctoPrint without having problems.

tckb, 2021-05-16:
the problem is with the stock firmware. Flashforge firmware is only pseudo marlin, meaning only few subset of marlin gcode is supported and this hinders full utilization of octoprint.  I have been work on Marlin fork for Flashforge (https://github.com/tckb/FlashForge_Marlin) if your printer is compatible -- you could give it a try.

Oxize, 2021-05-16:
Well i dont have a flashforge, but a Bresser Saurus, but its all Flashforge inside. Its actually an exact copy of the FlashForge Finder II.

When i asked difficult questions to bresser, i didnt get any normal answer back, so i went to Flashforge, and they give me fully support now. They said they making products for Bresser and all Flashforge inside,  but they have 2 different firmware, but are almost 99,5% the same. Dunno if it wise to use other firmware now. Do you mess up your guarantee with that?

tckb, 2021-05-16:
>have 2 different firmware, but are almost 99,5% the same
How do you know this? I've never heard of bresser saurus but this seems to be a white-label flashforge. Did you take a peek inside the hood?   

>Do you mess up your guarantee with that?
Most likely if you still have one. I'd say you wait

Oxize, 2021-05-16:
No, i didnt look under the hood, but when i was searching for info in the copyright disclaimer of the Bresser Saurus, i saw that Flashforge was mentioned alot of times in there. So i went Flashforge site to check around, and saw the same model "FlashForge Finder II", as me. They have red. I have Lime and Black.

Since Bresser was a bit anoyyed of my difficult questions, i went to flashforge support. And there saying freely that they deliver 3d printer to Bresser, ofcourse with their bresser their logo, and stuff, but inside its Flashforge. Thats why it can communicate with Octoprint, because there are no support for Bresser Printers at all at Octoprint if you check the list. Lucky me i find this out, because the standard software they deliver is shit 🗡️ 

Ye, its still within the guarantee. Flashforge said, i could even use their firmware, but better wait till guarantee ends, otherwise you will get no replacements or whatsoever from Bresser if they found out that your are running other firmware.

tckb, 2021-05-16:
yeah Flashprint works 'just fine' but was too basic for me. I used prusa since the beginning  even while I was on flashforge.

Oxize, 2021-05-16:
Do you any other Slicer which works with FlashForge Finder II? Prusa does?

Which i tried and dont work:

Cura
Simplfy3D (didnt checked fully)

But do you think MNT will try make it full compatible for all devices, or is this all?

tckb, 2021-05-17:
@Oxize  yeah I use Prusa for my DreamerNX, there's already a profile for DreamerNx for Cura