Mrnt/OctoPrint-FlashForge #74: Checking out my Saurus with Flashforge under OctoPrint

Oxize, 2021-05-18:
Since my Bresser Saurus is an refined Flashforge Finder II. Bresser Put some of their own logo stickers on it, but thats it.

[image]

It can connect under Octoprint which @Mrnt his plugin. /Bows

I can play with the lights, move the head, move the bed. 
[image]

What i not can is Put motors on or off, and Fan on or off. But when you press Fan off, and the fan is blowing from old session. And you spamm on Fan off, you hear actually slowing down the fan, but when your stopped spamming the button, its spool up to its normal rpm again.

So i made an bed level test, to test out printing directly from Octoprint to the Saurus/Flashforge

[image]

I loaded it into OctoPrint. When i press print, i get a warning

[image]

The Extruder heats up till 220c, until i cancel the session or when i go to temperature tab and put the tool manual off.

[image]

So i wanted to make a debug file for the guys who are working on this communication-plugin.

So i went to logging and enable as shown below
[image]
I pressed save, and checked again.

[image]

But i experienced more often in Octoprint when you press save buttons, that it isnt really saving. Dunno why, but i adressed this to the maker of OctoPrint.

So what i did was rebooting the whole system and redo all the logging settings again for your flashforge plugin. and Voila!

[image]

Lets redo the bed level test again and gets some files out of this machine....

[image]
This time after a reboot system, when i started the bed level test again, the fans are spinning, the extruder is warming up, and then i get the "Not sending "M140 S0 T0", printer profile has no heated bed. Either configure a heated bed or remove bed commands from your GCODE" warning again.

But i can press print. Last time when i was messing with this, the extruder bangs into the side of the bed. Now actually the bed move a little center middle height, what it normally also does when i print from Rexprint (flashprint).

Pressed print. Silence.......Printer warms up......and cools down again..

https://youtu.be/jIpRoj9Ffp0

Then i realise when i sliced the bed level test, i saw nothing on the bed. Maybe that is the problem, why its not printing. So this time we take a simple cube. A 20x20x10mm Cube.

An brimm around it, because its verry tiny.

[image]

Load and locked and pressed print. Watching carefully.

Printer is heating up, Extruder is warming up. I check temp, check, i check G-code Viewer, and then uhm

[image]

Hey Cube, dont you need to be in the Center of the bed?

Still get warning from " Not sending "M140 S0 T0", printer profile has no heated bed. Either configure a heated bed or remove bed commands from your GCODE".

Waiting...Waiting...Waiting...and then i suddenly hear the extruder/head moving. Oh boy....

He went this time to the center of the bed (but remember that g-code viewer showed left under?), instead banging into the side, where it still left his marks (2 weeks ago)

Made some whole video from the situation.
https://youtu.be/CxScKizio0k

It does something. It wants to print, makes weird noises which arent normal. Is verry slow, like he needs to think what he must do next. So i let him do this a few square rounds, but then canceled the session because i dont want to break my printer.

Do you know where i can get the flashforge logging which i enabled?

[image]

@Mrnt I am pretty techinical. I worked for over 18 years as an Senior Networkspecialist at a big dutch company. But since 2018 i got sick, and anxiety problems. Problem was, is that i was walkinkg around with 18 years of mental problems. Keeping away emotions. Just keep going on work, even if you have alot of downs. This resulted in heavy physicial health problems.

Anyway i am in therapy now, and i have a long way to again, so i have plenty time for this 3d printing Job. If you need any help, let me know.

sorry for my long text/picture story 🍡

Oxize, 2021-05-18:
I have the octoprint log, but i guess you need the debug from flashforgeplugin. Its not stated above in the list. 

[octoprint.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/6500871/octoprint.log)

Oxize, 2021-05-30:
@Mrnt Any chance you can do anything with above?

Oxize, 2021-06-06:
Really wished Finder/Saurus can work under OctoPrint.

Mrnt, 2021-06-08:
Sorry for the really long delay in responding - I have lot going on right now plus problems with my own printer :(

The second video you posted it looks like it is going very slowly - was the head actually too close to the bed and dragging on it? I presume that you already leveled it?

re the G Code Viewer not showing the code centered - you will need to create or edit the printer profile and select the "print bed and volume" tab:

[image]
and make sure to select "Origin: Center"

 
**NOTE: There are two ways to print using the plugin:**

[image]

1) Upload the file directly to the printer's SD card using the "Upload to SD" button, which should also automatically start the print once the file has uploaded. You will only be able to monitor approximate print progress in OctoPrint and not be able to see what is being printed using the G-Code Viewer. HOWEVER, this is the most reliable way to print since it directly uploads the file to the printer is not reliant on any subsequent communication with OctoPrint for the print to finish.

2) Upload a file to OctoPrint using the "Upload" button. You can then subsequently select the file in OctoPrint and print it, see what is being printed using the G-Code Viewer, etc. This method sends commands one by one from OctoPrint to the printer down the USB port. This is NOT reliable but USUALLY works on the two printers I have tested with. Sending print commands to FlashForge printers sequentially like this is not supported or documented by FlashForge probably because it just is not completely reliable and on some printer models seems to be not viable due to the fact that the printer will not support some G code commands delivered via USB even though the same printer will process the same command if it is part of a file uploaded to the printer's SD card. Very frustrating for me and everyone else...

**Please provide (relevant) debug files to help diagnose issues:**

Does "Upload to SD" work for you? If not can you:
1) Go to the "Logs" screen in OctoPrint and delete "octoprint.log"
2) Restart OctoPrint, reconnect OctoPrint to your printer, etc
3) Try uploading a file to your printer using "Upload to SD" and when you get to the point where the process fails, download the "octoprint.log" file and then upload it here as a zip file.

If "Upload to SD" works but it is the "Upload" and subsequent print that fails, can you:
1) Go to the "Logs" screen in OctoPrint and delete "octoprint.log"
2) Restart OctoPrint, reconnect OctoPrint to your printer, etc
3) Try uploading a file to OctoPrint using "Upload"
4) Start printing the file and when you get to the point where the process fails, download the "octoprint.log" file and then upload it here as a zip file.

Mrnt, 2021-06-08:
Ah, just saw you DID upload a log file - GitHub marked it as "off topic". I'll take a look at that.

re the things you noted in the OctoPrint control pane:

You SHOULD be able to change LED color - I did test that on the Finder but it is possible that the firmware is slightly different in your machine since it is newer. 
The extrude/retract SHOULD also work BUT you will need to heat the filament to be able to see anything happen.

Are you familiar with WireShark? If so you can use that to sniff a USB connection between RexPrint/FlashPrint and the printer to see what command(s) are actually sent when you change LED color, extrude/retract, fan on/off.

Oxize, 2021-06-08:
I am not sure about an RGB Led in the Saurus.

Ye i know Wireshark, but never used it on a USB connection. The Saurus is connected with USB to the PI. Can i still sniff that out on my windows machine? Using wireshark under Rasbian or something i never did.

Mrnt, 2021-06-08:
What I had to do was connect the printer to my desktop via USB and then control the printer LED color, fan, extruder, x/y/z movement, using FlashPrint while sniffing the USB traffic using Wireshark. I did this on a Mac but I believe you can do it on a Windows machine too. It's a pain in the you know what :)

Have to tried disabling all plugins that you installed except FlashForge? For example I see Astroprint which generated an error early in one print.

Oxize, 2021-06-08:
Will that tommorrow (today). Its 03.22 am here. Kinda sleepy.

[image]

Oxize, 2021-06-08:
Wireshark Log: (only selected the saurus in wireshark, because you can have more devices on 1 usb controller)
[saurus_new.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/6614121/saurus_new.zip)

New Fresh log. Bed lvled before and Extruder head also lined out -1.0 Z-axes. Also made the change centered in profiles before starting.

Log:

[octoprint_last.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/6614133/octoprint_last.log)

And video of it:

**https://www.youtube.com/watch?v=GKIpnRu_s20**

Mrnt, 2021-06-08:
Thanks for the Wireshark log - I see the LED command is slightly different from what the plugin uses. Did you try turning the fan on/off or test anything else while capturing that log?

In the video:
When you have the camera on the terminal window the terminal output pauses and then several messages pass by too quickly to read then I see the error about M140 - do you know what those messages are? 
It looks like the printer starts printing before the extruder has reached the target temperature (I cannot quite read the temperatures on the display)? That noise might be the extruder slipping on the filament if the filament is not hot enough to extrude. 

The log file you just uploaded does not contain any debug messages from the falshforge plugin - did debugging get turned off somehow? Also please disable any plugins that are not part of the OctoPrint bundle while isolating this issue.

Oxize, 2021-06-08:
Rexprint Control Panel is slightly different than Flashprint. I cant control the fans in Rexprint. I pushed the extruder in forward and reverse position through the control panel and played with some axes and temperature. More i cant do in that Control Panel.

I should have copied the terminal output, but didnt really think about that. The warning was about the heatbed which wasnt there. You see me also focussing on that warning on the right where that M140 pops up.

And yes the temperature was way to low. What normally happens is that the bed moves like halfway up, waits like 10 degrees before the configured temperature and than moves up and you see the extruder already slowly dripping filament.

What in the video happend is that it went way to early, and made that slippery noises. But the movement wasnt verry smooth like you saw in the video. It stopped something and making a big puddle of filament and then went on for a few cm, stopped agian, and moved again.

I will do a fresh testcube print again. I will check the flashforge plugin and disable any other plugin that arent from the Octoprint Bundle.

Oxize, 2021-06-08:
Terminal output: 
[terminal.txt](https://github.com/Mrnt/OctoPrint-FlashForge/files/6618673/terminal.txt)

Octoprint with less plugins and flashforge log on debug.

[octoprint_with_flash.log](https://github.com/Mrnt/OctoPrint-FlashForge/files/6618675/octoprint_with_flash.log)

This time i paused the print for like 1min to get the extruder to 220c. But octoprint didnt really liked that, disconnected from the printer, so i reconnected redid the print. The extruder was still at 220c.

When he started printing he started in the center. First lines goes okay, but then on the corner the extruder stops moving and start flooding the bed with filament. At this point you hear that rattle noise when he is stop moving and making those puddles.

Disconnected power after that.

Oxize, 2021-06-11:
[image]