Mrnt/OctoPrint-FlashForge #78: Flashforge Creator Pro 2 issues

Parker-Bergen, 2021-12-03:
I am able to connect and print with the creator pro 2, but when I try to print a file in duplicate or mirror mode, one one extruder moves, but both heat up.
I also can not save files or access the SD card files from in octoprint.
Other than those issues the creator pro 2 works great with this plugin

Parker-Bergen, 2021-12-03:
SD Card Write Error Output:
`
2021-12-03 20:15:29,877 - octoprint.plugins.flashforge - INFO - Starting SDCard upload from SpacerDup.gx to SpacerDup.gx
2021-12-03 20:15:31,173 - octoprint.plugins.flashforge - INFO - Upload failed: Unable to upload to SD card: b'CMD M28 Received.\r\nopen failed, File: 0:/user/SpacerDup.gx.\r\nok\r\n'
`

Mrnt, 2021-12-03:
Can you explain what you mean by duplicate/mirror mode?

re SD access by OctoPrint:
Unfortunately this is severely limited in FlashForge printers, which only support a subset of the G code commands commonly used by 3D printers. Although OctoPrint supports interrogating a printer for a list of files available in the printer memory or SD card (as well as file management such as deletion, etc), the FlashForge printers do not support this. They will allow a file that is on the SD card to be printed if the name is passed using the [M23 command](https://github.com/Mrnt/OctoPrint-FlashForge/wiki/G-Code-Reference#m23---print-file-stored-on-sd-card), which is a command generated by OctoPrint when a file is successfully uploaded to the printer SD card using the "Upload to SD Card" button (where the file is uploaded to the Printer SD card rather than than uploaded to OctoPrint), but since there is no way for OctoPrint to know what is on the SD card the command is limited to that one use case.
However, as you noted, a limitation of printing from the printer's SD card is that OctoPrint no longer controls the printing directly, so while you can see approx percentage print progress you are unable to see progress layer by layer in the OctoPrint interface, although you can still monitor using a camera, see temperatures and abort the print.

re the SD card upload error you showed above - it is hard to say what happened without more context, but things to check would be:
- SD card being full or corrupt
- make sure you don't have the touch panel on the printer displaying something other than the home screen (eg if it is showing the print job progress of a previous job that was paused or cancelled, then I have noticed some erratic behavior wrt OctoPrint). 

Printing to the FlashForge printers via upload to the SD card is generally pretty reliable, printing directly from OctoPrint to the printer is a bit more hit or miss even for me and I have spent many hours trying to track down why :(

Parker-Bergen, 2021-12-03:
Okay.
As for the duplicate and mirror mode, with an IDEX printer you can enable duplicate mode and it will print the same object at the same time by splitting the build plate in half and using both independent extruders to build. It is the same thing for mirror mode but mirrored objects instead of duplicated. When I try to print in either of these modes then I do not get the intended result. The gcode does not contain the gcode command for it but the printer will still understand when duplication/mirror mode is active. I can only assume its in the encoded header at the beginning of the file but I am not sure.

Mrnt, 2021-12-04:
Ah got it, I had forgotten that there are independent head dual extruders and was not aware of Idex. I’ll need to investigate, but there’s probably one or more g-code commands right at the beginning of the file to indicate an Idex mode and it may be filtered out by the plug-in.

Mrnt, 2021-12-04:
Can you confirm the observed behavior is this: when uploading an IDEX file generated by FlashPrint that contains a mirror OR duplicate print to OctoPrint (not SD card) and printing the file from OctoPrint, only one of the two prints is printed (I'm guessing the right hand one)?

I'm presuming that you are running this on a Raspberry Pi? Would you be comfortable editing one of the plugin source files directly as a test?

Parker-Bergen, 2021-12-04:
Yes this is the behavior that I am seeing, I think occasionally the left extruder will possibly extrude but not move. The right hand one is the only one moving as well as you described. 
Yes I am running it on a raspberry pi and I am willing to edit a file to test.

Mrnt, 2021-12-06:
Can you delete lines 304,305 in octoprint_flashforge/__init__.py:
```
			# M109 in Marlin = wait for extruder temp : M6 in FlashForge (this may need to be moved to the write() method)
			elif gcode == "M109":
				cmd = [cmd.replace("M109", "M6")]

```
and then try printing an IDEX file (generated by FlashPrint).

On my OctoPi install the file can be found at the following path:
```
./oprint/lib/python3.7/site-packages/octoprint_flashforge/__init__.py
```
and so you could edit it with nano as follows:
```
nano ./oprint/lib/python3.7/site-packages/octoprint_flashforge/__init__.py
```

Parker-Bergen, 2021-12-10:
Okay so I commented out those lines of code and the problem was not resolved. I contacted FlashForge and the did say that the code for duplicate was M109. So those lines made sense to me. I can tell you exactly what happened however. Both extruders and the build plate heated up as normal. When the model went to print, only the right extruder was moving but both were extruding. I have linked the code for the print. It is a small 1 mm spacer with a pre-extrude outline. I see the M109 code in this file but farther down then I would have thought.
[SpacerDup.gx.zip](https://github.com/Mrnt/OctoPrint-FlashForge/files/7689211/SpacerDup.gx.zip)

Parker-Bergen, 2022-01-07:
Any updates or anyway I can help further to figure this issue out?

Kleinja, 2022-01-11:
I just picked up the Creator Pro 2 a few days ago and have been working to get Octopi up and running on it the last couple of days. I just wanted to post and say I am seeing a lot of the similar issues as OP. 

The issues I am most concerned with is the SD card (in the machine). I am unable to view the files on it from Octoprint, and unable to upload to SD as well. When attempting to upload I get the error on screen: "There was an error while trying to start a print job. Error: Unable to upload to SD card - could not create file on printer SD card.". Not sure if it saves a more detailed log somewhere else, I am pretty fresh to Octoprint still. My install of Octoprint is recent, and really the only changes I made were in the tutorial on the homepage (most auto filled, aside from the GCODE scripts needing clearing out). 

I tried to print once using the Upload button, but the raft was printing very weird (like it was dotting the filament, rather than printing a line). Cancelled it pretty quick once I noticed that. Not sure what exactly was going on, and I can try again and provide more detail if needed. Though I did copy the same file to the SD Card (manually) and it printed as expected, so the file seemed to be fine. 

When printing manually from touchscreen off of SD Card, I get temp data as well, but nothing else. State shows as operational, but no other parameters show up. This is ultimately fine for me, as I mainly want Octoprint to interface with my home automation hub, so I know when prints are complete. I can use the temperature of the extruders to accomplish this, though it would be nice to report a percentage or estimated time left, rather than just printing / complete (which I could complete with a power monitoring smart outlet). 

I have only printed one duplicate print, and that was before Octoprint was setup. 

Just wanted to introduce myself as another with a creator pro 2, and see if there are any ways I can help progress this plugin as well. So far its been awesome, and really extends the capability of the printer!

Kleinja, 2022-01-12:
Noticed a couple more odd things with the creator pro 2, figured I would share those as well. First one is simple, and for me the light control doesn't work. I have the buttons on the interface, but they do nothing. Not sure if they use a different Gcode command for the lights in the Pro 2, but that one really doesn't bother me. Just figured I would mention it. 

The second thing I noticed, is a little more interesting. If I connect to the printer with octoprint, and then start a print (via touchscreen on the printer) from SD card the state in octoprint stays as operational. Though, if I start the print, and then connect with octoprint while it is heating up, it shows the state as printing from SD. I also now have the pause and cancel buttons available to me, and the print counter starts counting. 

Sort of weird behavior, and not sure why it doesn't pickup its printing from SD card when already connected? Figured I would share though

Kleinja, 2022-02-25:
Any updates to this? I was really hoping to get network storage setup on my creator pro 2. Haven't had octoprint up and running much since because it is really no use to me in its current condition.