Mrnt/OctoPrint-FlashForge #79: Flashforge Dreamer not detected by OctoPi

DerpBlerp, 2022-01-08:
After setting up OctoPi, and instaling the plugin, making a connection attempt doesn't return anything, nothing about a permission error, just no candidate

Using a Raspberry Pi Zero 2W, the printer has Wifi enabled as it was setup prior to setting up the the raspberry pi

here's an image of the set up, the red cable providing power
[image]

and a copy of the terminal
```
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 0 port/baudrate candidates: 
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
```

DerpBlerp, 2022-01-09:
[octoprint (1).log](https://github.com/Mrnt/OctoPrint-FlashForge/files/7835813/octoprint.1.log)
here's the log

Mrnt, 2022-01-14:
Thank you for including the log - it's very hard to figure out what is happening without it!

Looking through the log you can see the section where it is trying to detect USB devices, for example:
```
2022-01-09 20:00:52,802 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2022-01-09 20:00:52,804 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2022-01-09 20:00:52,813 - octoprint.plugins.flashforge - DEBUG - Unable to get device name LIBUSB_ERROR_ACCESS [-3]
2022-01-09 20:00:52,813 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown device' with Vendor ID: 0X1D6B, USB ID: 0X0002
2022-01-09 20:00:53,141 - octoprint.plugins.flashforge - DEBUG - detect_printer()
2022-01-09 20:00:53,144 - octoprint.plugins.flashforge - DEBUG - Unable to get device name LIBUSB_ERROR_ACCESS [-3]
2022-01-09 20:00:53,145 - octoprint.plugins.flashforge - DEBUG - Found device 'unknown device' with Vendor ID: 0X1D6B, USB ID: 0X0002
```
Notice how it only finds the one device, which according to http://www.linux-usb.org/usb.ids is `1d6b  Linux Foundation USB 2.0 hub`

The fact that it is not listing the Vendor ID and device ID of the printer makes me wonder if somehow the printer is not connected to the Pi? Maybe you have the power connector in the USB Host/OTG port and the printer connected to the (USB) power port?

DerpBlerp, 2022-01-16:
it occurred as a possibility to me but i never tried to go through with it cus i don't fully understand linux/ raspberry pi

closed 2022-03-09 by Mrnt