Mrnt/OctoPrint-FlashForge #80: Trouble connecting to Dreamer, port/baudrate issue

schmooot, 2022-09-19:
This is a fresh install of octopi on a pi4.  The only plugin I have installed is this one.  I know the cable is good as I can plug in to my laptop and it connects with the slicer no problem.

Initially I got the permissions error and had to add the line in my 99-octoprint.rules file, I rebooted the pi and the printer.
Then it gives a solid attempt but still no success.
Here's the terminal log.

```
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: FlashForge Dreamer 3D Printer, port:1:3@115200, FlashForge Dreamer 3D Printer, port:1:3@250000, FlashForge Dreamer 3D Printer, port:1:3@230400, FlashForge Dreamer 3D Printer, port:1:3@57600, FlashForge Dreamer 3D Printer, port:1:3@38400, FlashForge Dreamer 3D Printer, port:1:3@19200, FlashForge Dreamer 3D Printer, port:1:3@9600
Trying port FlashForge Dreamer 3D Printer, port:1:3, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: <octoprint_flashforge.flashforge.FlashForge object at 0xac8ced50>, starting monitor
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port FlashForge Dreamer 3D Printer, port:1:3, baudrate 250000
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port FlashForge Dreamer 3D Printer, port:1:3, baudrate 230400
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port FlashForge Dreamer 3D Printer, port:1:3, baudrate 57600
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port FlashForge Dreamer 3D Printer, port:1:3, baudrate 38400
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port FlashForge Dreamer 3D Printer, port:1:3, baudrate 19200
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port FlashForge Dreamer 3D Printer, port:1:3, baudrate 9600
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

I've tried playing with a few of the settings in the serial connection tab but didn't get any results.  I don't know where else to go from here.  Any suggestions?

Mrnt, 2022-09-26:
Did you:
- check you had set up all the settings as described in the install/setup guide for the plugin?
- turn on debugging as described in the troubleshooting notes and then download the octoprint.log file Settings > Logging? There should be quite a bit of additional information to help troubleshoot.

schmooot, 2022-09-28:
> Did you:
>
> * check you had set up all the settings as described in the install/setup guide for the plugin?
> * turn on debugging as described in the troubleshooting notes and then download the [...]

Well that's embarassing.  For some reason I didn't read anything below the installation section.  Ha.  I followed the directions in the config section and it connected immediately.  

Thank you for that

closed 2023-02-07 by Mrnt