Mrnt/OctoPrint-FlashForge #85: how to fix this error it does not want to connect to the printer

gabbelitoV2, 2023-04-11:
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyS9@115200, /dev/ttyS9@250000, /dev/ttyS9@230400, /dev/ttyS9@57600, /dev/ttyS9@38400, /dev/ttyS9@19200, /dev/ttyS9@9600
Trying port /dev/ttyS9, baudrate 115200
Connecting to port /dev/ttyS9, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x7f7c1a1c40, open=True>(port='/dev/ttyS9', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 250000
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 230400
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 57600
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 38400
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 19200
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 9600
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyS9@115200, /dev/ttyS9@250000, /dev/ttyS9@230400, /dev/ttyS9@57600, /dev/ttyS9@38400, /dev/ttyS9@19200, /dev/ttyS9@9600
Trying port /dev/ttyS9, baudrate 115200
Connecting to port /dev/ttyS9, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x7f7c15a9a0, open=True>(port='/dev/ttyS9', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 250000
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
[... 147 lines cut ...]
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 57600
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 38400
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0
Handshake attempt #3 with timeout 2.0s
Send: M601 S0
Trying port /dev/ttyS9, baudrate 19200
Handshake attempt #1 with timeout 2.0s
Send: M601 S0
Handshake attempt #2 with timeout 2.0s
Send: M601 S0

gabbelitoV2, 2023-04-11:
does it even work with Adventurer 3?

gabbelitoV2, 2023-04-11:
eat the serial number you should have?

gabbelitoV2, 2023-04-11:
is it not possible to take a cable that goes from a usb to a ubs and use the usb hole on the front of the printer to connect it to the computer to get the id

Mrnt, 2023-05-08:
Sorry for the late response. The USB connector on the front is for plugging in a USB flash drive - the printer uses this port as a USB "host" and access the drive as a USB "client" or accessory. In order to use the printer with OctoPrint via USB, the printer has to act as a USB client with the RaspberryPi (or whatever is running Octoprint) acting as the host. The USB client ports are typically one of the following - USB-B, USB mini or USB micro:

[image]